VoodooI2C/VoodooI2C #26: Bug for Xiaomi Mi Notebook Air 12

GrampassoNordico, 2017-01-30:
Hello! First of all I'd like to thank you for your good work. 
When I inject the kext from the Terminal the trackpad somehow works, but as I move the pointer, it continue moving forever until I touch again the surface. There's something we can do to correct this bug?

Liryc-R, 2017-01-30:
[MacForceOne](https://github.com/MacForceOne/VoodooI2C/tree/patch-1)'s version resolve your issue.

GrampassoNordico, 2017-01-31:
Yes it did, wow! I've one last thing to ask. When I move my finger normally the pointer moves very slow, then when I move it quickly it moves fine. Changing mouse sensitivity in the Mouse panel didn't resolve this. Is there a way to change "motion curves" or some parameters that a noob like me could edit in an easy way?

closed 2017-03-01 by alexandred