VoodooI2C/VoodooI2C #31: trackpad Not Working after sleep

sooxiaotong, 2017-04-05:


alexandred, 2017-06-19:
While I am keen to fix this issue, I will have to close this as you have provided no details.

closed 2017-06-19 by alexandred

fubizitch, 2017-10-30:
I am also having this issue, and I know others have.  After the computer goes to sleep for a length of time, the track pad will no longer function after the computer wakes back up.  I have to reboot to get the track pad to work again.

Asus UX305CA-OHM7-WH

fubizitch, 2017-10-30:
I have El Capitan 10.11.6 with the latest VoodooI2C driver installed.  I used the guide from https://www.tonymacx86.com/threads/guide-asus-ux305ca-uhd.197229/ to install.

Any other information you need from me?

alexandred, 2017-10-30:
This issue has been fixed in the rewrite. No ETA for it yet. Also note that this issue is closed.