VoodooI2C/VoodooI2C #41: request for High Sierra and Elan I2C

thefiredragon, 2017-07-16:
Hey there, it's nice work,
I want to request if high Sierra will be supported in future?
At the moment we're getting kernel panics on the last beta build.

the next request would be, If it could be possible to bring up new elan touchpad working on macOS?

HID\VEN_ELAN&DEV_0651&Col01
HID\ELAN0651&Col01
HID*ELAN0651&Col01 

older ones like this are working correctly.
static const struct acpi_device_id elan_acpi_id[] = {
	{ "ELAN0000", 0 },
	{ "ELAN0100", 0 },
	{ "ELAN0600", 0 },
	{ "ELAN0605", 0 },
	{ "ELAN1000", 0 },
	{ }
};

ben9923, 2017-07-16:
I don’t think your issue is caused because of macOS High Sierra.
I think it’s a patch you are missing, or it’s something with the Lpss kexts (Check if they are loaded under your I2C device with IORegistryExplorer. If they are loaded there try removing AppleIntelLpssI2C + AppleIntelLpssI2CController + AppleHPM and try loading VoodooI2C again)
If injected with Clover, try installing to S/L/E or manually loading with terminal.
With macOS 10.12 it’s working? Same DSDT and similar config.plist (when the only change is 10.13 optimization)?

thefiredragon, 2017-07-16:
I had try it out but I only get force Restarts under high sierra. 

https://image.ibb.co/ieZYQF/Screenshot_20170716_130116.png

I will test it later on Sierra.

thefiredragon, 2017-07-18:
Okay, I'm sorry I am, I had read threads about I2C and how it works.

So the BootIssue is not related to the kext, so I need first to patch my ACPI Tables right.
The question there is that my Elan Model is not listed. 
Could the ELAN0651 be integrated when there's a supported binary for Linux?
It would be important to know. When not I will be switch to an DELL XPS
Thanks again for your work here

alexandred, 2017-07-18:
I am going to be closing this thread as it is not longer about High Sierra support. All questions pertaining to individual devices should be posted in the gitter chat: https://gitter.im/alexandred/VoodooI2C .

closed 2017-07-18 by alexandred