VoodooI2C/VoodooI2C #43: Gesture Configuration and Extended Gestures

kprinssu, 2017-07-26:
Currently VoodooI2C only has support for gestures up to 3 fingers, although the kext can theoretically support up to 15 fingers. To match functionality with macOS, the project should add support up to 5 fingers.

Secondly, only simple swipes are detected and support for more "advanced" features such as pinch and squeeze/splay can be added.

Lastly, the action for each swipe is hard coded into the kext and it would be nice to have customisability. We should consider following what VoodooPS2 has or implement something better.

coolstar, 2017-07-30:
csgesture is just a stop-gap that's in place temporarily. Ideally we would want to switch over to emulating a macbook or magic trackpad so we can use the native gesture engine instead.

Effort is probably better spent on reversing the apple trackpads instead of extending csgesture :)

alexandred, 2017-12-22:
I am closing this considering that we are homing in on native multitouch emulation.

closed 2017-12-22 by alexandred