VoodooI2C/VoodooI2C #44: cursor drifts without input

ldartez, 2017-08-24:
I'm not sure if this is the right place to bring this up. I followed [this post](https://www.tonymacx86.com/threads/wip-voodooi2c-i2c-trackpad-limited-support.204227/) on how to use VoodooI2C regain control of my laptop's trackpad. My method and results are [here](https://www.tonymacx86.com/threads/installing-sierra-on-razer-blade-14-2017-kabylake.230106/). I am not intricately familiar with the os x kernel, i might be able to help with guidance/input. 

Once I got my trackpad to work, it became only somewhat responsive: as if the sensitivity was turned almost all the way down. In addition, whenever I stop or lift my finger off the trackpad the cursor continues to move in the original direction. Only excessive clicking seems to stop the cursor. In effect, the trackpad is still unusable for my current configuration. 

Can someone point me in the right direction for getting this fixed?

alexandred, 2017-08-24:
We are currently rewriting the enrire code base and this issue has been fixed.

closed 2017-08-24 by alexandred

ldartez, 2017-08-25:
if it's been fixed can I please be pointed a working copy?

alexandred, 2017-08-25:
The rewrite is not yet ready for general consumption so unfortunately you will have to wait until we release it (or start beta testing). Porting the fix over to the original VoodooI2C codebase is also too complicated a task to be worth the effort.

I just remembered that MacForceOne's fork has a different fix that might work but it's on a fairly old version of the original codebase, you can try that if you like.