VoodooI2C/VoodooI2C #46: Wacom EMR support pressure

axel9546, 2017-10-16:
Hello! I just tested this awesome driver!
I can help you to dump something you need by linux or windows or osx, but wacom emr digitizer in my Galaxy Book 12 seem only to work with point pen, and not with rpessure.
Infact the pen is recognized well and work very well.. But no pressure at all!
Can i help to make this dream possible??

(ps also touch dont work in 10.13 in GB12)

axel9546, 2017-10-16:
Thanks! Sorry for my noob mind, but what is the rewrite release.?

blankmac, 2017-10-16:
@alexandred has rewritten the entire kext from the ground up...it should be ready for release soon.

axel9546, 2017-10-16:
@blankmac wow seriuoulsy! Thanks! This is an amazing thing!!
I need your help to decide. As i need this tablet for drawing, on windows i am feeling bad, i need osx like old times.
Should i install osx El Capitan and so get the pressure sensitivity work but no touch in that case (with the rewrite release will fix touch also on elcapitan??), or stick on High Sierra, knowing that i have no possibility to pressure sensitivity?
Also do you know if i can change the pressure sensitivity "Strings" in the HID event driver? 
Sorry but im a noob at this stage >)

blankmac, 2017-10-16:
I can appreciate not wanting to give up on something, I spent a considerable amount of time looking for a solution, but it's very simple -- there's nothing you can do to enable pressure sensitivity in 10.12+, Apple removed the capability, so if that's the make/break feature for you, then installing El Capitan is your best bet.  You can't add anything from El Capitan to enable it, those functions have been eliminated from the source.

closed 2017-10-16 by blankmac

axel9546, 2017-10-16:
Okay thannks!! Last question 👍 
With the rewrite of the kext, i will have touch support on elcapitan too?? thanks

blankmac, 2017-10-16:
It is untested to my knowledge, but there are no technical reasons that I know of at this time for it not to work.

axel9546, 2017-10-16:
Thank you very much for the information! Looking forward to the rewrite kext

alexandred, 2017-10-16:
The rewrite works fine on El Capitan, that's my main OS. We are targeting 10.11+ for the rewrite.

blankmac, 2017-10-16:
Thanks for clarifying.   Thought you were on Sierra for some reason.

axel9546, 2017-10-17:
Thanks for clarifyng me too! 
I think el capitan is one of my best os! I ve installed h sierra only.for seeing the effect of metal 2 (el capitan gpu api are?? Still opengl??)
Also i would like to thank you for the effort, and focus on this project! 
Looking forward to draw on the Galaxy Book, on elcapitan!!