VoodooI2C/VoodooI2C #55: Invalid signature

aweorit, 2018-01-15:
Hello

I clone the repo and built all the kexts in Xcode 9.2, but they won't load because of an invalid signature:

```
Invalid signature -67050 for kext <OSKext 0x7fbee6d088b0 [0x7fff9608e5b0]> { URL = "file:///Library/Extensions/VoodooI2C.kext/", ID = "com.alexandred.VoodooI2C" }
```

How can I make it load? I have already disabled SIP. Should I use kext dev mode?

aweorit, 2018-01-15:
Is there any way to enable right click with Elan touchpad?

alexandred, 2018-01-15:
No there isn't.

closed 2018-01-15 by alexandred

aweorit, 2018-01-15:
OK, what about the other questions?

alexandred, 2018-01-16:
The invalid signature warning is not an error. If you didn't know that then you probably shouldn't be fiddling with development kexts and you should just wait for the release.