VoodooI2C/VoodooI2C #56: compile from source error

1and1get2, 2018-01-17:
I noticed that the pre-build binary hasn't been updated for quite some time, so I decided to compile myself and maybe get some ideas about the functionalities, however, I'm facing some problem trying to do so.

at first, it complained about unable to find cldoc, solved by running sudo pip install cldoc
(for future reference)

1and1get2, 2018-01-17:
and then, i was greeted by this:

```Showing Recent Issues
PhaseScriptExecution Generate\ Documentation /Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Intermediates.noindex/VoodooI2C.build/Release/VoodooI2C.build/Script-AC4EEE2D1F3B2F1D00D0A6E0.sh
    cd /Users/derek/Workspace/Hackintosh/Projects/Clone/Others/VoodooI2C/VoodooI2C
    export ACTION=build
    export ALTERNATE_GROUP=staff
    export ALTERNATE_MODE=u+w,go-w,a+rX
    export ALTERNATE_OWNER=derek
    export ALWAYS_EMBED_SWIFT_STANDARD_LIBRARIES=NO
    export ALWAYS_SEARCH_USER_PATHS=YES
    export ALWAYS_USE_SEPARATE_HEADERMAPS=NO
    export APPLE_INTERNAL_DEVELOPER_DIR=/AppleInternal/Developer
    export APPLE_INTERNAL_DIR=/AppleInternal
    export APPLE_INTERNAL_DOCUMENTATION_DIR=/AppleInternal/Documentation
    export APPLE_INTERNAL_LIBRARY_DIR=/AppleInternal/Library
    export APPLE_INTERNAL_TOOLS=/AppleInternal/Developer/Tools
    export APPLICATION_EXTENSION_API_ONLY=NO
    export APPLY_RULES_IN_COPY_FILES=NO
    export ARCHS=x86_64
    export ARCHS_STANDARD=x86_64
    export ARCHS_STANDARD_32_64_BIT="x86_64 i386"
    export ARCHS_STANDARD_32_BIT=i386
    export ARCHS_STANDARD_64_BIT=x86_64
    export ARCHS_STANDARD_INCLUDING_64_BIT=x86_64
    export AVAILABLE_PLATFORMS="appletvos appletvsimulator iphoneos iphonesimulator macosx watchos watchsimulator"
    export BITCODE_GENERATION_MODE=marker
    export BUILD_ACTIVE_RESOURCES_ONLY=NO
    export BUILD_COMPONENTS="headers build"
    export BUILD_DIR=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Products
    export BUILD_ROOT=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Products
    export BUILD_STYLE=
    export BUILD_VARIANTS=normal
    export BUILT_PRODUCTS_DIR=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Products/Release
    export CACHE_ROOT=/var/folders/05/x1pr_s8n1655dk8q_h_35fl80000gn/C/com.apple.DeveloperTools/9.2-9C40b/Xcode
    export CCHROOT=/var/folders/05/x1pr_s8n1655dk8q_h_35fl80000gn/C/com.apple.DeveloperTools/9.2-9C40b/Xcode
    export CHMOD=/bin/chmod
    export CHOWN=/usr/sbin/chown
    export CLANG_ANALYZER_NONNULL=YES
    export CLANG_CXX_LANGUAGE_STANDARD=c++0x
    export CLANG_CXX_LIBRARY=libc++
    export CLANG_ENABLE_MODULES=YES
    export CLANG_ENABLE_OBJC_ARC=YES
    export CLANG_MODULES_BUILD_SESSION_FILE=/Users/derek/Library/Caches/Xcode/DerivedData/ModuleCache/Session.modulevalidation
    export CLANG_WARN_BOOL_CONVERSION=YES
    export CLANG_WARN_CONSTANT_CONVERSION=YES
    export CLANG_WARN_DIRECT_OBJC_ISA_USAGE=YES_ERROR
    export CLANG_WARN_DOCUMENTATION_COMMENTS=NO
    export CLANG_WARN_EMPTY_BODY=YES
    export CLANG_WARN_ENUM_CONVERSION=YES
    export CLANG_WARN_INFINITE_RECURSION=YES
    export CLANG_WARN_INT_CONVERSION=YES
    export CLANG_WARN_OBJC_ROOT_CLASS=YES_ERROR
    export CLANG_WARN_SUSPICIOUS_MOVES=YES
    export CLANG_WARN_UNREACHABLE_CODE=YES
    export CLANG_WARN__DUPLICATE_METHOD_MATCH=YES
    export CLASS_FILE_DIR=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Intermediates.noindex/VoodooI2C.build/Release/VoodooI2C.build/JavaClasses
    export CLEAN_PRECOMPS=YES
    export CLONE_HEADERS=NO
    export CODESIGNING_FOLDER_PATH=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Products/Release/VoodooI2C.kext
    export CODE_SIGNING_ALLOWED=YES
    export CODE_SIGN_IDENTITY=-
    export COLOR_DIAGNOSTICS=NO
    export COMBINE_HIDPI_IMAGES=YES
    export COMMAND_MODE=legacy
    export COMPILER_INDEX_STORE_ENABLE=Default
    export COMPOSITE_SDK_DIRS=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Intermediates.noindex/CompositeSDKs
    export COMPRESS_PNG_FILES=NO
    export CONFIGURATION=Release
    export CONFIGURATION_BUILD_DIR=/Users/derek/Library/Caches/Xcode/DerivedData/VoodooI2C-brzqagexrxlryehdvgrxmwevsawr/Build/Products/Release
[... 349 lines cut ...]
Traceback (most recent call last):
  File "/usr/local/bin/cldoc", line 9, in <module>
    load_entry_point('cldoc==1.9', 'console_scripts', 'cldoc')()
  File "/Library/Python/2.7/site-packages/cldoc/__init__.py", line 57, in run
    run_generate(rest)
  File "/Library/Python/2.7/site-packages/cldoc/__init__.py", line 27, in run_generate
    cmdgenerate.run(args)
  File "/Library/Python/2.7/site-packages/cldoc/cmdgenerate.py", line 125, in run
    t.process()
  File "/Library/Python/2.7/site-packages/cldoc/tree.py", line 212, in process
    self.visit(tu.cursor.get_children())
  File "/Library/Python/2.7/site-packages/cldoc/tree.py", line 517, in visit
    self.visit(item.get_children(), node)
  File "/Library/Python/2.7/site-packages/cldoc/tree.py", line 502, in visit
    node = cls(item, None)
  File "/Library/Python/2.7/site-packages/cldoc/nodes/variable.py", line 24, in __init__
    self.type = Type(cursor.type, cursor=cursor)
  File "/Library/Python/2.7/site-packages/cldoc/nodes/ctype.py", line 58, in __init__
    self._kind = tp.kind
  File "/Library/Python/2.7/site-packages/cldoc/clang/cindex.py", line 1627, in kind
    return TypeKind.from_id(self._kind_id)
  File "/Library/Python/2.7/site-packages/cldoc/clang/cindex.py", line 1519, in from_id
    raise ValueError,'Unknown type kind %d' % id
ValueError: Unknown type kind 119
Command /bin/sh failed with exit code 1`` `

1and1get2, 2018-01-17:
So, I've downloaded the latest release and put VoodooI2C.kext VoodooI2CELAN.kext VoodooI2CHID.kext under /EFI/Clover/kexts/Other. As I'm on Haswell (i7-4720hq + HM87 motherboard), as I understand, I dont need to patch my dsdt other than windows patch(correct?)

 after reboot, internal keyboard and trackpad don't work. and a kextstat shows:

```
derek-mac:~ derek$ kextstat |grep -y "Voodoo"
   46    0 0xffffff7f834fb000 0x5000     0x5000     com.alexandred.VoodooI2CServices (1) 71EEA5EA-0F7E-3988-B2A2-95C6F9FEFC40 <4 3>
```

I checked my dsdt (both original and patched) and i was able to find supposedly supported device
I2C1 _HID, "INT33C3" and I2C0 _HID, "INT33C2". 

my dsdt can be found [AML](https://github.com/1and1get2/CP65S/tree/master/dump/dsdt_origin) and [fixed_dsl](https://github.com/1and1get2/CP65S/tree/master/DSDT/original_dsl_error_free)

1and1get2, 2018-01-17:
diagnostic files (clover log ioreg etc)
https://mega.nz/#!CRh3xAQQ!oe7HVlnRC2cq6_ZphV_N7IbBNyC5KfcKZEA8p7H8J2I

1and1get2, 2018-01-17:
I'm following (trying to) this [guide](https://github.com/alexandred/VoodooI2C/blob/master/Documentation/GPIO%20Pinning.md) and I'm totally lost.

first of all, it seems my devices are \_SB.PCI0.LPCB.PS2K  (for the internal keyboard, IOInterruptSpecifiers: 0x01) and \_SB.PCI0.LPCB.SYNM (for the trackpad IOInterruptSpecifiers: 0x0c) and none of the IOInterruptSpecifiers can be found on the given [link](https://github.com/coreboot/coreboot/blob/master/src/soc/intel/skylake/include/soc/gpio_defs.h#L43) I supposed it's cuz mine's not sky lake?

alexandred, 2018-01-17:
The official cldoc is no good for VoodooI2C. I have modified it for use in VoodooI2C but haven't made it public yet. You don't need Cldoc anyway, you can just remove the documentation generation.

There are many issues here which suggest to me that you haven't properly read the documentation page. First of all, I cannot find an IOReg in your zip. Secondly, you do not need both VoodooI2CHID and VoodooI2CElan, this is outlined in the satellite kext page found [here](https://voodooi2c.github.io/#Satellite Kexts/Satellite Kexts).

Moreover, your system is Haswell so you do not need to apply GPIO pinning, this is outlined on the instruction page found [here](https://voodooi2c.github.io/#Installation/Installation).

Furthermore, `PS2K` and `SYNM` are clearly not your devices - they are PS2 devices (and why you think you would need to use an ELAN driver on a Synaptics trackpad even if it was your trackpad is beyond me). 

Lastly, this is the Github issues page, it is not intended for end user support. It is intended for developer support. As such, I will be closing this issue. If you want support, you should get it in the gitter chat room found [here](https://gitter.im/alexandred/VoodooI2C).

closed 2018-01-17 by alexandred

1and1get2, 2018-01-17:
My apologies, I was only intended to report the compiling error.