VoodooI2C/VoodooI2C #58: elan1200

onejay09, 2018-01-26:
2 finger click/drag only works the first time, could a deadzone for the buttons be added?
and 2 finger right click is intermittent when tapping, it works ok when two finger click is used

alexandred, 2018-01-26:
What do you mean "the first time"?

And the problem with the right click is known, I don't have a solution for it yet.

onejay09, 2018-01-26:
I can drag icons with my second finger while holding the click button down
with my fist finger, but it only works once from login, the cursor won't
move the second time I try.

alexandred, 2018-01-26:
You should come to the Gitter chatroom and upload the archives requested on the Troubleshooting page.

alexandred, 2018-01-26:
Closed as confirmed not a bug. Fixed by following the drag instructions here: https://voodooi2c.github.io/#Supported Gestures/Supported Gestures

closed 2018-01-27 by alexandred