VoodooI2C/VoodooI2C #59: VoodooI2C version 2.x causing kernel bug on High Sierra 10.13.3 while version 1.x no longer working

TomHeaven, 2018-02-15:
Hi, I updated my Hackintosh laptop from 10.13 to 10.13.3 yesterday. While everything seemed flawless, I found my I2C touchpad no longer working.  Previously, I installed VoodooI2C 1.0.4 to S/L/E and everything worked fine. So I updated VoodooI2C to 2.0.1 ( also tried 2.0.0) and this caused kernel bug and immediate reboot during booting process. This happens so fast that I cannot take a clear picture of the bug screen. It's a long screen with with many driver names and their versions.

alexandred, 2018-02-15:
Github issues is for genuine bug reports, not installation issues. Post your problem on the tonymac thread or on Gitter.

closed 2018-02-15 by alexandred

TomHeaven, 2018-02-20:
Well, using VoodooI2C 2.x on 10.13.3 will trigger a kernel panic. I think it's a genuine bug.

TomHeaven, 2018-02-21:
It turned out that I need to recompile DSDT after updating BIOS. I've reverted back to v1.0.4 and it works fine now. I didn't realize that there were new instructions on installing new versions. I'll try following the instructions to test new versions.