VoodooI2C/VoodooI2C #60: intermittent kernel panic

onejay09, 2018-02-21:
every other boot up i get this kp
im unsure why

```
Anonymous UUID:       2073F6BC-80BB-B7DE-8CDE-3D40FCF51A03

Wed Feb 21 20:06:43 2018

*** Panic Report ***
panic(cpu 2 caller 0xffffff800d9fe39d): Kernel trap at 0xffffff800debc8f9, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000023a89000, CR4: 0x00000000003627e0
RAX: 0xffffff8035766980, RBX: 0x0000000000000000, RCX: 0x0000000000000017, RDX: 0xffffff80357a1100
RSP: 0xffffff812241be40, RBP: 0xffffff812241be60, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0xffffff8035865c60, R9:  0xffffff912bd95000, R10: 0x0000000000800000, R11: 0xffffff81220ee058
R12: 0x0000000000800000, R13: 0x0000000000000000, R14: 0xffffff8035766680, R15: 0x0000000000000017
RFL: 0x0000000000010202, RIP: 0xffffff800debc8f9, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 1

Backtrace (CPU 2), Frame : Return Address
0xffffff812241bad0 : 0xffffff800d8e953c 
0xffffff812241bb50 : 0xffffff800d9fe39d 
0xffffff812241bd30 : 0xffffff800d89a593 
0xffffff812241bd50 : 0xffffff800debc8f9 
0xffffff812241be60 : 0xffffff7f8e55f05d 
0xffffff812241beb0 : 0xffffff7f8e55ec58 
0xffffff812241bee0 : 0xffffff800debc46f 
0xffffff812241bf30 : 0xffffff800debacd1 
0xffffff812241bf80 : 0xffffff800deba2b6 
0xffffff812241bfb0 : 0xffffff800d8978f7 
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[125807B8-4E2E-341B-AE46-F06983D562A3]@0xffffff7f8e55d000->0xffffff7f8e56afff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[4F7FB6AD-2498-3F71-827C-ED7AA4BF2511]@0xffffff7f8e54d000

BSD process name corresponding to current thread: kernel_task
Boot args: darkwake=0 kext-dev-mode=1 shikigva=4 

Mac OS version:
16G29

Kernel version:
Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64
Kernel UUID: D3314D98-5D40-3CD8-98A4-F1DD46C20E03
Kernel slide:     0x000000000d600000
Kernel text base: 0xffffff800d800000
__HIB  text base: 0xffffff800d700000
System model name: MacBookPro13,1 (Mac-473D31EABEB93F9B)

System uptime in nanoseconds: 6460094323
last loaded kext at 5504073313: com.apple.filesystems.ntfs	3.13 (addr 0xffffff7f8ef8f000, size 397312)
loaded kexts:
com.alexandred.VoodooI2CHID	1
com.nvidia.web.GeForceWeb	10.1.8
com.nvidia.web.NVDAGP100HalWeb	10.1.8
com.nvidia.web.NVDAResmanWeb	10.1.8
org.emlydinesh.driver.AsusACPIBackLightPanel	1.7
com.intel.driver.EnergyDriver	2.0
org.voodoo.driver.VoodooHDA	2.9.1
com.nvidia.NVDAStartupWeb	10.1.8
com.alexandred.VoodooI2C	2.0.1
com.alexandred.VoodooI2CServices	1
org.coolstar.VoodooGPIO	1.1
com.no-one.BrcmPatchRAM2	2.2.7
com.no-one.BrcmFirmwareStore	2.2.7
org.rehabman.driver.FakePCIID	1.3.11
com.insanelymac.RealtekRTL8111	2.2.1
org.netkas.driver.FakeSMC	1765
as.vit9696.Shiki	2.2.3
com.rehabman.driver.USBInjectAll	0.6.4
as.vit9696.AppleALC	1.2.2
as.lvs1974.NvidiaGraphicsFixup	1.2.2
as.vit9696.Lilu	1.2.1
org.emlydinesh.driver.EAPDFix	2.3
org.rehabman.driver.AppleSmartBatteryManager	1.81.4
com.apple.filesystems.ntfs	3.13
com.apple.filesystems.autofs	3.0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.AGPM	110.23.17
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.AppleHDAHardwareConfigDriver	279.48
com.apple.driver.AppleHDA	279.48
com.apple.driver.AudioAUUC	1.70
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.iokit.IOBluetoothSerialManager	5.0.5f1
com.apple.driver.AppleIntelPCHPMC	1.1
com.apple.driver.AppleLPC	3.1
com.apple.driver.AppleMuxControl	3.14.49
com.apple.driver.AppleOSXWatchdog	1
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHV	1
com.apple.driver.AppleBacklight	170.9.20
com.apple.driver.ACPI_SMC_PlatformPlugin	1.0.0
com.apple.driver.AppleIntelSlowAdaptiveClocking	4.0.0
com.apple.driver.AppleMCCSControl	1.3.4
com.apple.driver.AppleFIVRDriver	4.1.0
com.apple.nvidia.NVDAStartup	10.1.7
com.apple.driver.AppleFileSystemDriver	3.0.1
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0
com.apple.BootCache	40
com.apple.filesystems.hfs.kext	366.70.1
com.apple.iokit.IOAHCIBlockStorage	295.20.1
com.apple.driver.AirPort.Brcm4360	1150.12.1a1
com.apple.driver.AppleAHCIPort	326.60.1
com.apple.driver.AppleACPIButtons	5.0
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleACPIEC	5.0
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	172
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.kext.triggers	1.0
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.5f1
com.apple.iokit.IOBluetoothHostControllerUSBTransport	5.0.5f1
com.apple.iokit.IOBluetoothHostControllerTransport	5.0.5f1
com.apple.driver.DspFuncLib	279.48
com.apple.kext.OSvKernDSPLib	525
com.apple.iokit.IOSurface	159.9
com.apple.iokit.IOBluetoothFamily	5.0.5f1
com.apple.iokit.IOAcceleratorFamily2	311.14
com.apple.driver.AppleSMBusPCI	1.0.14d1
com.apple.driver.AppleGraphicsControl	3.14.49
com.apple.AppleGraphicsDeviceControl	3.14.49
com.apple.driver.AppleSSE	1.0
com.apple.driver.AppleBacklightExpert	1.1.0
com.apple.driver.IOPlatformPluginLegacy	1.0.0
com.apple.driver.X86PlatformPlugin	1.0.0
com.apple.driver.IOPlatformPluginFamily	6.0.0d8
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IONDRVSupport	516.1
com.apple.driver.AppleSMC	3.1.9
com.apple.driver.AppleSMBusController	1.0.18d1
com.apple.driver.AppleHDAController	279.48
com.apple.iokit.IOGraphicsFamily	515.3
com.apple.iokit.IOHDAFamily	279.48
com.apple.iokit.IOAudioFamily	205.15
com.apple.vecLib.kext	1.2.0
com.apple.driver.AppleIntelLpssUARTv1	3.0.60
com.apple.driver.AppleIntelLpssUARTCommon	3.0.60
com.apple.driver.AppleIntelLpssDmac	3.0.60
com.apple.iokit.IOSerialFamily	11
com.apple.iokit.IOSCSIArchitectureModelFamily	394.50.1
com.apple.filesystems.hfs.encodings.kext	1
com.apple.driver.usb.IOUSBHostHIDDevice	1.1
com.apple.driver.usb.cdc	5.0.0
com.apple.driver.usb.networking	5.0.0
com.apple.driver.usb.AppleUSBHostCompositeDevice	1.1
com.apple.driver.AppleXsanScheme	3
com.apple.iokit.IO80211Family	1200.12.2
com.apple.driver.mDNSOffloadUserClient	1.0.1b8
com.apple.iokit.IONetworkingFamily	3.2
com.apple.driver.corecapture	1.0.4
com.apple.iokit.IOAHCIFamily	288
com.apple.driver.AppleIntelLpssGspi	3.0.60
com.apple.driver.usb.AppleUSBXHCIPCI	1.1
com.apple.driver.usb.AppleUSBXHCI	1.1
com.apple.driver.usb.AppleUSBHostPacketFilter	1.0
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.1
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.DiskImages	444.50.16
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	1.0

```

alexandred, 2018-02-21:
Please provide the KP log after booting with the argument "Keep Debug Symbols" in Clover.

onejay09, 2018-02-21:
ok

onejay09, 2018-02-21:
keepsyms=1 ?

onejay09, 2018-02-21:
here you go Mr Doud

onejay09, 2018-02-22:
```
2018-02-22 19:47:07.662254+0000 0x5854     Activity    0x8000000000002dc0   607    kextcache: (Security) SecTrustEvaluateIfNecessary
2018-02-22 19:47:07.662885+0000 0x3ea1     Default     0x8000000000002dc0   134    trustd: [com.apple.securityd.policy] cert[2]: AnchorTrusted =(leaf)[force]> 0
2018-02-22 19:47:07.665633+0000 0x3ea1     Default     0x8000000000002dc0   134    trustd: [com.apple.securityd.rvc] asynchronously fetching CRL (http://crl.apple.com/root.crl) for client (kextcache[607])
2018-02-22 19:47:07.670313+0000 0x5854     Activity    0x8000000000002dc1   607    kextcache: (Security) SecTrustEvaluateIfNecessary
2018-02-22 19:47:07.671073+0000 0x3ea1     Default     0x8000000000002dc1   134    trustd: [com.apple.securityd.rvc] asynchronously fetching CRL (http://crl.apple.com/timestamp.crl) for client (kextcache[607])
2018-02-22 19:47:07.671290+0000 0x3ea1     Default     0x8000000000002dc1   134    trustd: [com.apple.securityd.rvc] asynchronously fetching CRL (http://crl.apple.com/root.crl) for client (kextcache[607])
2018-02-22 19:47:07.672377+0000 0x5854     Activity    0x8000000000002dc2   607    kextcache: (Security) SecTrustEvaluateIfNecessary
2018-02-22 19:47:07.673004+0000 0x3ea1     Default     0x8000000000002dc2   134    trustd: [com.apple.securityd.policy] cert[2]: AnchorTrusted =(leaf)[force]> 0
2018-02-22 19:47:07.679322+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.679339+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.682074+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.682087+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.682116+0000 0x5854     Default     0x0                  607    kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f5230 [0x7fff998a1da0]> { URL = "VoodooI2CHID.kext/ -- file:///System/Library/Extensions/", ID = "com.alexandred.VoodooI2CHID" }
2018-02-22 19:47:07.684023+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.684035+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.685874+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.685887+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.685910+0000 0x5854     Default     0x0                  607    kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f91c0 [0x7fff998a1da0]> { URL = "VoodooI2CServices.kext/ -- file:///System/Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "com.alexandred.VoodooI2CServices" }
2018-02-22 19:47:07.687826+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.687838+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.689962+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.689974+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.689997+0000 0x5854     Default     0x0                  607    kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f7910 [0x7fff998a1da0]> { URL = "VoodooGPIO.kext/ -- file:///System/Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "org.coolstar.VoodooGPIO" }
2018-02-22 19:47:07.692634+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.692646+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.696213+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.696225+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
2018-02-22 19:47:07.696248+0000 0x5854     Default     0x0                  607    kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f2300 [0x7fff998a1da0]> { URL = "VoodooI2C.kext/ -- file:///System/Library/Extensions/", ID = "com.alexandred.VoodooI2C" }
2018-02-22 19:47:07.697233+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67030
2018-02-22 19:47:07.697655+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67030
[... cut ...]
2018-02-22 19:47:07.699522+0000 0x5854     Default     0x0                  607    kextcache: Invalid signature -67030 for kext <OSKext 0x7fbe472f0160 [0x7fff998a1da0]> { URL = "VoodooHDA.kext/ -- file:///System/Library/Extensions/", ID = "org.voodoo.driver.VoodooHDA" }
2018-02-22 19:47:07.699922+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67030
2018-02-22 19:47:07.699978+0000 0x5854     Default     0x0                  607    kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67030
```

not sure if its related but theres this in log

onejay09, 2018-02-22:
`2018-02-22 18:15:50.598157+0000 0x385      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a161 SSCN not implemented in ACPI tables
2018-02-22 18:15:50.598165+0000 0x385      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a161 FMCN not implemented in ACPI tables
2018-02-22 18:15:50.598171+0000 0x385      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Warning: Error getting bus config, using defaults where necessary
2018-02-22 18:15:50.598206+0000 0x385      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Publishing device nubs
2018-02-22 18:15:50.598216+0000 0x385      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Found I2C device: ELAN1200
2018-02-22 18:15:50.598268+0000 0x385      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointH`

```
    If (USTP)
    {
        Scope (_SB.PCI0.I2C1)
        {
            Method (SSCN, 0, NotSerialized)
            {
                Return (PKG3 (SSHI, SSLI, SSDI))
            }

            Method (FMCN, 0, NotSerialized)
            {
                Return (PKG3 (FMHI, FMLI, FMDI))
            }

            Method (FPCN, 0, NotSerialized)
            {
                Return (PKG3 (FPHI, FPLI, FPDI))
            }

            Method (M0D3, 0, NotSerialized)
            {
                Return (PKG1 (M0CI))
            }

            Method (M1D3, 0, NotSerialized)
            {
                Return (PKG1 (M1CI))
            }
        }
```

onejay09, 2018-02-22:
note:
after removing the (UTSP) IF statement the kext finds the changes for pci8086,a161

kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a160 SSCN not implemented in ACPI tables
kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a160 FMCN not implemented in ACPI tables
kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a160 Warning: Error getting bus config, using defaults where necessary
kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a160 Publishing device nubs

kernel: (VoodooI2CServices) class: VoodooI2CControllerNub
kernel: (VoodooI2CServices) class: VoodooI2CControllerDriver
kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Got bus configuration values
kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Publishing device nubs
kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Found I2C device: ELAN1200
kernel: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointH
kernel: (VoodooI2CServices) class: VoodooI2CControllerDriver`

however this didnt fix the Kernel Panic

onejay09, 2018-02-22:
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/1749671/Archive.zip)

alexandred, 2018-02-22:
Can you stop spamming this page with comments? Most of what you have sent isn't helpful, I only needed what I asked of you. I will investigate when I have the time.

onejay09, 2018-03-01:
Boot hang but touchpad still works:

kernel: busy timeout[0], (60s), kextd wait(0): 'X86PlatformPlugin', 'IOHDACodecFunction', 'BCM20702A0', 'ETPD', 'IOResources'

boots but touchpad doesn’t work:

2018-02-28 19:13:23.265834+0000 0x3e6      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a160 SSCN not implemented in ACPI tables
2018-02-28 19:13:23.268749+0000 0x3de      Default     0x0                  0      kernel: (VoodooGPIO) VoodooGPIOSunrisePointH::VoodooGPIO Initializing Community 0
2018-02-28 19:13:23.269130+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Probing controller
2018-02-28 19:13:23.269203+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Found valid Synopsys component, continuing with initialisation
2018-02-28 19:13:23.269228+0000 0x3de      Default     0x0                  0      kernel: (VoodooGPIO) VoodooGPIOSunrisePointH::VoodooGPIO Initializing Community 1
2018-02-28 19:13:23.269433+0000 0x3de      Default     0x0                  0      kernel: (VoodooGPIO) VoodooGPIOSunrisePointH::VoodooGPIO Initializing Community 2
2018-02-28 19:13:23.269767+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Got bus configuration values
2018-02-28 19:13:23.277532+0000 0x117      Default     0x0                  0      kernel: (VoodooGPIO) VoodooGPIOSunrisePointH::GPIO Controller is already awake! Not reinitializing.
2018-02-28 19:13:23.277706+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Publishing device nubs
2018-02-28 19:13:23.280621+0000 0x3d7      Default     0x0                  0      kernel: (BrcmPatchRAM2) BrcmPatchRAM2: Published new IOKit personality for BrcmFirmwareStore.
2018-02-28 19:13:23.281078+0000 0x3ce      Default     0x0                  0      kernel: (VoodooI2CServices) class: IOPCIDevice
2018-02-28 19:13:23.284624+0000 0x3e6      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a160 FMCN not implemented in ACPI tables
2018-02-28 19:13:23.284632+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Found I2C device: ELAN1200
2018-02-28 19:13:23.284871+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::ETPD Could not find GPIO controller, exiting
2018-02-28 19:13:23.284875+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a161 Could not initialise nub for ELAN1200
2018-02-28 19:13:23.285056+0000 0x3e6      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a160 Warning: Error getting bus config, using defaults where necessary
2018-02-28 19:13:23.285110+0000 0x3ff      Default     0x0                  0      kernel: (VoodooI2CServices) class: VoodooI2CControllerNub
2018-02-28 19:13:23.285617+0000 0x418      Default     0x0                  0      kernel: (VoodooI2CServices) class: VoodooI2CControllerDriver
2018-02-28 19:13:23.286141+0000 0x3e6      Default     0x0                  0      kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a160 Publishing device nubs
2018-02-28 19:13:23.286270+0000 0x419      Default     0x0                  0      kernel: (VoodooI2CServices) class: VoodooI2CControllerDriver
2018-02-28 19:13:23.293037+0000 0x3ce      Default     0x0                  0      kernel: (VoodooI2CServices) class: IOPCIDevice
2018-02-28 19:13:23.293061+0000 0x3e6      Default     0x0                  0      kernel: (VoodooI2CServices) class: VoodooI2CControllerNub
2018-02-28 19:13:23.293206+0000 0x3ce      Default     0x0                  0      kernel: (VoodooI2CServices) class: VoodooI2CPCIController
2018-02-28 19:13:23.293402+0000 0x3ce      Default     0x0                  0      kernel: (VoodooI2CServices) class: VoodooI2CPCIController

onejay09, 2018-04-11:
this was while using aptiomemoryfix.efi, it doesnt panic on aptiofixdrv.efi.

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!

xuhadixu, 2018-09-26:
> Please provide the KP log after booting with the argument "Keep Debug Symbols" in Clover. Also provide your troubleshooting archives as requested on the troubleshooting page.
/ Linux equivalent of FTE_i2c_write_cmd function
 IOReturn VoodooI2CFTETouchpadDriver::write_FTE_cmd(UInt16 reg, UInt16 cmd) {
     UInt16 buffer[] {
         reg,
         cmd
     };
     IOReturn retVal = kIOReturnSuccess;
     retVal = api->writeI2C(reinterpret_cast<UInt8*>(&buffer), sizeof(buffer));
     return retVal;
 }
 
 IOReturn VoodooI2CFTETouchpadDriver::message(UInt32 type, IOService* provider, void* argument)
 {
     switch (type)
     {
         case kKeyboardGetTouchStatus:
         {
 #if DEBUG
             IOLog("%s::getEnabledStatus = %s\n", getName(), ignoreall ? "false" : "true");
 #endif

pusolexuze, 2018-09-26:
Автор на вопросы не отвечает а только даны кидает!будьте внимательны

vcreatyv, 2018-09-26:
> Can you stop spamming this page with comments? Most of what you have sent isn't helpful, I only needed what I asked of you. I will investigate when I have the time.
kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f5230 [0x7fff998a1da0]> { URL = "VoodooI2CHID.kext/ -- file:///System/Library/Extensions/", ID = "com.alexandred.VoodooI2CHID" }
kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f91c0 [0x7fff998a1da0]> { URL = "VoodooI2CServices.kext/ -- file:///System/Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "com.alexandred.VoodooI2CServices" }
kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f7910 [0x7fff998a1da0]> { URL = "VoodooGPIO.kext/ -- file:///System/Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "org.coolstar.VoodooGPIO" }
kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67050
kextcache: Invalid signature -67050 for kext <OSKext 0x7fbe472f2300 [0x7fff998a1da0]> { URL = "VoodooI2C.kext/ -- file:///System/Library/Extensions/", ID = "com.alexandred.VoodooI2C" }
kextcache: (Security) [com.apple.securityd.security_exception] MacOS error: -67030
kextcache: Invalid