VoodooI2C/VoodooI2C #63: kernel panic intermittent

zysuper, 2018-03-17:
Anonymous UUID:       8D1A334E-D043-B8C6-6366-2541EB74B352

Sat Mar 17 11:02:45 2018

*** Panic Report ***
panic(cpu 0 caller 0xffffff800156e339): Kernel trap at 0xffffff7f84050c5c, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000016dc4000, CR4: 0x00000000003627e0
RAX: 0x0000000000000000, RBX: 0xffffff8046c0ae30, RCX: 0xffffff8001e721c0, RDX: 0xffffff803190c6f0
RSP: 0xffffff81f002bd80, RBP: 0xffffff81f002bd90, RSI: 0xffffff8001e18c60, RDI: 0x0000000000000000
R8:  0x000000000000ffff, R9:  0x0000000000000001, R10: 0xffffff805a6d9040, R11: 0x0000000000000003
R12: 0xffffff806090f440, R13: 0xffffff8060a1eaf0, R14: 0xffffff8046c0ae30, R15: 0x0000000000000000
RFL: 0x0000000000010246, RIP: 0xffffff7f84050c5c, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff81f002b830 : 0xffffff800144f606 
0xffffff81f002b880 : 0xffffff800157c654 
0xffffff81f002b8c0 : 0xffffff800156e149 
0xffffff81f002b940 : 0xffffff8001401120 
0xffffff81f002b960 : 0xffffff800144f03c 
0xffffff81f002ba90 : 0xffffff800144edbc 
0xffffff81f002baf0 : 0xffffff800156e339 
0xffffff81f002bc70 : 0xffffff8001401120 
0xffffff81f002bc90 : 0xffffff7f84050c5c 
0xffffff81f002bd90 : 0xffffff7f840519aa 
0xffffff81f002bdc0 : 0xffffff7f82123e7f 
0xffffff81f002be20 : 0xffffff7f840527bf 
0xffffff81f002be40 : 0xffffff8001a6a69d 
0xffffff81f002be80 : 0xffffff8001a6a3e1 
0xffffff81f002bf00 : 0xffffff8001a69907 
0xffffff81f002bf50 : 0xffffff8001a6b516 
0xffffff81f002bfa0 : 0xffffff80014004f7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[56274755-65D2-3435-99D5-63101C02DFDF]@0xffffff7f82112000->0xffffff7f8218cfff
         com.alexandred.VoodooI2CHID(1.0)[AC15AC00-AAEC-39C0-BB1A-C23C613F73F3]@0xffffff7f8404e000->0xffffff7f8405dfff
            dependency: com.alexandred.VoodooI2C(2.0.1)[ABF613A5-D745-398B-8567-5A8E406D37E2]@0xffffff7f84022000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[8794C760-FDD9-3664-ADED-4A9BBEC6E517]@0xffffff7f821a3000
            dependency: com.apple.iokit.IOGraphicsFamily(517.22)[2AEA02BF-2A38-3674-A187-E5F610FD65B7]@0xffffff7f82435000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[56274755-65D2-3435-99D5-63101C02DFDF]@0xffffff7f82112000

BSD process name corresponding to current thread: kernel_task
Boot args: -igfxbeta abm_firstpolldelay=10000 -disablegfxfirmware -igfxdvmtbeta darkwake=8 

Mac OS version:
17D102

Kernel version:
Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64
Kernel UUID: 18D901F1-4A03-3FF1-AE34-C26B2732F13C
Kernel slide:     0x0000000001000000
Kernel text base: 0xffffff8001200000
__HIB  text base: 0xffffff8001100000
System model name: MacBookPro14,3 (Mac-551B86E5744E2388)

System uptime in nanoseconds: 7359180928465
last loaded kext at 3053740366427: com.apple.driver.AppleXsanScheme	3 (addr 0xffffff7f8405e000, size 32768)
last unloaded kext at 3438999018544: com.apple.driver.AppleXsanScheme	3 (addr 0xffffff7f8405e000, size 32768)
loaded kexts:
(none)

EOF

alexandred, 2018-03-18:
What am I supposed to do with this? If you have an issue you should go to the gitter chat room and provide troubleshooting archives. Providing a KP and saying nothing is useless.

closed 2018-03-18 by alexandred

zysuper, 2018-03-19:
Sorry, I don't know dump log doesn't help anything.

I apologize for the inconvenience caused.

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!