VoodooI2C/VoodooI2C #64: Some proble I have met

dgtutu, 2018-03-26:
Hello，
My device: Lenovo Mixx510 [i5 7200u]
Both touch screen and touch pens can be used，but It still has some bug：
1、Removing the magnetic keyboard can cause the crash and restart automatically
2、Volume and brightness shortcut key failure
3、Keyboard touch panel failure
4、Shutdown kernel panic

alexandred, 2018-03-26:
The first issue you listed is a known bug with v2.0.1 on devices with removable digitisers and is fixed in existing code (but hasn't yet been released). For that reason, I will leave this issue up.

However, I don't understand the 2nd part and the 3rd part. The 4th part is something that should be debugged in the regular support channels (gitter, tonymac/insanelymac threads) and not in the Github issues (unless an actual VoodooI2C bug is found and is not something related to how you installed it).

Please continue all further discussion in one of the previously mentioned support channels. In particular, you should provide your troubleshooting archives there as requested on the troubleshooting page.

dgtutu, 2018-08-30:
With using VoodooI2c.kext 2.03
Now I have met a new problem，
When I enter the login interface, after entering the password, the mouse will continue to circle, then the black screen will automatically return to the login interface, repeating the same problem, only the drive reinstall system

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!

> With using VoodooI2c.kext 2.03
> Now I have met a new problem，
> When I enter the login interface, after entering the password, the mouse will continue to circle, then the black screen will automatically return to the login interface, repeating the same problem, only the drive reinstall system
because of this driver breaks laptops - be careful!