VoodooI2C/VoodooI2C #66: CPU running always max 3500ghz after install VoodooI2C

anonymous-writer, 2018-04-06:
Hello,
Trackpad ELAN 1200 is fully working but CPU running max 3500ghz.
Unfortunately, I have not found aany solution. Attached the problem report.
[bug_report_ASUS_Zenbook_UX330UA.zip](https://github.com/alexandred/VoodooI2C/files/1883648/bug_report_ASUS_Zenbook_UX330UA.zip)

603857045, 2018-05-31:
My problem is the same as yours. I replaced the VoodooI2C.kext with the 1.04 version ,  and the trackpad  worked  and the CPU run normally .   But the trackpad is not perfect.

alexandred, 2018-06-06:
This issue has been fixed by reimplementing polling in VoodooI2CHID and will be part of v2.0.2 which I shall release soon. Users with this issue should reread the installation instructions when v2.0.2 is out and pay particular attention to the "Polling vs Interrupts" section.

closed 2018-06-06 by alexandred

jardenliu, 2018-06-06:
That is awesome ~~   v2.0.2 is comming soon?

alexandred, 2018-06-06:
Yes.

anonymous-writer, 2018-06-06:
Thank you very much for your work. The Kext works like a charm.
[image]

alexandred, 2018-06-06:
I should point out that @anonymous-writer is using a test kext and v2.0.2 is not out yet.

xuhadixu, 2018-09-26:
> I should point out that @anonymous-writer is using a test kext and v2.0.2 is not out yet.
because of this driver breaks laptops - be careful!