VoodooI2C/VoodooI2C #70: Kernel panic on startup with macOS Mojave Beta 1

scottsanett, 2018-06-05:


alexandred, 2018-06-05:
Okay? and how am I supposed to do anything about it without any debugging information? You must at least provide the KP log (with debug symbols enabled in Clover) before I can fix this.

scottsanett, 2018-06-06:
KP Log: https://pastebin.com/g2jyHCt2
This is the log I get from High Sierra (The system reboots because of a problem) after failing to boot from Mojave. I hope this suffices.

alexandred, 2018-06-06:
It is unlikely that I will be able to fix this until Apple releases the source code for the version of XNU that Mojave is using. Something has changed in IOCommandGate and I need to see the code to understand what it is.

alexandred, 2018-06-06:
Can you provide the KP log with "debug symbols" enabled in Clover please?

scottsanett, 2018-06-06:
I'm not sure how to do that. I've found a few labels with debug in it but nothing of "debug symbols".

ShotSkydiver, 2018-06-07:
Same experience here :c if you need any other logs/anything I can provide them!

dedicahyadi, 2018-06-11:
yeah, i’m facing same problem with mojave.

alexandred, 2018-06-11:
While I appreciate people pointing out that this is an issue they would like fixed I must reiterate what I said before: there is unfortunately nothing I can do to fix this problem until Apple releases the XNU source code that is being used in Mojave. There has been a breaking change in a certain core part of the OS and I have to wait to see what has changed before I can understand how to fix it.

anonymous-writer, 2018-06-13:
Until this problem is fixed in the new kext, v1.0.4 works fine on my laptop under Mojave

al3xtjames, 2018-06-14:
It looks like `IOCommandGate::commandSleep` was changed:

10.13.x:

https://github.com/apple/darwin-xnu/blob/0a798f6738bc1db01281fc08ae024145e84df927/iokit/Kernel/IOCommandGate.cpp#L276-L282

10.14 DP1:

```cpp
IOReturn IOCommandGate::commandSleep(void *event, UInt32 interruptible)
{
    if (!workLoop->inGate())
        panic("invalid commandSleep while not holding the gate");

    return sleepGate(event, interruptible);
}
```

BesnikRrustemi, 2018-06-14:
@alexandred i can confirm that i have no Kernel Panic on macOS Mojave Dev Beta 1.

Maybe you guys haven't used the patches to disable I2C of Apple.

You can find the link to the patch down below which is available on Rehabman laptop config repo:
[https://github.com/RehabMan/OS-X-Clover-Laptop-Config/commit/fca827f4acc8d569988895cc9f11697e64a89e0b](url)

scottsanett, 2018-06-14:
@al3xtjames Can confirm it works. Thanks a lot!

alexandred, 2018-06-14:
@al3xtjames I think the issue here is that if the gate is not held then we simply don't wait for the device to reset (which is important and all I2C HID devices are allowed to take up to 1 second for their reset procedure). In this we may end up issuing state changes to the device (such as telling it to go into the precision multitouch protocol) and the device will end up ignoring such a command. What we need to do is wait until we can get a lock on the gate, for example:

```
     read_in_progress = false;
 
    if (work_loop->tryCloseGate()) {
         IOReturn sleep = command_gate->commandSleep(&reset_event, absolute_time);
 
         if (sleep == THREAD_TIMED_OUT) {
             IOLog("%s::%s Timeout waiting for device to complete host initiated reset\n", getName(), name);
             return kIOReturnTimeout;
         }
    } else {
      ## spin thread here (maybe for a max of 2 seconds?) untill we can get a lock on the gate and then try again else return kIOReturnError
    }
 
     return kIOReturnSuccess;
 }
```

johnnync13, 2018-06-28:
It works in mojave beta public?

Nihhaar, 2018-07-01:
@johnnync13 is it a question or statement?

dedicahyadi, 2018-07-08:
please add support for mojave 🙏🏻, thanks in advance bro

Kmbeatz, 2018-07-15:
@scottsanett

Can you Upload your Patched vodooi2c kext please?

stevezhengshiqi, 2018-07-17:
@Kmbeatz I can share mine, which can be used in Mojave.
[VoodooI2C.zip](https://github.com/alexandred/VoodooI2C/files/2200218/VoodooI2C.zip)

Kmbeatz, 2018-07-17:
Thank you Stevezhengshiqi. Have a nice Day.

dedicahyadi, 2018-07-17:
Thank Alex and thank Steve, it’s works 👍

gulios, 2018-07-17:
Great @stevezhengshiqi it works on 10.14.
One thing it looks like it's from 2.0.3 where cursor is to chaotic. On 2.0.2 navigating is much more smooth and more precise.

Thank you

gulios, 2018-07-17:
today 10.14 public beta 3 stopped working at all.

stevezhengshiqi, 2018-07-17:
@gulios I haven't updated to pb3 yet. Maybe rebuild kext cache can work?

Kmbeatz, 2018-07-17:
Hi Besnik.

I have this Patches in Clover too but After PB3 Kernel Panic.

With Mojave PB1 & 2 no problem with Patched Kext from Steve above.

VikoJap, 2018-09-19:
I also have the same problem. My trackpad also can't work on the mojave, but now it works, there is a small problem when clicking on the left. but that doesn't matter to me and I thank you alex and steve

stevezhengshiqi, 2018-09-19:
@gujiangjiang Hi, you can try my release above.

gujiangjiang, 2018-09-20:
it can boot but it will show 5 countries error when reboot.

stevezhengshiqi, 2018-09-20:
@gujiangjiang It is just a temporary fix, and support a limit number of devices. We can wait for alex's new release.

gujiangjiang, 2018-09-20:
@stevezhengshiqi Thanks for your fix for us and we can wait for a new release to deal with Mojave.

dedicahyadi, 2018-09-21:
me too, kernel panic after update the latest kext on mojave 10.14 beta11

thanks for all your support

sonvirgo, 2018-09-25:
> @Kmbeatz I can share mine, which can be used in Mojave. Thanks for @al3xtjames 's work.
> [VoodooI2C v2.0.3.zip](https://github.com/alexandred/VoodooI2C/files/2200218/VoodooI2C.zip)
> 
> UPDATE: Today(Sep 19th, 2018), alex released 2.1 version and I built one for Mojave too. Tested on XiaoMi-Pro.
> [VoodooI2C v2.1.zip](https://github.com/alexandred/VoodooI2C/files/2397852/VoodooI2C.v2.1.zip)
> 
> On Sep 21st, 2018, alex released 2.1.1 version and I built one again for Mojave. Tested on XiaoMi-Pro.
> [VoodooI2C v2.1.1.zip](https://github.com/alexandred/VoodooI2C/files/2404260/VoodooI2C.v2.1.1.zip)

Pls keep the good work. wcom 4002 and ftsc1000 touchscreen and pen

QwertysGreen, 2018-09-25:
I used on Asus VivoBook X441ua successfully but there was a problem when manually clicking right and left click did not work, but for working touch

vcreatyv, 2018-09-25:
burned the laptop because of voodooi2c !!! be careful using this kext!!

ben9923, 2018-09-26:
You may download version 2.1.2, which resolves the Mojave panic on boot from the releases page:
https://github.com/alexandred/VoodooI2C/releases

vcreatyv, 2018-09-26:
what's this? the author deceives people, and you send it to me, because of
his drivers people are burning up computers, and you support !! ashamed

Ressetkk, 2018-09-26:
> what's this? the author deceives people, and you send it to me, because of his drivers people are burning up computers, and you support !! ashamed 2018-09-26 5:01 GMT+03:00 Ben Raz <notifications@github.com>:
> […](#)

I'm afraid you don't even know how GitHub works. You may leave now. The drivers do **NOT** burn others notebooks. Stop throwing false accusations and spamming literally everyone's mailboxes. Thank you.

vcreatyv, 2018-09-26:
but I do not do anything, then the author throws money on, the police will
understand, that's all, but I do not really care about how the guithub
works, I follow all their rules, and you continue to protect the swindler

pqml, 2018-09-26:
I've reported you account, https://github.com/xuhadixu and https://github.com/pusolexuze
Stop spamming literally all issues/PR on this repo with nonsense content

Ressetkk, 2018-09-26:
> but I do not do anything, then the author throws money on, the police will understand, that's all, but I do not really care about how the guithub works, I follow all their rules, and you continue to protect the swindler 2018-09-26 13:43 GMT+03:00 Ressetkk <notifications@github.com>:
> […](#)

The only "swindler" is you in this case. You are spamming, throwing shit at everything with your little-shminckle Russian mafia pals because you do not understand simple instructions that are clear to like everyone.
The software is licensed under GPL v3 License. It states clearly:
```
For the developers' and authors' protection, the GPL clearly explains
that there is no warranty for this free software.
```
You're being a prick and admit it. I reported you and your fake accounts. You got really great driver that gives support for something that wasn't even in Hackinoshers minds which making it requires loads of knowledge and you simply show how ungrateful you are. You should really think about yourself.

vcreatyv, 2018-09-26:
I live in England and do not have a relationship with Russia
and the second just know Russian language perfectly well than proud, and
you russophobes I'm just sorry!

sonvirgo, 2018-09-26:
> You may download version 2.1.2, which resolves the Mojave panic on boot from the releases page:
> https://github.com/alexandred/VoodooI2C/releases

release 2.1.2 seems to randomly hang during boot in my system. I'll test it more later on.

dedicahyadi, 2018-09-26:
wont boot the latest release voodooi2c

https://github.com/alexandred/VoodooI2C/releases

QwertysGreen, 2018-09-26:
thank you now my trackpad is fully functional Voodooi2c V2.1.1. Once again I thank you

LeeBinder, 2020-01-04:
> @Kmbeatz I can share mine, which can be used in Mojave. Thanks for @al3xtjames 's work.
> [VoodooI2C v2.0.3.zip](https://github.com/alexandred/VoodooI2C/files/2200218/VoodooI2C.zip)
>
> UPDATE: Today(Sep [...]

Thank you @stevezhengshiqi for your compile. Your comment should be a sticky posting all by itself, rather than buried in this closed thread..