VoodooI2C/VoodooI2C #86: Mouse drift problem

zzyandzzy, 2018-08-13:
Hello, my problem is the same as [issues/44](https://github.com/alexandred/VoodooI2C/issues/44) After I used another driver [MacForceOne/VoodooI2C](https://github.com/MacForceOne/VoodooI2C) the mouse no longer drifts. The gestures are perfect (five fingers will cause to restart), how can I fix it? Thank you.

zzyandzzy, 2018-08-13:
By the way, **MacForceOne/VoodooI2C** can cause hibernation instability and **kernel panic**

alexandred, 2018-08-13:
Without troubleshooting archives I can only guess. If you are using the latest release then it is because you are running in polling mode and your device does not support it. You should run it in interrupt mode.

Macforceone's branch is old and should nto be used.

closed 2018-08-13 by zzyandzzy