VoodooI2C/VoodooI2C #87: Trackpad for ASUS ZenBook UX300UAR Broken in v2.0.0

AwlsomeAlex, 2018-08-26:
I am using a ASUS ZenBook UX330UAR [8th Generation i5] (I don't know what trackpad is used) but when I use version 1.0.4 of the kext, it works **BUT** it is close to unusable since if you drag it will go on infinitely, and clicking is very glitchy. However starting with 2.0.0 it just doesn't work at all. Hoping to find a fix for this, or is there a possible DSDT patch for this? Thanks!

alexandred, 2018-08-26:
Such questions should be asked in the Gitter chat, not on Github issues.

closed 2018-08-26 by alexandred

AwlsomeAlex, 2018-08-28:
But it’s an issue. With your kext. Who else am I suppose to ask but the developer?

ben9923, 2018-08-28:
You aren't describing an issue with the kext.
If you complete patching and correct setup for it, and you still have an issue (For example - kext prevents sleep on your machine, kext is causing a panic on shutdown...), then you should open an appropriate GitHub issue.
Please, ask for help in the Gitter chat room, as advised. You should also probably read the installation guide.

xuhadixu, 2018-09-26:
> Such questions should be asked in the Gitter chat, not on Github issues.
because of this driver breaks laptops - be careful!