VoodooI2C/VoodooI2C #89: Trackpad only activating after sleep.

Subscor, 2018-09-19:
Hi, I've been having a very strange problem.

After reading this thread, https://github.com/alexandred/VoodooI2C/issues/70 I was able to finally get VoodooI2C working in my mojave hackintosh system for the internal trackpad (the touchscreen worked out of the box).

However, my internal trackpad only starts working after I force my laptop to sleep and it wakes.

Here's all my debug files including clover DSDT folder
[debug_9170.zip](https://github.com/alexandred/VoodooI2C/files/2395876/debug_9170.zip)

alexandred, 2018-09-25:
In the `_INI` method of your `TPD0` in the DSDT, try replacing the following:

```
                If (LEqual (SDM0, Zero))
                {
                    SHPO (GPDI, One)
                }
```
with this:
```
                    SHPO (GPDI, One)
```

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!

yoasif, 2018-10-04:
Not sure if this is related, but I opened #117 where the trackpad begins to respond after having stopped responding. However, my trackpad works as soon as the machine boots.

Subscor, 2018-11-02:
HI, I'm really sorry for the delay in responding, I've been busy at work.

I modified the _INI method of my TPD0 in my DSDT as suggested but no change, I still have to manually sleep/wake the laptop for the trackpad to work.

I tried with your latest releases (2.1.3 and 2.1.4) to see if I could get native gestures but I had no luck making them work even after a sleep cycle, when using those, I get a TPD0(busy)_timeout(1)...
on power up. The laptop eventually starts but the drivers are not loaded for TPD0.

Any help would be greatly appreciated. 

Heres my new debug files with a "working" trackpad using the kexts from post #70

let me know if you need them with your new kexts too.
[debug_30700.zip](https://github.com/alexandred/VoodooI2C/files/2543907/debug_30700.zip)

alexandred, 2019-01-14:
Do you still have this issue with the latest upstream?

ben9923, 2019-10-25:
Closing as no further information was provided by issue opener.
Will reopen if needed.

closed 2019-10-25 by ben9923

fredwilliamtjr, 2020-01-27:
Hello guys, good night, I am part of the Olalira community in Brazil, with the help of MaLd0n, we set up the EFI of my notebook. it is an Asus VivoBook X512FJ-EJ228T, it all works except the tackpad that is an ELAN1200 model, it only works if I put the computer to sleep, when he wakes up the trackpad is working normally. exactly like this issue.

AndreAbascal, 2020-01-27:
Hello. I am also a part of the Brazilian Hackintosh community and I also own an ASUS VivoBook X512FJ (ASUS VivoBook X512FJ-EJ226T). @fredwilliamtjr has the i7 version and uses Olarila, I have the I5 version and using pure vanilla. All other specs are equal, as far as I know. 

My trackpad, also, only works if I put the computer to sleep. After waking up from sleep, the trackpad starts to work. I am using latest version: 2.3

Think you can help us both?

Thank you for your patience,
Andre.

heberdasilva, 2020-01-27:
Hello guys, good morning! I am also part of the Hackintosh Brazil community, and with the help of @fredwilliamtjr  I managed to install Catalina 10.15.2 on my laptop, an Asus VivoBook X512FJ-EJ228T. Exactly like his, the trackpad only starts to function after putting the system to sleep, and when he wakes up, the trackpad is working normally.
We would be very grateful if anyone knows how to help us!
Thanks!

raspberryenvoie, 2020-02-20:
Hello everybody !
I had the same issue with an ASUS X510 with an FTE 1200 touchpad. After struggling for 2 days I randomly found this solution: I removed VoodooI2C.kext from L/E, I rebuilt  KextCache and Repaired permissions and I only kept it in EFI/CLOVER/KEXT/OTHER. 
Now it works like a charm ! Let me know if this also works for you.

AndreAbascal, 2020-02-20:
@raspberryenvoie Are you using Catalina (10.15.x)? Would you mind posting your DSDT and a list of all kexts you have in EFI/CLOVER/kexts/Other ? Thank you for the help.

raspberryenvoie, 2020-02-20:
Hi AndreAbascal! 
Thanks for your answer.
I'm using Catalina 10.5.3 and I followed this guide to get macOS running on my  laptop https://github.com/tctien342/Asus-Vivobook-S510UA-Hackintosh

I changed few things to the provided configuration.
1. I disabled  SSDT-I2C1_USTP.aml because I'm using an FTE1200 trackpad.
2. I updated VoodooI2C.kext to the lastest version.
3. I did not install the kexts in the L/E folder. I only installed CodecCommander.kext in L/E.

I hope this could help you.

fillipeags, 2020-04-05:
Does anyone with the Asus vivbook x512FJ managed to make the trackpad works properly without having to put to sleep ?

ben9923, 2020-04-11:
All please join the Gitter channel (See the README) and ask for help, with a troubleshooting archive attached.

mollicksj, 2021-01-28:
> Hello. I am also a part of the Brazilian Hackintosh community and I also own an ASUS VivoBook X512FJ (ASUS VivoBook X512FJ-EJ226T). @fredwilliamtjr has the i7 version and uses Olarila, I have the I5 version and I am using pure vanilla, built from scratch, version. All other specs are equal, as far as I know.
> 
> My trackpad, also, only works if I put the computer to sleep. After waking up from sleep, the trackpad starts to work. I am using latest version: 2.3
> 
> Think you can help us both?
> 
> Thank you for your patience,
> Andre.
 I HAVE SAME PROBLEM WITH ASUS VIVO BOOK X407UA 
CORE I 5 6TH GENARATION
TRACKPAD HID COMPLAINT

MOJAVE 10.14 NIRESH WITH CLOVER
CATALINA 10.15.7 NIRESH AND OLARILA WITH CLOVER AND OPEN CORE
NOW USEING BIG SUR OLARILA 11.2 WITH CLOVER AND OPEN CORE
TRACKPAD WORKS GREAT EVEN FOUR FINGER GESTURE
THE PROBLEM IS TOUCHPAD WORKS BUT AFTER SLEEP NOT FROM THE BEGGING THE SYSTEM BOOT UP .
THANKS IN ADVANCE IF ANYONE ALL READY FIX THE PROBLEM OR KNOW HOW TO FIX IT ...
OR ANY TUTORIAL VIDEO IS MORE APPRECIATE...

kprinssu, 2021-01-28:
@mollicksj Spamming on a closed and solved issue is not cool. I have blocked you from commenting on any other VoodooI2C tickets.

 _**Please read and follow the troubleshooting guide**_.