VoodooI2C/VoodooI2C #90: hidd CPU Usage

tboock, 2018-09-21:
First off: Thank you very much for your work. With latest releases the touchpad of my Dell XPS 13 works just perfect.

However I'm experiencing hidd to cause relatively high CPU usage with VoodooI2C 2.1 and 2.1.1 when the mouse cursor is being moved. With VoodooI2C 2.03 hidd CPU usage was around 5% for me. With latest releases it went up to 20%. Due to the CPU clock rate being raised, battery drain is caused.

alexandred, 2018-09-24:
I am currently still investigating this. It turns out that kernel_task had a heavy CPU usage on v2.0.3 as well (upwards of 17%) so I am attempting to fix that first. I am not quite sure what the hidd cpu usage but we will get to it eventually.

tboock, 2018-09-25:
I'm moving to another city and I'm pretty stuck in work. Will provide a troubleshooting archive when I finished some things. Thanks for having a look at this issue!

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!

xuhadixu, 2018-09-26:
> I am currently still investigating this. It turns out that kernel_task had a heavy CPU usage on v2.0.3 as well (upwards of 17%) so I am attempting to fix that first. I am not quite sure what the hidd cpu usage but we will get to it eventually.
/ Linux equivalent of FTE_i2c_write_cmd function
 IOReturn VoodooI2CFTETouchpadDriver::write_FTE_cmd(UInt16 reg, UInt16 cmd) {
     UInt16 buffer[] {
         reg,
         cmd
     };
     IOReturn retVal = kIOReturnSuccess;
     retVal = api->writeI2C(reinterpret_cast<UInt8*>(&buffer), sizeof(buffer));
     return retVal;
 }
 
 IOReturn VoodooI2CFTETouchpadDriver::message(UInt32 type, IOService* provider, void* argument)
 {
     switch (type)
     {
         case kKeyboardGetTouchStatus:
         {
 #if DEBUG
             IOLog("%s::getEnabledStatus = %s\n", getName(), ignoreall ? "false" : "true");
 #endif

vcreatyv, 2018-09-26:
> Can you please provide a troubleshooting archive.
However I'm experiencing hidd to cause relatively high CPU usage with VoodooI2C 2.1 and 2.1.1 when the mouse cursor is being moved. With VoodooI2C 2.03 hidd CPU usage was around 5% for me. With latest releases it went up to 20%. Due to the CPU clock rate being raised, battery drain is caused.

ben9923, 2018-09-27:
This issue is fixed in v2.1.2+
You may get latest release (currently 2.1.4) from the [releases page](https://github.com/alexandred/VoodooI2C/releases).

closed 2018-09-27 by ben9923