VoodooI2C/VoodooI2C #95: 八代设备触控驱动问题

chuntaojun, 2018-09-24:
Hello, when I refer to your tutorial, I found that the touch driver code has not been seen for a while. I have never seen it with reference to other people. The whole tutorial looks like only the code here is different.

[image]

alexandred, 2018-09-24:
`I2CM (I2CX, BADR, SPED)` is effectively `SBFB`. So just replace `SBFI` with `SBFG` while GPIO pinning.

chuntaojun, 2018-09-24:
Ok, I'll try it.  thank you

chuntaojun, 2018-09-25:
> `I2CM (I2CX, BADR, SPED)` is effectively `SBFB`. So just replace `SBFI` with `SBFG` while GPIO pinning.

What frustrates me is that it still doesn't work.

alexandred, 2018-09-25:
You should seek further help on the Gitter chat instead of here. Please visit the chat and provide your full troubleshooting archives.

chuntaojun, 2018-09-26:
> You should seek further help on the Gitter chat instead of here. Please visit the chat and provide your full troubleshooting archives.

I temporarily solved the problem with this trackpad.

xuhadixu, 2018-09-26:
хорошо что комп не сгорел,фейковый драйвер

xuhadixu, 2018-09-26:
> You should seek further help on the Gitter chat instead of here. Please visit the chat and provide your full troubleshooting archives.
/ Linux equivalent of FTE_i2c_write_cmd function
 IOReturn VoodooI2CFTETouchpadDriver::write_FTE_cmd(UInt16 reg, UInt16 cmd) {
     UInt16 buffer[] {
         reg,
         cmd
     };
     IOReturn retVal = kIOReturnSuccess;
     retVal = api->writeI2C(reinterpret_cast<UInt8*>(&buffer), sizeof(buffer));
     return retVal;
 }
 
 IOReturn VoodooI2CFTETouchpadDriver::message(UInt32 type, IOService* provider, void* argument)
 {
     switch (type)
     {
         case kKeyboardGetTouchStatus:
         {
 #if DEBUG
             IOLog("%s::getEnabledStatus = %s\n", getName(), ignoreall ? "false" : "true");
 #endif

closed 2018-09-27 by ben9923