VoodooI2C/VoodooI2C #96: Duel monitor

grantbi, 2018-09-24:
I have a 9560 setup with a duel monitor. When I use 2.1.1 drivers and touch my laptop screen, it moves the mouse relative to my second screen if my mouse is left within the 2nd screen display.

Expected to have it move my mouse to the touch screen where I'm touching.

alexandred, 2018-09-25:
Thanks for pointing this out - this is something we hadn't previously considered and is probably something that is quite tricky to fix. I'll have a think about it.

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!

xuhadixu, 2018-09-26:
> I have a Dell 9560 setup with a duel monitor. When I use 2.1.1 drivers and touch my laptop screen, it moves the mouse relative to my second screen if my mouse is left within the 2nd screen display.
> 
> Expected to have it move my mouse to the touch screen where I'm touching.

отлично!

xuhadixu, 2018-09-26:
> Thanks for pointing this out - this is something we hadn't previously considered and is probably something that is quite tricky to fix. I'll have a think about it.
/ Linux equivalent of FTE_i2c_write_cmd function
 IOReturn VoodooI2CFTETouchpadDriver::write_FTE_cmd(UInt16 reg, UInt16 cmd) {
     UInt16 buffer[] {
         reg,
         cmd
     };
     IOReturn retVal = kIOReturnSuccess;
     retVal = api->writeI2C(reinterpret_cast<UInt8*>(&buffer), sizeof(buffer));
     return retVal;
 }
 
 IOReturn VoodooI2CFTETouchpadDriver::message(UInt32 type, IOService* provider, void* argument)
 {
     switch (type)
     {
         case kKeyboardGetTouchStatus:
         {
 #if DEBUG
             IOLog("%s::getEnabledStatus = %s\n", getName(), ignoreall ? "false" : "true");
 #endif

xuhadixu, 2018-09-26:
> Thanks for pointing this out - this is something we hadn't previously considered and is probably something that is quite tricky to fix. I'll have a think about it.
define super IOService
OSDefineMetaClassAndStructors(VoodooI2CController, IOService);

void VoodooI2CController::free() {
    IOFree(physical_device, sizeof(VoodooI2CControllerPhysicalDevice));

    super::free();
}

bool VoodooI2CController::init(OSDictionary* properties) {
    if (!super::init(properties)) {
        if (debug_logging)
            IOLog("%s super::init failed\n", getName());
        return false;
    }

    physical_device = reinterpret_cast<VoodooI2CControllerPhysicalDevice*>(IOMalloc(sizeof(VoodooI2CControllerPhysicalDevice)));
    physical_device->awake = true;

    return true;

vcreatyv, 2018-09-26:
https://www.youtube.com/watch?v=SVlP4mW2ERE

vcreatyv, 2018-09-26:
> Thanks for pointing this out - this is something we hadn't previously considered and is probably something that is quite tricky to fix. I'll have a think about it.
interesting Аpple approves this driver, it is necessary to ask

gongzhenjeef, 2018-09-26:
Also have this issue (?) on my touchscreen. But on the other hand, I'm able to control 2nd screen with finger when the cursor's there. If you decide to implement this feature, can you add something like a boolean in config.plist as a switch?

BTW, the gitter chat seems inaccessible with a 404?

grantbi, 2018-09-26:
I would imagine that the trackpad would still work on the second screen, however on the touch screen you select - it would move to that one. It would be strange to touch the screen and use as a "trackpad" for another.

Another part to this may be if the second screen is a touch screen. :-/ I don't have one to test this with.

hacker1024, 2019-06-23:
Any progress on this?

hacker1024, 2020-07-06:
I guess not

kprinssu, 2020-07-06:
@hacker1024 Please do not spam the Github issue as your comment was not constructive nor additive to the conversation.