VoodooI2C/VoodooI2C #105: With VoodooI2C 2.1.2 and 2.1.3 have problem in sleep and wake.

gujiangjiang, 2018-09-26:


asoco, 2018-09-26:
Aspire A517-51G-532B, 2.1.3
Confirm for sleep problems, right click with to fingers work perfectly, notification center gesture not wokring

anonymous-writer, 2018-09-26:
Same problem on my Laptop. No Trackpad after sleep and then seconds later black screen with kernel panic.

alexandred, 2018-09-27:
Sleep issues should be fixed in v2.1.4 . Right click with right corner and two fingers is broken for the moment due to scaling - same reason notification centre gesture is broken. A fix for this will come in a future release when we figure out the scaling issues. I will therefore close this issue since the title refers to sleep.

closed 2018-09-27 by alexandred

gujiangjiang, 2018-09-27:
With latest VoodooI2C 2.1.4 it has fixed sleep issues.
But scaling issues was still as before.

anonymous-writer, 2018-09-27:
Hi alexandred,
Now works great for me too. Thank you for your outstanding work.

LeeBinder, 2018-09-28:
Alexandre, @hieplpvip has found a work-around for the scaling issue in his fork, see [https://github.com/hieplpvip/VoodooI2C/commit/0a8208f992a364fe931752299c9625fa9280f50a](https://github.com/hieplpvip/VoodooI2C). It's just that he uses a smaller zone for his Zenbook trackpad than the one in your 2.0.3 driver.

It would be awesome if you could use his work-around but with the values from your 2.0.3 version.

hieplpvip, 2018-09-28:
@LeeBinder My fork of VoodooI2C is made for my laptop, nothing guarantee that it will work on all laptops. It's also the reason why Alexandre hasn't fixed the scaling issue, as he hasn't found a universal solution

gujiangjiang, 2018-09-28:
@hieplpvip 
Tested your build with my XPS15 and the **[Open Notification Center]** works perfectly.

However,the **[Secondary click (right-click) Click or tap with two fingers]** still unreliable and sometime with no response.

LeeBinder, 2018-09-28:
> @LeeBinder My fork of VoodooI2C is made for my laptop, nothing guarantee that it will work on all laptops. It's also the reason why Alexandre hasn't fixed the scaling issue, as he hasn't found a universal solution

@hieplpvip To keep discussions here limited to Alexandre's release I was going to reply in your github account issues section, but that's deactivated, so I'll reply to you directly.