VoodooI2C/VoodooI2C #98: Unresponsive trackpad on XPS15 9560 (mojave and voodooi2c 2.1.2)

yonient, 2018-09-26:
Hi guys,

First thank you for your great work! You have achieved a very huge success.

I got a XPS15 9560 and I am trying to enable the I2C trackpad on it. However, after patching the I2C device by SSDT hotpatch, the trackpad is unresponsive at all. The cursor will not move. Even though I can see the trackpad settings in the SysPre. Very weird. Could someone help me find the solution?

Attached is the generated debug file in case someone need it.

Thanks.

PS: I'm using the macos 10.14 18A391 with voodooi2c 2.1.2. The trackpad is under I2C1.TPD1 as shown in Windows 10.
[debug_29252.zip](https://github.com/alexandred/VoodooI2C/files/2418112/debug_29252.zip)

xuhadixu, 2018-09-26:
because of this driver breaks laptops - be careful!

vcreatyv, 2018-09-26:
If your hexadecimal pin number is greater than `0x2F` then proceed to the next step.
 - If IOInterruptSpecifiers is not present then you have completed this guide and you can continue with installing the kexts

 ##### Step 2a: Ensuring your device is GPIO-pinned
 - If IOInterruptSpecifiers is present, then expand this property and note the first two hexadecimal digits of the value shown. This is your APIC Pin number in hexadecimal notation. Note this down as 0xyy, where yy is replaced by these first two digits.

 You have arrived at this step because Apple's drivers do not support APIC pins greater than `0x2F` (and it would be very difficult to make them support them). In this case, we thus make use of a kext called [VoodooGPIO](https://github.com/coolstar/VoodooGPIO) which comes bundled with all copies of VoodooI2C. VoodooGPIO allows us to get around this limitation of macOS by using GPIO interrupts instead which most new machines support.
 If your APIC Pin Number is less than or equal to 0x2f, then you have completed this guide and you can proceed with the installation of the kexts.

 We must first determine whether or not your device is properly configured to support GPIO pins. Search for your device ACPI ID in your DSDT until you reach its device entry. Look for a `Name` that looks like this:
 If your APIC Pin Number is outside the range 05x-0x77, then VoodooI2C is not the correct driver for your touchpad. You are advised to seek support in the relevant forums or usergroups.

 If your APIC Pin Number is between 0x5c and 0x77, then calculate the corresponding GPIO Pin Number as follows (you can use Calculator.app for hexadecimal conversions and calculations):

vcreatyv, 2018-09-26:
[image]

vcreatyv, 2018-09-26:
https://www.youtube.com/watch?v=SVlP4mW2ERE

vcreatyv, 2018-09-26:
interesting Аpple approves this driver, it is necessary to ask

yonient, 2018-09-26:
I have fixed it by myself. I forgot to replace the XOSI with the one that simulates windows10 (windows2015). After replacing the latest one, my trackpad works well with voodooi2c 2.1.3.

Thanks guys for your great work.

closed 2018-09-26 by yonient