VoodooI2C/VoodooI2C #107: Four fingers kneading will lead to restart.

QiuChenly, 2018-09-27:
My laptop has an ELAN 1200 touch pad, and I'm using the latest Release version of the I2C driver, but I found the system crashed and restarted when I tried to open Launcher with a four-finger pinch.
If you need detailed information about my equipment, please do let me know.

alexandred, 2018-09-27:
@QiuChenly Please provide the kernel panic log with "debug symbols" enabled in Clover (this is `keepsyms=1` as a boot argument).

QiuChenly, 2018-09-27:
I am very glad to see your reply.
I'm sorry I added this startup symbol in the startup parameter, but when I started again after repeated crashes, I couldn't find the panic log from the Library / Logs directory. Did I do something wrong? Please let me know.

alexandred, 2018-09-27:
You need to have NVRAM working for the panic logs to appear. They will become visible in a popup window when the PC reboots.

QiuChenly, 2018-09-27:
Do you mean adding "-v" to boot parameters?
Forgive me for not understanding how to make "NVRAM work"?

QiuChenly, 2018-09-27:
Thanks for helping, I got the crash log successfully.
But this log was caught by a sudden automatic reboot during the boot process of the system crash automatic reboot. I restarted it again before I could switch on properly.

-----------------------------    logs  --------------------------------

Anonymous UUID:       13F830B3-4799-F0BD-F36A-32E608BAEE8C

Fri Sep 28 01:50:13 2018

*** Panic Report ***
panic(cpu 4 caller 0xffffff80142d87bd): Kernel trap at 0xffffff80142d15b3, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000008, CR3: 0x0000000029984000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff7f9777bf74, RCX: 0x0000000000000000, RDX: 0xffffff805397e800
RSP: 0xffffff82e097bd58, RBP: 0xffffff82e097be00, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0xffffff805397e800, R9:  0xffffff8014c2da48, R10: 0xffffff8014a9e350, R11: 0xffffff8014a9af58
R12: 0xffffff7f9777ca90, R13: 0xffffff8052616820, R14: 0xffffff7f9777ca78, R15: 0xffffff805397e800
RFL: 0x0000000000010246, RIP: 0xffffff80142d15b3, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000008, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffff82e097b820 : 0xffffff80141aca1d mach_kernel : _handle_debugger_trap + 0x48d
0xffffff82e097b870 : 0xffffff80142e6b13 mach_kernel : _kdp_i386_trap + 0x153
0xffffff82e097b8b0 : 0xffffff80142d859a mach_kernel : _kernel_trap + 0x4fa
0xffffff82e097b920 : 0xffffff8014159ca0 mach_kernel : _return_from_trap + 0xe0
0xffffff82e097b940 : 0xffffff80141ac437 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff82e097ba60 : 0xffffff80141ac283 mach_kernel : _panic + 0x63
0xffffff82e097bad0 : 0xffffff80142d87bd mach_kernel : _kernel_trap + 0x71d
0xffffff82e097bc40 : 0xffffff8014159ca0 mach_kernel : _return_from_trap + 0xe0
0xffffff82e097bc60 : 0xffffff80142d15b3 mach_kernel : _lck_mtx_lock + 0x3
0xffffff82e097be00 : 0xffffff7f9777b9ac com.blankmac.VoodooI2CUPDDEngine : __Z23initialiseGestureSocketv + 0xe5
0xffffff82e097be20 : 0xffffff7f9777be2e com.blankmac.VoodooI2CUPDDEngine : __ZN19VoodooI2CUPDDEngine5startEP9IOService + 0x6a
0xffffff82e097be40 : 0xffffff801482c2fb mach_kernel : __ZN9IOService14startCandidateEPS_ + 0x6b
0xffffff82e097be80 : 0xffffff801482c041 mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0x911
0xffffff82e097bf00 : 0xffffff801482b597 mach_kernel : __ZN9IOService14doServiceMatchEj + 0x2c7
0xffffff82e097bf50 : 0xffffff801482d066 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x1a6
0xffffff82e097bfa0 : 0xffffff80141590ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.blankmac.VoodooI2CUPDDEngine(1.0)[B475F4D9-2398-3DBA-90E0-84F56B4A70FB]@0xffffff7f9777b000->0xffffff7f9777efff
            dependency: com.alexandred.VoodooI2C(2.1.4)[E7230338-1AFE-3830-B753-09751D6BABA3]@0xffffff7f97590000

BSD process name corresponding to current thread: Unknown
Boot args: -applealcbeta brcmfx-country=US keepsyms=1 -disablegfxfirmware -v -lilubetaall 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 18.0.0: Wed Aug 22 20:13:40 PDT 2018; root:xnu-4903.201.2~1/RELEASE_X86_64
Kernel UUID: DF83AD37-501E-3B4F-B1F0-04F3AC90FE35
Kernel slide:     0x0000000013e00000
Kernel text base: 0xffffff8014000000
__HIB  text base: 0xffffff8013f00000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)

System uptime in nanoseconds: 3727800606
last loaded kext at 3245488101: com.apple.iokit.IOUSBFamily	900.4.2 (addr 0xffffff7f95122000, size 634880)
loaded kexts:
com.blankmac.VoodooI2CUPDDEngine	1
me.kishorprins.VoodooI2CELAN	1
me.kishorprins.VoodooI2CFTE	1
org.rehabman.driver.FakePCIID	1.3.12
com.sinet3k.Sinetek-rtsx	1
com.insanelymac.RealtekRTL8111	1.3.0d1
org.hwsensors.driver.LPCSensors	1800
com.alexandred.VoodooI2CHID	1
org.rehabman.driver.AppleSmartBatteryManager	1.90
org.hwsensors.driver.ACPISensors	1800
org.hwsensors.driver.SMMSensors	1800
org.hwsensors.driver.CPUSensors	1800
as.lvs1974.AirportBrcmFixup	1.1.5
[... 68 lines cut ...]
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	1.0

EOF
Model: MacBookPro14,1, BootROM MBP141.0178.B00, 4 processors, Intel Core i7, 2.81 GHz, 24 GB, SMC 2.43f6
Graphics: Intel HD Graphics 630, Intel HD Graphics 630, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR4, 2400 MHz, Samsung, M471A1K43CB1-CRC
Memory Module: BANK 1/DIMM0, 16 GB, DDR4, 2400 MHz, Crucial Technology, CT16G4DFS824A.8FB1
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x19), Broadcom BCM43xx 1.0 (7.21.190.32 AirPortDriverBrcm4360-1302)
Bluetooth: Version 6.0.8f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
PCI Card: Intel HD Graphics 630, Display Controller, Built In
Serial ATA Device: SAMSUNG SSD PM830 2.5" 7mm 128GB, 128.04 GB
Serial ATA Device: ST1000LM035-1RK172, 1 TB
USB Device: USB 3.0 Bus
USB Device: ITE Device(8910)
USB Device: BCM20702A0
USB Device: USB2.0 HD UVC WebCam
USB Device: MI 5s
Thunderbolt Bus:

QiuChenly, 2018-09-27:
This time we caught the log of system collapse caused by four fingers kneading.
The log above shows that the system will occasionally reboot abruptly during startup, which is also a problem - he is not well supported during startup.

--------------------------------------       by four fingers kneading   logs    -------------------

Anonymous UUID:       13F830B3-4799-F0BD-F36A-32E608BAEE8C

Fri Sep 28 01:56:37 2018

*** Panic Report ***
panic(cpu 0 caller 0xffffff80058d87bd): Kernel trap at 0xffffff7f88d7bd20, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000260, CR3: 0x000000001af84000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000005, RCX: 0x0000000000000001, RDX: 0x0000000000000004
RSP: 0xffffff9afb55bdb0, RBP: 0xffffff9afb55bde0, RSI: 0xffffff7f88ba5ba0, RDI: 0x0000000000000000
R8:  0x0000000000000004, R9:  0x0000000000000000, R10: 0x0000000000000001, R11: 0x0000000000000009
R12: 0xffffff8045218040, R13: 0x0000000000000006, R14: 0xffffff8045263120, R15: 0x0000000000000000
RFL: 0x0000000000010246, RIP: 0xffffff7f88d7bd20, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000260, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff9afb55b880 : 0xffffff80057aca1d mach_kernel : _handle_debugger_trap + 0x48d
0xffffff9afb55b8d0 : 0xffffff80058e6b13 mach_kernel : _kdp_i386_trap + 0x153
0xffffff9afb55b910 : 0xffffff80058d859a mach_kernel : _kernel_trap + 0x4fa
0xffffff9afb55b980 : 0xffffff8005759ca0 mach_kernel : _return_from_trap + 0xe0
0xffffff9afb55b9a0 : 0xffffff80057ac437 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff9afb55bac0 : 0xffffff80057ac283 mach_kernel : _panic + 0x63
0xffffff9afb55bb30 : 0xffffff80058d87bd mach_kernel : _kernel_trap + 0x71d
0xffffff9afb55bca0 : 0xffffff8005759ca0 mach_kernel : _return_from_trap + 0xe0
0xffffff9afb55bcc0 : 0xffffff7f88d7bd20 com.blankmac.VoodooI2CUPDDEngine : __ZN19VoodooI2CUPDDEngine21handleInterruptReportE24VoodooI2CMultitouchEventy + 0x8e
0xffffff9afb55bde0 : 0xffffff7f88b96bbd com.alexandred.VoodooI2C : __ZN28VoodooI2CMultitouchInterface21handleInterruptReportE24VoodooI2CMultitouchEventy + 0x85
0xffffff9afb55be30 : 0xffffff7f88caa11f com.alexandred.VoodooI2CHID : __ZN33VoodooI2CMultitouchHIDEventDriver21handleInterruptReportEyP18IOMemoryDescriptor15IOHIDReportTypej + 0x123
0xffffff9afb55be80 : 0xffffff7f865b3ead com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice20handleReportWithTimeEyP18IOMemoryDescriptor15IOHIDReportTypej + 0x2ff
0xffffff9afb55bf20 : 0xffffff7f865b242c com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice12handleReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x56
0xffffff9afb55bf60 : 0xffffff7f88cad211 com.alexandred.VoodooI2CHID : __ZN18VoodooI2CHIDDevice14getInputReportEv + 0xbd
0xffffff9afb55bfa0 : 0xffffff80057590ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[0101D74C-CA38-370E-88CB-8BCFB2F42810]@0xffffff7f865aa000->0xffffff7f86627fff
         com.alexandred.VoodooI2C(2.1.4)[E7230338-1AFE-3830-B753-09751D6BABA3]@0xffffff7f88b90000->0xffffff7f88bbdfff
            dependency: com.alexandred.VoodooI2CServices(1)[409F4A01-3D35-32FF-B7E5-DB77CB7F52AF]@0xffffff7f88b7e000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[8A2C5602-298A-3199-AED0-979018ECBE16]@0xffffff7f8663e000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[0101D74C-CA38-370E-88CB-8BCFB2F42810]@0xffffff7f865aa000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[2CE7BCB3-0766-3A94-A8D4-29BF3EBAEFBC]@0xffffff7f86095000
            dependency: org.coolstar.VoodooGPIO(1.1)[125807B8-4E2E-341B-AE46-F06983D562A3]@0xffffff7f88b83000
         com.alexandred.VoodooI2CHID(1.0)[0C2FCF61-54DA-3FE5-99E3-B9E62A0AC8D2]@0xffffff7f88ca7000->0xffffff7f88cc6fff
            dependency: com.alexandred.VoodooI2C(2.1.4)[E7230338-1AFE-3830-B753-09751D6BABA3]@0xffffff7f88b90000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[8A2C5602-298A-3199-AED0-979018ECBE16]@0xffffff7f8663e000
            dependency: com.apple.iokit.IOGraphicsFamily(530.9)[1EEBFCDD-6965-336F-AE19-D177CE0E7BC1]@0xffffff7f8681a000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[0101D74C-CA38-370E-88CB-8BCFB2F42810]@0xffffff7f865aa000
         com.blankmac.VoodooI2CUPDDEngine(1.0)[B475F4D9-2398-3DBA-90E0-84F56B4A70FB]@0xffffff7f88d7b000->0xffffff7f88d7efff
            dependency: com.alexandred.VoodooI2C(2.1.4)[E7230338-1AFE-3830-B753-09751D6BABA3]@0xffffff7f88b90000

BSD process name corresponding to current thread: kernel_task
Boot args: -applealcbeta brcmfx-country=US keepsyms=1 -disablegfxfirmware -v -lilubetaall 

Mac OS version:
18A391

Kernel version:
Darwin Kernel Version 18.0.0: Wed Aug 22 20:13:40 PDT 2018; root:xnu-4903.201.2~1/RELEASE_X86_64
Kernel UUID: DF83AD37-501E-3B4F-B1F0-04F3AC90FE35
Kernel slide:     0x0000000005400000
Kernel text base: 0xffffff8005600000
__HIB  text base: 0xffffff8005500000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)

System uptime in nanoseconds: 357582875291
last loaded kext at 13775687995: com.apple.fileutil	1 (addr 0xffffff7f868b4000, size 110592)
loaded kexts:
com.no-one.BrcmPatchRAM2	2.2.10
com.no-one.BrcmFirmwareStore	2.2.10
[... 156 lines cut ...]
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
[... cut ...]
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x19), Broadcom BCM43xx 1.0 (7.21.190.32 AirPortDriverBrcm4360-1302)
Bluetooth: Version 6.0.8f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
PCI Card: Intel HD Graphics 630, Display Controller, Built In
Serial ATA Device: SAMSUNG SSD PM830 2.5" 7mm 128GB, 128.04 GB
Serial ATA Device: ST1000LM035-1RK172, 1 TB
USB Device: USB 3.0 Bus
USB Device: ITE Device(8910)
USB Device: BCM20702A0
USB Device: USB2.0 HD UVC WebCam
USB Device: MI 5s
Thunderbolt Bus:

ben9923, 2018-09-27:
As you only have an ELAN1200, you should only use the VoodooI2CHID satellite (With VoodooI2C, of course).
The others (VoodooI2CELAN, VoodooI2CFTE and VoodooI2CUPDDEngine) should be removed.
Please test again after properly removing them, and rebuilding your kextcache.

QiuChenly, 2018-09-27:
Thank you for your help.
The above-mentioned drivers have been removed, showing no problem at all, and everything seems to be normal.
Just four-finger kneading and opening don't seem to work very well on my ELAN 1200 device, and it often fails to detect my four-finger kneading and opening.

closed 2018-09-27 by QiuChenly

ben9923, 2018-09-28:
The panic was caused by VoodooI2CUPDDEngine. It was accidentally reverted back to an old version that could cause a panic in some cases.
This issue should be fixed in the next release of VoodooI2C.

Removing it was just a workaround, but in your case you don't need it anyway. Make sure you only install kexts you need to avoid problems, not just with VoodooI2C ;)

About 4 finger gestures - swipe gestures can be configured to use 3 fingers in the SysPrefs. Pinch/Spread gestures are problematic, I hope we can improve them in the future.