VoodooI2C/VoodooI2C #108: two-finger actions involving pressing down the left trackpad button do not work anymore

LeeBinder, 2018-09-27:
Dear Alexandre,

merci for your trackpad driver. I have been using v.2.0.3 for a while which has been acting up every once in a while with the mouse pointer suddenly jumping across the screen, not analog anymore with my movements on the trackpad, so I've been hoping for a fix in a future update.

Now ever since v.2.1 and also still with your current 2.1.4 I have been experiencing another bug, so I cannot say if the bug described above is solved.

The new bug is: any two-finger actions involving pressing down the left trackpad button do not work anymore, like:
- dragging a window
- highlighting text in a document
- moving a slider left <-> right

- Hardware: Asus Vivobook S15 with ELAN 1300 (8086 : 9d60 / 9d61)
- DSDT patches applied: https://bit.ly/2N9pc3A (I2C PATCH, Window10 Patch, Add GPIO Controller Enable)
- both relevant kexts (VoodooI2C.kext, VoodooI2CHID.kext) installed to /L/E with correct permissions and kernel/ kext cache rebuilt (I am aware that the other 3 kexts are NOT for the ELAN 1300)
- latest Clover

I would love to move beyond v.2.0.3 because bug #1 is too annoying over time.

Please let me know if you already know which code from 2.0.3 to reimplement into 2.1.4, or if you'd like me to supply a complete gen_debug problem reporting dump for you.

Thanks for all your efforts here,
Lee

PS: hieplpvip's VoodooI2C Asus Zenbook fork has been having an acceleration issue with my ELAN 1300 (even his latest fork of your 2.1.4), so I can not use his fork.

gujiangjiang, 2018-09-27:
https://github.com/alexandred/VoodooI2C/issues/105#issuecomment-424909888

Here maybe the answer.

LeeBinder, 2018-09-27:
> [#105 (comment)](https://github.com/alexandred/VoodooI2C/issues/105#issuecomment-424909888)
> 
> Here maybe the answer.

thank you, @gujiangjiang !

alexandred, 2018-09-29:
I have renamed this issue and will leave it open until we figure out a universal issue to the scaling problems.

LeeBinder, 2018-09-29:
hm OK - even though I do not understand how the topic title reflects my issue description, because 
- notification center is not invoked
- the press action is about pressing down ("click + hold") the LEFT trackpad button

Alexandre, I do have Xcode installed. If you point me to the appropriate .cpp file, I can go ahead with trying to import the 2.0.3 code into the current 2.1.4, compile and see if again works fine for my ELAN 1300.

Do I have to look at

https://github.com/alexandred/VoodooI2C/blob/master/Multitouch%20Support/Native/VoodooI2CMT2SimulatorDevice.cpp

or a different .cpp?

LeeBinder, 2018-09-29:
OK. Hieplpvip actually managed to enable scaling for his Asus Zenbook trackpad, and it does work for my Asus Vivobook trackpad, too, however is the sensitive zone too small so that I almost always miss the "sweet spot".

Can you point me to the part in the code where I can enlargen the clickable area? Above all it needs to be widened towards the top:

https://github.com/hieplpvip/VoodooI2C/blob/master/Multitouch%20Support/Native/VoodooI2CMT2SimulatorDevice.cpp

I already asked Hieplpvip but he does not know.

alexandred, 2018-09-29:
The magic values are `factor_x` and `factor_y` found here: https://github.com/hieplpvip/VoodooI2C/blob/e0f5ab2140920f33ec323bc6d7b919b03f1ff905/Multitouch%20Support/Native/VoodooI2CMT2SimulatorDevice.cpp#L97 .

@hieplpvip fixed it for his laptop by simply removing the two occurrences of `factor_x` and the one occurrence of `factor_y` in the part of the code I linked above. You could try fiddling with the values there to see if it makes a difference. On my laptop, for example, I need those factors for the scaling to be correct.

LeeBinder, 2018-09-29:
great, very valuable hint, I'll do asap.

Another thing which changed from 2.0.3 to 2.1.4 and which hieplpvip didn't know how to change, either, is the pointer accelleration which is too fast even at slow pointer speed. Sadly there is no separate slider for acceleration in the trackpad pref pane (maybe Apple did not implement it at all?).

Do you happen to know where  (in which .cpp and line #) that value is set?

alexandred, 2018-09-29:
That is also related to scaling - if you manage to scale the trackpad correctly then the speed set in the system preferences should be very natural.

LeeBinder, 2018-09-29:
aha, I would never have guessed that, thanks for taking the time to explain all this.

Do you actually know if the random trackpad pointer jumps from 2.0.3 and below are resolved with the Magic Trackpad 2 code?

alexandred, 2018-09-29:
I'm not sure I know what the "random trackpad pointer jumps" are. I didn't experience such a thing myself as far as I remember. Could you perhaps describe it in more detail?

LeeBinder, 2018-09-29:
In the middle of moving over the trackpad, the pointer "goes crazy" - moves in a different direction than how one moves the finger. To get back to normal, some odd keyboard strokes always help (CMD + H, Fn 11, CMD + Tab to switch back into the effected app).

This is experienced by many if not all who have been using VoodooI2C 2.0.3 or below on Asus Zenbook/ Vivobook with ELAN 1300 (and possibly other ELAN, can't exactly recall from memory).

alexandred, 2018-09-29:
This is the first I've heard of such an issue so I can't promise that it has been fixed in v2.1+ . That being said, the entire multitouch engine has changed since v2.0.3 so its possible that it has been fixed now.

LeeBinder, 2018-09-29:
wow really, was I the first to break the ice, can hardly believe. This bug has been described all over in Asus Vivobook/ Zenbook threads on tonymac, YouTube.. 

So I do hope that with 2.1.4+ this bug is a matter of the past :)

Andrw0830, 2018-09-30:
I can confirm that removing factor_x and factor_y fixes the scaling as one-finger click does work, but it removes all the benefit that v2.1 has had and has re-introduced the scrolling inertia issue. I had no idea that factor_x and factor_y was so tied into how the Trackpad responds.

LeeBinder, 2018-10-01:
I also confirm the scrolling inertia issue with hieplpvip's 2.1.4 branch which has factor_x and factor_y removed, which is another reason I reverted back to alexandred/VoodooI2C 2.0.3

BesnikRrustemi, 2018-10-02:
@alexandred I have a Lenovo V330-15IKB with ELAN I2C Trackpad/Touchpad
TPD0 "ELAN0618" 8086:9d60

I can't recall correctly from which version but I believe updates after 2.0.3 broke the left and right click functionality on the VoodooI2CElan.kext.

I can use VoodooI2C.kext + VoodooI2CHID.kext but with this combination I only have the Left & Right Click working and normal mouse function on the touchpad.

If I install VoodooI2C.kext + VoodooI2CElan.kext I have the gestures but the left and right click (physical clicks) are broken.

Previously they used to work 2.0.3 and older (even though the clickable area was to small, I had to click the right and left click to the edge to invoke the physical left or right click function).

Let me know what kind of logs do you need to analyze the issue & which file do I need to change If I decide to use 2.0.3 and older to increase the clickable area of physical left & right click of the trackpad/touchpad.

My laptop touchpad is all flat, but the surface near the bottom can be clicked on left and right (physical left and right click).

yoasif, 2018-10-03:
FWIW, using VoodooI2CHiD with ELAN0651 and I am seeing my cursor randomly jump across the screen while I use the trackpad - it seems to end up on the top right of the screen (notification center icon area). 

Happy to provide any further details that may be helpful, but I wanted to post a "me too" with the trackpad model where I am seeing the issue in case you wanted to know other models were affected.

dderrida, 2018-10-16:
Same issue reported on Dell XPS13
8086:9d61
TDP1 DLL075B
Not sure how to report properly these issues. Happy to send logs if needed.
Merci

BesnikRrustemi, 2018-10-16:
@alexandred reverting back to 2.0.3 works just fine on my Lenovo V330-15IKB running macOS Mojave.
My ELAN0618 Right click works now but I have a question if you can point me on where I can do some adjustments to this kext:

Left click and right click work fine but I have to press the very bottom edge of the laptop in order for the left click and right click to work.

What values can I edit to adjust those settings so the surface can be larger for the clicks

coolstar, 2018-12-17:
Should be fixed in commit ecabd7461c549f29a8a5b776a8b10041232dff25

LeeBinder, 2018-12-17:
> Should be fixed in commit [ecabd74](https://github.com/alexandred/VoodooI2C/commit/ecabd7461c549f29a8a5b776a8b10041232dff25)

splendid, coolstar - your code changes look promising. Have you compiled and tested successful with your rig?
Any idea when the release with your commit might be published?
If you need a beta tester, feel free to send a pre-release my way; I have a bit of time right now.

BesnikRrustemi, 2018-12-17:
Just came back from Gym, while i was going there, i got the email about @coolstar changes.

Download repositories right now and will try to make a build, test and upload it for others to try it.

LeeBinder, 2018-12-17:
yee-hah!!
PS: you got the correct order down - 1st physical work-out, then computer stuff ;)

alexandred, 2018-12-17:
Just a heads up that this will not yet fix the right click issue but if everything checks out then we should be able to sort it out easily with scaling fixed.

BesnikRrustemi, 2018-12-17:
> Just a heads up that this will not yet fix the right click issue but if everything checks out then we should be able to sort it out easily with scaling fixed.

Thats great to hear, what im more concerned is Right Click, it's really painful to borrow my mouse with myself everywhere i go.

Im using VoodooI2C.kext & VoodooI2CELAN.kext as i have a ELAN I2C precision touchpad on my Lenovo V330-15IKB. (it's ELAN0618)

coolstar, 2018-12-17:
Got a better fix for the scaling problem. Will push shortly.

LeeBinder, 2018-12-17:
> Thats great to hear, what im more concerned is Right Click, it's really painful to borrow my mouse with myself everywhere i go.
@BesnikRrustemi : what about two-finger tap instead right click? I hardly ever (if at all ever) have been using physical right click anymore for ages. Two-finger tap is way more convenient.

BesnikRrustemi, 2018-12-17:
> @BesnikRrustemi : what about two-finger tap instead right click? I hardly ever (if at all ever) have been using physical right click anymore for ages. Two-finger tap is way more convenient.

That's what i have been using till your commit. but now even two fingers tap doesn't work, neither gestures. i built with you recent fix again, but didn't fix the issue with two fingers tap and neither gestures.

I got used to two fingers tap but really, i would much more appreciate the physical right click. this is pretty much my only issue with this laptop.

coolstar, 2018-12-17:
Scaling should be completely fixed as of 33db3b99ffeb057dd7d3ea8985332b60fc343912

BesnikRrustemi, 2018-12-18:
> Scaling should be completely fixed as of [33db3b9](https://github.com/alexandred/VoodooI2C/commit/33db3b99ffeb057dd7d3ea8985332b60fc343912)

Cursor still jumps to random locations when i double tap for right click but it jumps closer to where it was when i tapped.
2 Fingers tap for right click still not working, no gestures working.

Although i definitely noticed improvements.

coolstar, 2018-12-18:
For new commenters, please check trackpad operation with FingerMgmt. Gestures can break for a number of reasons; we need to verify the touch data is working correctly (and if not, which part is broken): https://github.com/jnordberg/FingerMgmt/releases

coolstar, 2018-12-20:
Scaling is completely fixed as of latest commit. To verify scaling, check the corners of your trackpad show up as the corners in FingerMgmt. Any other issues remaining are likely not from scaling.

closed 2018-12-20 by coolstar

LeeBinder, 2019-01-02:
I never manage to compile from https://github.com/alexandred/VoodooI2C/archive/master.zip, even though I also downloaded https://github.com/coolstar/VoodooGPIO/archive/9e0b3e2e6a64c5d057cad16d94df32f962c45292.zip and applied its content into VoodooI2C-master/Dependencies/VoodooGPIO. I've been reading the documentation, but I'm new to Xcode and I admit it's above my head for now.

@BesnikRrustemi (or anybody else): can you please compile the most recent source code as you already did before and supply a link here? That would be awesome. Thank you

LeeBinder, 2019-01-02:
@andrewwyatt : thank you, but with it my trackpad is dead. Most likely because of the 10.14 SDK? I only installed the same kexts I have been using (VoodooI2C, VoodooI2CHID), and did nothing with any of the dSYM files.

Andrew, can you maybe also upload the source package containing all the files? Maybe I did already something wrong assembling the source. I would sure be able to compile myself, and I think there's just one or two steps missing because I am inexperienced with Xcode (I somewhat know Microsoft Visual Studio, which does not seem to help much here LOL).

andrewwyatt, 2019-01-02:
@LeeBinder https://github.com/andrewwyatt/VoodooI2C

It's just a fork that I've been using for some local testing, but it has the linter and documentation generation steps disabled which was necessary to build.

LeeBinder, 2019-01-02:
Partial success: working trackpad with correct scaling (!!!) aka drag & drop / highlighting/ moving windows incl. a nice large sensitive area on the trackpad as with v.2.0.3 :) !!

Via work-around:

VoodooI2C: your compile

VoodooI2CHID: current original AlexandreD 2.1.4

So your VoodooI2CHID kext not  but your VoodooI2C working in my 10.13.6 should imply that this part is NOT a compile SDK issue, shouldn't it..

I'd still be interested in your complete source code package, also because I am experiencing two new annoying issues now (just mentioning them, not expecting any fixing here):

1. after any combined keyboard operation incl. a modifier keyvia , (cmd+ c copy, cmd + s save, cmd + x cut etc.), the mouse pointer is "paralyzed"/ like in stasis for maybe half a second (I think I read AlexandreD somewhere mentioning this is a unfortunate standard behavior of the MT driver? Or has this been fixed in the most current VoodooI2CHID?)

2. About 1/3 of the time, in a text file, instead of highlighting, the context menu appears, because the driver thinks I'm doing a double tap.

I hope this engine change turns from a win/ lose into a win/ win package.

LeeBinder, 2019-01-02:
@@andrewwyatt : Partial success: working trackpad with correct scaling (!!!) aka drag & drop / highlighting/ moving windows incl. a nice large sensitive area on the trackpad as with v.2.0.3 :) !!

Via work-around:

VoodooI2C: your compile Andrew Wyatt (blue)

VoodooI2CHID: current original AlexandreD 2.1.4 (pink)

So your VoodooI2CHID kext not  but your VoodooI2C working in my 10.13.6 should imply that this part is NOT a compile SDK issue, shouldn't it..

I'd still be interested in your complete source code package, also because I am experiencing two new annoying issues now (just mentioning them, not expecting any fixing here):

1. after any combined keyboard operation incl. a modifier keyvia , (cmd+ c copy, cmd + s save, cmd + x cut etc.), the mouse pointer is "paralyzed"/ like in stasis for maybe half a second (I think I read AlexandreD somewhere mentioning this is a unfortunate standard behavior of the MT driver? Or has this been fixed in the most current VoodooI2CHID?)

2. About 1/3 of the time, in a text file, instead of highlighting, the context menu appears, because the driver thinks I'm doing a double tap.

I hope this engine change turns from a win/ lose into a win/ win package.

LeeBinder, 2019-01-02:
@andrewwyatt : great, managed to compile! Before I set any occurrence of 10.13 to 10.14. Will report back soon :)

LeeBinder, 2019-01-02:
@andrewwyatt : hm I don't get it :( Xcode keeps on displaying "Build succeeded" but no matter where I look, I don't find the output. I compiled some other kexts before, and the output folder was simply called "build" in the project folder. Tried cmd+B and also all 3 options in the "Build for" menu.

BesnikRrustemi, 2019-01-02:
> @andrewwyatt : hm I don't get it :( Xcode keeps on displaying "Build succeeded" but no matter where I look, I don't find the output. I compiled some other kexts before, and the output folder was simply called "build" in the project folder. Tried cmd+B and also all 3 options in the "Build for" menu.
You can find your builds at:
/Users/**yourusername**/Library/Developer/Xcode/DerivedData

LeeBinder, 2019-01-02:
@andrewwyatt 👍 yes, just figured it out 10 sesonds ago LOL. Thanks man for standing by. Another week and I'll be an Xcode Guru (with a digital turban or something). Will report back in a bit.

LeeBinder, 2019-01-03:
@andrewwyatt 👍 yeeha, self-compiled release kexts work! But the two issues [described above](https://github.com/alexandred/VoodooI2C/issues/108#issuecomment-451006810) remain. Not sure if swapping one or both of the plugins kexts inside VoodooI2C.kext with current official 2.1.4 release one could do any good. Might try tomorrow, or might not (I'm pretty sure I already experienced this mouse pointer lag on combo keyboard actions with the release 2.1.4).

For now back to 2.0.3. That one performs even smoother than the Windows drivers (and no lag), but the random but frequent pointer jumping is somewhat annoying.

Thanks again for being my Xcode jumper cable tonight :)

LeeBinder, 2019-01-05:
For completion, my 10.13.6 compile also attached:

[VoodooI2C 2.14 10.13.6 2019-01-03.zip](https://github.com/alexandred/VoodooI2C/files/2729990/VoodooI2C.2.14.10.13.6.2019-01-03.zip)

LeeBinder, 2019-02-12:
Hello @andrewwyatt . I finally installed Mojave on a separate partition and am now experiencing two challenges:

1. VoodooI2C 2.0.3 (the version running close to perfect with my ELAN 1300 in High Sierra 10.13.6) is not Mojave compatible --> Kernel Panic :(

I downloaded the source code from [here](https://github.com/alexandred/VoodooI2C/releases/tag/v2.0.3) but am getting the same compile errors I was getting with the stock 2.1.4 source code

2. You seem to have removed your fork you've been using for some local testing with the linter and documentation generation steps disabled which was necessary to build:

> git clone https://github.com/andrewwyatt/VoodooI2C
> cd VoodooI2C
> git submodule update --init --recursive

Can you give me some pointers? Like:

1. working git commands which download a fork of the most recent 2.1.4 code which compiles w/o errors?

2. or step-by-step instrus how to disable linter and documentation generation?

3. I might prefer a 2.0.3 Mojave compile from [the 2.0.3 code](https://github.com/alexandred/VoodooI2C/releases/tag/v2.0.3) until there is a working magic trackpad version for the Asus ELAN 1300. Can you tell me what I need to do so it compiles in the latest Xcode? I did not find a 2.0.3 compiled for Mojave.

Again, I am still new to Xcode but am willing to learn.

Hopefully you're still around. If not, maybe someone else reading this can give me some hints 👍