VoodooI2C/VoodooI2C #113: Teleporting Mouse

Lockzi, 2018-09-29:
Dear Alexandred,

as per the Dell XPS 9550 thread on insanelymac.

**Problem description**
If I draw a circle with the mousepointer using the trackpad continuously I can see how it teleports in the last known direction of the circle - exactly every second, which makes it impossible to do a circle.[Teleporting mouse screenrecording.mov.zip](https://github.com/alexandred/VoodooI2C/files/2430489/Teleporting.mouse.screenrecording.mov.zip)

**Actions already taken**
Disabled the ACPIBatteryManager kext

**Hardware**
Dell XPS 15 9550

Motherboard: Dell 0N7TVV | South Bridge: Intel Skylake-H rev 07 | North Bridge: Intel Skylake-H PCH rev 31 | BIOS: 1.8.0 CPU: i7-6700HQ | Memory: 2x Samsung DDR4 SDRAM 16384MB @ 2133 MHz | Integrated GPU: Intel Skylake Graphics HD530 GOP 1035 64MB | Dedicated GPU: Nvidia GTX 960M (GM107M) | SSD: THNSN5102DU7 NVMe Toshiba 1Tb | Screen: 15" UHD-1 (4K) ELAN Touchscreen | Audio: Realtek ALC3266 | WiFi: BCM43602 802.11AC | Battery: Dell 1P6KD5A

**Mac OS**
10.13.3

**DSDTs and IOReg**
DSDTs and ioreg.zip](https://github.com/alexandred/VoodooI2C/files/2430488/DSDTs.and.ioreg.zip)

**Kextstat**
❯ kextstat | grep -v com.apple
Index Refs Address Size Wired Name (Version) UUID 
16 1 0xffffff7f86031000 0x5000 0x5000 com.alexandred.VoodooI2CServices (1) 409F4A01-3D35-32FF-B7E5-DB77CB7F52AF <4 3>
18 1 0xffffff7f86036000 0xd000 0xd000 org.coolstar.VoodooGPIO (1.1) 125807B8-4E2E-341B-AE46-F06983D562A3 <11 5 4 3>
19 1 0xffffff7f86043000 0x2e000 0x2e000 com.alexandred.VoodooI2C (2.1.4) E7230338-1AFE-3830-B753-09751D6BABA3 <18 17 16 12 11 5 4 3 1>
21 0 0xffffff7f81245000 0xf000 0xf000 com.displaylink.driver.DisplayLinkDriver (4.1 (11)) 5A240BCF-1B8D-3A9E-B21B-B12D639AC9D9 <20 12 5 4 3>
44 7 0xffffff7f86071000 0x21000 0x21000 as.vit9696.Lilu (1.2.7) 4EA74271-18D4-38CC-AAD2-03F2875BCF14 <7 5 4 3 2 1>
45 0 0xffffff7f86092000 0x55000 0x55000 as.lvs1974.IntelGraphicsFixup (1.2.7) C2ADBDCC-C5DD-3CF4-A8AF-A6146F0F5D4B <44 12 7 5 4 3 2 1>
46 0 0xffffff7f860e7000 0xd3000 0xd3000 as.vit9696.AppleALC (1.3.2) 42E1908A-664A-3B2C-903A-070129B0F27A <44 12 7 5 4 3 2 1>
47 0 0xffffff7f861ba000 0x4000 0x4000 org.vanilla.driver.CoreDisplayFixup (1.3.7) A081F0C6-9E16-3680-8D6E-4863A18DBD53 <44 7 5 4 3 2 1>
48 2 0xffffff7f861be000 0x11000 0x11000 org.netkas.driver.FakeSMC (1800) FC7031CC-C949-35D1-AC1A-CA8DB349793E <11 7 5 4 3 1>
49 0 0xffffff7f861cf000 0x4000 0x4000 as.lvs1974.HibernationFixup (1.2.3) 67C31FBF-C503-3F5F-B5E1-2B75B2A5DBDA <44 7 5 4 3 2 1>
50 0 0xffffff7f861d3000 0x6000 0x6000 com.XPS.VerbStub (1.0.3) 94CAA03E-C91B-3DC7-AD1F-943A4D5E08B5 <12 4 3>
51 0 0xffffff7f861de000 0x3000 0x3000 xyz.racermaster.NoTouchID (1.0.0) D699F26D-AFA3-34CA-B727-C437DCAD0E1C <44 7 5 4 3 2 1>
52 0 0xffffff7f861e1000 0x3000 0x3000 as.lvs1974.BT4LEContiunityFixup (1.1.1) 6A75BE3C-3F52-307D-BA56-F65E39F08F96 <44 7 5 4 3 2 1>
53 0 0xffffff7f861e4000 0x6000 0x6000 as.lvs1974.AirportBrcmFixup (1.1.5) 2156FBDA-3468-3C95-8486-B96435A86E32 <44 22 12 7 5 4 3 2 1>
54 0 0xffffff7f861ea000 0x5000 0x5000 org.hwsensors.driver.CPUSensors (1800) ED90DA0F-5876-3DB5-82EC-CEEA470A7479 <48 7 5 4 3>
64 0 0xffffff7f861f0000 0x8000 0x8000 org.hwsensors.driver.ACPISensors (1800) FA4DF6C0-0D3A-3CE2-8569-0AB2C84053E7 <48 11 7 5 4 3>
66 0 0xffffff7f861fa000 0x9000 0x9000 org.rehabman.driver.AppleSmartBatteryManager (1.90) 0EE476DC-884B-35FF-BD9E-E24BEC25A521 <11 4 3>
69 0 0xffffff7f812c2000 0x2000 0x2000 org.rehabman.driver.NullEthernet (1.0.3) 4B72A7F4-04C2-3EAC-BF31-168CFB428701 <22 11 5 4 3 1>
70 2 0xffffff7f86206000 0xb000 0xb000 org.rehabman.voodoo.driver.PS2Controller (1.9.0) 2207C797-FB9F-30CB-B3A0-D07FC4883BE5 <11 7 5 4 3 1>
75 0 0xffffff7f80fcc000 0x17e000 0x17e000 at.obdev.nke.LittleSnitch (5187) 530932AE-E8F4-3521-8065-642CB7C7B135 <7 5 4 3 1>
80 0 0xffffff7f86211000 0x7000 0x7000 org.rehabman.driver.FakePCIID (1.3.13) 215C3F8E-9145-3D5C-80A5-08CFAB003727 <12 7 5 4 3 1>
96 0 0xffffff7f86218000 0x6000 0x6000 org.rehabman.voodoo.driver.PS2Keyboard (1.9.0) C61C243B-5496-35FA-9F10-09764929F8A7 <70 24 17 7 6 5 4 3 1>
97 0 0xffffff7f8621e000 0xc000 0xc000 org.rehabman.voodoo.driver.PS2Trackpad (1.9.0) 49F63682-0BBA-34EB-9974-AAB604CCD656 <70 24 17 7 6 5 4 3 1>
100 1 0xffffff7f8622a000 0x25a000 0x25a000 com.no-one.BrcmFirmwareStore (2.2.10) FEAD3CAB-F1B9-329A-8291-EAC222EBC0F0 <5 4 3>
101 0 0xffffff7f86484000 0x6000 0x6000 com.no-one.BrcmPatchRAM2 (2.2.10) EA8FADAF-E7F3-3CDF-8440-49C15C0E25F8 <100 29 5 4 3 1>
105 0 0xffffff7f8648a000 0x11000 0x11000 com.alexandred.VoodooI2CHID (1) 0C2FCF61-54DA-3FE5-99E3-B9E62A0AC8D2 <20 19 17 11 5 4 3>
154 0 0xffffff7f8127f000 0x6000 0x6000 com.kairos.driver.DuetDisplay (1) 0A9F5387-316E-3D61-B9ED-D066346F8797 <20 5 4 3>
172 0 0xffffff7f8649b000 0x88000 0x88000 com.avast.FileShield (4.0.0) AF1C89B5-14E1-3AB2-90C7-EFA7C75AF6FC <5 4 1>
173 0 0xffffff7f86523000 0x3000 0x3000 com.avast.PacketForwarder (2.1) 23DA7908-63C7-3658-8EEF-70B8C602D4FD <4 1>
177 0 0xffffff7f86591000 0x7000 0x7000 com.joshuawise.kexts.HoRNDIS (6) 2FF769BE-1667-3CE2-A8AE-C9D86BAF30D4 <73 22 5 4 3 1>

alexandred, 2018-09-29:
This appears to be an issue with how we are simulating the MT2 protocol; we used to have similar issues back when we were figuring out the protocol and I thought I had fixed it. The bizarre thing is that I have the exact same laptop and I don't have this issue...

Does it still happen when the laptop is not plugged in to the mains?

Lockzi, 2018-09-30:
Yes, it happens when running on battery. 

I'm glad you have seen the issue before, maybe it's a simple fix then.

Lockzi, 2018-09-30:
Wow!

Absolutely amazing!

It seems that CLOVER still injects kexts even if you rename it from **.kext** to **.kextbackup**!
After I got it removed, it is no longer teleporting, but rather freezing for 1 ms every second. It is however usable now compared to the teleporting.

alexandred, 2018-09-30:
What do you mean by "freezing for 1ms every second" ?

Lockzi, 2018-09-30:
[Freezing mouse screenrecording.mov.zip](https://github.com/alexandred/VoodooI2C/files/2432065/Freezing.mouse.screenrecording.mov.zip)

It is now much more subtle, but it's the 1ms freeze that "breakes the circle" and moves the center point so to speak.

alexandred, 2018-09-30:
Yes this is now what I experience and I am not sure what the cause of the issue is. I'm glad to hear its now usable for you though.

alexandred, 2018-09-30:
I will rename this issue again to reflect the freezing and leave it open until we figure out what the cause of it is.

Lockzi, 2018-09-30:
Very good! Let me know if I can be of more assistant with troubleshooting.

Since you're running the same hardware.
Can you recommend an alternative to ACPIBatteryManager that works well with your voodooi2c so that I can see the battery remaining?

Awesome work you're doing here. Many thanks!

alexandred, 2018-09-30:
I simply followed RehabMan’s battery guide. I installed macOS myself and didn’t follow any of the XPS guides so not sure what they recommend.

Lockzi, 2018-10-06:
@Jern97

I went with Ben9923s suggested VirtualSMC. It works well!

Jern97, 2018-10-06:
I'm having a hard time getting it to work. I wiped my cache, added VirtualSMC and SMCBatteryManager to my kext folder and updated to Rehabman's fork of Clover (Pre-Built). 
Is there anything i'm missing?

Lockzi, 2018-10-07:
You can not also use FakeSMC

I would follow Ben9923s suggestion and read through the documentation for virtualSMC to see if there's anything you've missed. 

Lilu needs to be at least 1.2.6

I did not update to RehabMan's clover fork.

Lockzi, 2018-11-08:
Sorry to open this one up again,

but I updated (airportbrcmfixup, applealc, bt4lecontiunityfixup, fakepciid, lilu, notouchid, nullethernet virtualsmc, voodoops2) all my kexts, and upgrading to 10.14 Mojave and now the cursor is acting the same except every 2 seconds...

Any ideas? Downgrading VirtualSMC?

alexandred, 2018-11-09:
I have the same setup and I'm not facing any issues here. You could try downgrading VirtualSMC, yeah.

Lockzi, 2018-11-09:
Apparently it was the videoplayer running on Insanelymac thread that caused the stuttering. Took a while to figure out the source. Sorry for opening it up again...

ThatsNiceGuy, 2019-07-24:
Sorry for maybe opening this again but I’m having the same issue on a Dell G7 7588. Using Voodoo I2C and the VoodooI2CHID satellite. No third-party software has been installed on my hack and I’m confused as to what’s causing this.

ben9923, 2019-07-24:
> Sorry for maybe opening this again but I’m having the same issue on a Dell G7 7588. Using Voodoo I2C and the VoodooI2CHID satellite. No third-party software has been installed on my hack and I’m confused as to what’s causing this. On macOS 10.14

@ThatsNiceGuy Read the previous comments, it's usually ACPIBatteryManager. Using SMCBatteryManager instead resolves the issue.
If you're already using SMCBatteryManager (or it doesn't help), upload a full troubleshooting archive to the Gitter support chat:
https://gitter.im/alexandred/VoodooI2C