VoodooI2C/VoodooI2C #115: Mutitouch for touch screen not work well。

jieeraoyun, 2018-09-30:
My device is lenove miix 520

I have installed Mac Os10.14,and then I use your voodooi2c to drive the touch pad and touch screen(only placing VoodooI2C.kext and VoodooI2CHID.kext to Clover/kext/other/)

I found that the touch pad and touch screen works well except mutitouch for touch screen.

When i touch screen with 2 fingers,it not perform any thing.
When i touch screen with 3 fingers,it performs like 2 fingers for touch pad,and very slowly.
When i touch screen whith 4 fingers,it perform like 3 fingers and settings in mac for 3 fingers gesture can response.

could you please fix this。

acai66, 2018-09-30:
hello,I'm using miix 520 now,and i also meet this problem. my touchpad cant works well,could you share your efi?

alexandred, 2018-09-30:
This is likely because your touchscreen has stylus support - it should be fixed in a future release.

jieeraoyun, 2018-10-01:
It also has a problem is that we must removing the following key from VoodooI2CHID.kext/Info.plist if we want the touch pad and touch screen work together.

If not removing this,for mac 10.13,only the touch screen can be drivered.For mac 10.14 only the touch pad can be drivered.

<key>VoodooI2CHIDDevice Multitouch HID Event Driver</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>$(PRODUCT_BUNDLE_IDENTIFIER)</string>
			<key>DeviceUsagePairs</key>
			<array>
				<dict>
					<key>DeviceUsage</key>
					<integer>4</integer>
					<key>DeviceUsagePage</key>
					<integer>13</integer>
				</dict>
				<dict>
					<key>DeviceUsage</key>
					<integer>5</integer>
					<key>DeviceUsagePage</key>
					<integer>13</integer>
				</dict>
				<dict>
					<key>DeviceUsage</key>
					<integer>2</integer>
					<key>DeviceUsagePage</key>
					<integer>13</integer>
				</dict>
			</array>
			<key>IOProbeScore</key>
			<integer>200</integer>
			<key>IOClass</key>
			<string>VoodooI2CMultitouchHIDEventDriver</string>
			<key>IOProviderClass</key>
			<string>IOHIDInterface</string>
		</dict>

acai66, 2018-10-01:
I konw you,你是远景上分享miix 520的老哥.

gongzhenjeef, 2018-10-08:
> It also has a problem is that we must removing the following key from VoodooI2CHID.kext/Info.plist if we want the touch pad and touch screen work together.
> 
> If not removing this,for mac 10.13,only the touch screen can be drivered.For mac 10.14 only the touch pad can be drivered.
> 
> VoodooI2CHIDDevice Multitouch HID Event Driver
> 
> CFBundleIdentifier
> $(PRODUCT_BUNDLE_IDENTIFIER)
> DeviceUsagePairs
> 
> 
> DeviceUsage
> 4
> DeviceUsagePage
> 13
> 
> 
> DeviceUsage
> 5
> DeviceUsagePage
> 13
> 
> 
> DeviceUsage
> 2
> DeviceUsagePage
> 13
> 
> 
> IOProbeScore
> 200
> IOClass
> VoodooI2CMultitouchHIDEventDriver
> IOProviderClass
> IOHIDInterface

It's because your touchpad is switched to PTP mode while the driver can't recognize it well. You can also add following like "VoodooI2CHIDDevice Generic Mouse HID Event Driver" for "VoodooI2CHIDDevice Multitouch HID Event Driver" to prevent that.

`			<key>IOPropertyMatch</key>
			<dict>
				<key>Transport</key>
				<string>I2C</string>
			</dict>
`

jieeraoyun, 2019-03-17:
> > It also has a problem is that we must removing the following key from VoodooI2CHID.kext/Info.plist if we want the touch pad and touch screen work together.
> > If not removing this,for mac 10.13,only the touch screen can be drivered.For mac 10.14 only the touch pad can be drivered.
> > VoodooI2CHIDDevice Multitouch HID Event Driver
> > CFBundleIdentifier
> > $(PRODUCT_BUNDLE_IDENTIFIER)
> > DeviceUsagePairs
> > DeviceUsage
> > 4
> > DeviceUsagePage
> > 13
> > DeviceUsage
> > 5
> > DeviceUsagePage
> > 13
> > DeviceUsage
> > 2
> > DeviceUsagePage
> > 13
> > IOProbeScore
> > 200
> > IOClass
> > VoodooI2CMultitouchHIDEventDriver
> > IOProviderClass
> > IOHIDInterface
> 
> Hi, do you have any updates?
> In the latest VoodooI2CHID commits, USB device support is dropped and maybe your touchpad won't work any more via I2C connection.

hello,
I have tested the latest code.My touch pad can also work well with the latest VoodooI2CHID commits.
The muti-touch screen  issue is still present.

JayMonkey, 2019-04-04:
Hi Guys,

Any update on the touchscreen multitouch issue detailed in the first post of this issue report ?

I have 2017 HP Spectre X360 and i also have to use three fingers to scroll using the touchscreen where as in the past it only needed two like when using the touchpad.

The touchpad on my laptop is a Synaptics PS2 device and am using rehabman's VoodooPS2Controller kext, so VoodooI2C is only being used for the touchscreen and active pen/stylus.

Cheers
Jay

ben9923, 2019-04-04:
> Hi Guys,
>
> Any update on the touchscreen multitouch issue detailed in the first post of this issue report ?
>
> I have 2017 HP Spectre X360 and i also have to use three fingers to scroll using the [...]

Have a look at #167 and try compiling the code from @blankmac 's fork.

JayMonkey, 2019-04-04:
> Have a look at #167 and try compiling the code from @blankmac 's fork.

@ben9923,

Many Thanks for the tip will give it a go when i get home ...

Cheers
Jay

jieeraoyun, 2019-04-04:
I have test the patch #167,It works well with my Miix 520.
Many thanks to blankmac!

alexandred, 2019-04-06:
This is now fixed in the latest upstream. You will have to wait for an official release.

closed 2019-04-06 by alexandred

JayMonkey, 2019-04-06:
Thanks Alex

look forward to the next official release ....

Cheers
Jay

spurd0, 2020-01-08:
Hi there.
Got lenovo L380 with 10.13 and len2130 elan touchpad. Can't get multitouch working. 
Here is my fixed [Info.plist](https://github.com/alexandred/VoodooI2C/files/4035518/Info.plist.txt) from VoodooI2CHID.kext.
I didn't disable Apple kext because as i know i don't need to do it until 10.14.

Any ideas whats wrong?

rE8007, 2024-09-12:
I need help with a touch screen problem on my HP elite x2 g1. All of the multi touch functions work but the top half of my screen doesn't seem to be recognized. I have attached the computer specs as well as a video showing the problem I am having. It works fine with the track pad and I didn't have this issue when windows was installed. Booting with OpenCore

https://github.com/user-attachments/assets/a13e4309-c57f-4bb2-9a8d-181745579867

[image]