VoodooI2C/VoodooI2C #116: ignore built-in trackpad when mouse is present not work

lollycc, 2018-10-01:
Hello
when I upgrade to 10.14 and use v2.1.4, the trackpad doesn't work with no mouse connected when this function is on.
it works fine in 10.13 with v2.0.3.

ben9923, 2018-10-01:
I assume you are using the VoodooI2CHID satellite.
Please make sure you have latest version of it too (Have a look at the info.plist of the satellite. ProcessUSBMouseStopsTrackpad and ProcessBluetoothMouseStopsTrackpad will be set to false with latest release).

lollycc, 2018-10-01:
I'm using the latest version of VoodooI2CHID
I have change VoodooI2CStylusHIDEventDriver.cpp to show attached_hid_pointer_devices->getCount() and ignore_all's status at system log and it seems no problem
if I don't connect the mouse, I can press the trackpad to click but can't move pointer and tap to click
when the mouse connected, even press the trackpad can't click.
already set ProcessUSBMouseStopsTrackpad and  ProcessBluetoothMouseStopsTrackpad true.

ben9923, 2018-10-01:
They should be set to `false`, as I've said. Trackpad input will _still_ be ignored when external mouse/trackpad is attached, when this option is enabled in SysPrefs.

lollycc, 2018-10-02:
I set them false but still not work when this option is enabled.
I put a code at VoodooI2CMultitouchHIDEventDriver::handleInterruptReport to show the ignore_all's value at log and confirmed it was false but just can't move the pointer with trackpad.

penghubingzhou, 2018-11-17:
> I set them false but still not work when this option is enabled.
> I put a code at VoodooI2CMultitouchHIDEventDriver::handleInterruptReport to show the ignore_all's value at log and confirmed it was false but just can't move the pointer with trackpad.

there's a temporary  solution for it. Remove other code  from function VoodooI2CMT2PointingWrapper::start and only keep this word:

``bool VoodooI2CMT2PointingWrapper::start(IOService *provider){

    
    return true;
}``

Then the mouse detection will work. Tested by @stevezhengshiqi   and me but still in test.

penghubingzhou, 2019-01-04:
@alexandred  this  bug  has  been  fixed  by @coolstar  now and  I  think  you  can  close this  issue now.

closed 2019-01-14 by alexandred