VoodooI2C/VoodooI2C #117: ELAN0651 stops responding randomly

yoasif, 2018-10-04:
Steps to reproduce:

1. Boot up
2. Use laptop/trackpad for a while 

What happens:

Trackpad randomly stops responding

Expected result:

Trackpad should work as long as the laptop is running. Attaching a debug log after the trackpad stops responding. 

[debug_32167.zip](https://github.com/alexandred/VoodooI2C/files/2444655/debug_32167.zip)

yoasif, 2018-10-04:
I found that sleeping the machine and waking it up causes the trackpad to respond again - here's a log after sleep/wake in the hopes that it is useful. 
[debug_13300.zip](https://github.com/alexandred/VoodooI2C/files/2444659/debug_13300.zip)

coolstar, 2018-11-02:
ELAN0651 works fine here. ELAN0651 requires interrupts to be set up correctly and for every interrupt to be followed by an I2C read (or it eventually stops responding until it is reset -- which sleep/wake does). Make sure you're using interrupts rather than polling.

closed 2018-11-02 by coolstar