VoodooI2C/VoodooI2C #118: All works like a charm , but there are some issues

giuligon, 2018-10-07:
Congrats to the developers, the drivers works very well, all multi-touch gestures work perfectly but there are two things that do not work: the single click , this is registered as double and the right click...
I want to report another issue but probably this is lack of Mojave's trackpad prefpane , I'm unable to change trackpad prefs... I must try with prefpane from Sierra or High Sierra, but I have not any installation of these working...

ben9923, 2018-10-07:
About the first issue - see #111 

Do you have battery status working? It's a requirement for prefpane to show.
No known prefpane issue in Mojave. Make sure you revert the prefpane back to native from Mojave.

giuligon, 2018-10-08:
Hi Ben thanks you fo the answer, yes I saw  #111, but I'm still unable to fix the issue. No I haven't battery status working because i can't fix DSDT due Maciasl founding error on native DSDT even if I take it from Windows.... Can I try with a dummy kext for battery ?

giuligon, 2018-10-08:
Hi ben thanks very much , I didn't know that requirement ! I installed a kext for battery status , even if it does not report the percent but let me to use the prefpane ! Now I have no issues : instead clicking the button I can simply tap my trackpad ! Thanks again :)

closed 2018-10-08 by giuligon