VoodooI2C/VoodooI2C #119: ELAN0618 Right Click not working

BesnikRrustemi, 2018-10-09:
Hi @alexandred 
I have a Lenovo V330-15IKB with a ELAN I2C Touchpad.
On IOReg it shows as ELAN0618, Device name "TPD0" 8086; 9d60

Since 2.0.3 and above my Right click button stopped working, the kexts that I use is VoodooI2C.kext + VoodooI2CELAN.kext
Gestures work fine and the rest.

However if i use VoodooI2C.kext + VoodooI2CHID.kext I do have perfectly working Left and Right click buttons but I have no more gestures & scrolling function + Touchpad not recognized on SysPrefs>TrackPad.
I do rely on gestures a lot & use mission control a lot, not to mention the lack of scroll...

So is right click which basically makes my laptop sort of touchpad-less.

Even on 2.0.3 and below I had to press the very edge & bottom sides of the touchpad (which touchpad surface can be pressed for left & right click physical buttons function) in order to have left & right click work, with VoodooI2CHID.kext the physical buttons work phenomenal but no scroll and gestures.

Shall I upload you the Logs & Problem Reporting files to have a look or it's an issue that you are already aware of, also if there's a known fix until an update, can you point out so I can make the change locally and build the kext as workaround for personal use until an update comes up with the fix ?.

Thanks for the wonderful work for the hackintosh community.

TomHeaven, 2018-10-15:
I have met the same problem after updating to VoodooI2C 2.1.4 on macOS Mojave.  My device is Elan1000 with PCI address 8086,a160. Everything works fine except right click (or two finger click) won't work.

dderrida, 2018-10-15:
I have the same problem on my Dell XPS13 9360 with VoodooI2C 2.14
Right click does not work. Evertyhing else works beautifully on this TrackPad now