VoodooI2C/VoodooI2C #120: The driver can't work in /L/E

foxni, 2018-10-15:
I have installed VoodooPS2Controller.kext ,VoodooI2C.kext and VoodooI2CHID.kext for keyboard and touchpad. They work well in clover kext directory , but after I moved they into the /L/E ,the touchpad can't work and keyboard still can works well.
some info may help:
XPS 9560 1080p with 10.14 Mojave installed.
Two clover patchs and one ssdt were applied.(change original _STA to XSTA , _CRS to XCRS and insert a new function).
I moved all kexts to /L/E before I use VoodooI2C drivers , they work well.

alexandred, 2018-10-15:
Did you refresh your kextcache with `sudo kextcache -i /` ?

alexandred, 2018-10-15:
You need to apply Coolstar's I2C patches to clover. You can find them in the main patch repos.

foxni, 2018-10-17:
Thank you, it works.
Anyone who needs the patch can them here: [https://github.com/RehabMan/OS-X-Clover-Laptop-Config/commit/fca827f4acc8d569988895cc9f11697e64a89e0b](url)  RehabMan's OS-X-Clover-Laptop-Config

closed 2018-10-17 by foxni