VoodooI2C/VoodooI2C #122: wird memory high

satcity, 2018-10-19:
I have a dell precision3530, the touchpad is pci8086,a369, which is not in the support list.

But I add this the device ID to the plist of VoodooI2C.kext and put this kext, as well as VoodooI2CHID.kext to other folder of clover EFI, and reboot, the touchpad is working well, 2 finger and multi-finger gestures work well.

but after awhile, no more than 1 hour, the **wired memory** as showed in the activity monitor is increasing, when it reaches about 5GB+, the system broken. 

if I removed this 2 kexts from other folder, the wired memory stay about 1GB+, so I think if there is any bug within the drive caused the kernel memory leak, or is it because the compatibility with my notebook?

please have a look at this issue, it's really convenient to have a working touchpad, and I'm willing to help testing the new kext, thanks.

jafd, 2018-10-28:
I can confirm that on a Dell XPS13 9360, the trackpad not being used at all still causes a kernel panic due to a memory leak after mere few hours of uptime. Which makes me think that this is due to polling somehow (the driver complains about reverting to polling because interrupts cannot be used).

alexandred, 2018-10-28:
This does sound like a polling issue (OP has coffelake controller IDs so would only be able to use polling for the moment). Unfortunately polling is really only a temporary measure while either the user takes the time to implement the patches for interrupts or they wait for better support.

@jafd unless your Dell is Coffeelake, it should support interrupts fine. Have you tried going through the full installation guide?

jafd, 2018-10-28:
Probably not. I took ready-made files from someone else and just trusted they would work. Which, granted, they do.

But here's an interesting thing:
https://github.com/alexandred/VoodooI2CHID/blob/master/VoodooI2CHID/VoodooI2CHIDDevice.cpp#L175 does an `IOMalloc()` for which I don't see a corresponding `IOFree()`
Same here: https://github.com/alexandred/VoodooI2CHID/blob/master/VoodooI2CHID/VoodooI2CHIDDevice.cpp#L368
Could it be related?

FWIW if there is indeed a memory leak, switching to interrupts would only offset it, as I can see that the code paths are the same — the timer invokes the interrupt handler.

alexandred, 2018-10-28:
@jafd All VoodooI2C installs should really be done through the installation guide as opposed to using files from guides. I understand why it can be easy to trust such guides but, as you've now pointed out, many of the guides do not implement things properly. Polling is a last-resort option and the installation guide points this out. I would recommend you do a clean install of VoodooI2C (and its related files/patches) by following the guide. Polling mode will inevitably lead to higher CPU and memory usage by its nature.

In any case, the lines you point out are weird and, as far as I can tell, there should be a corresponding `IOFree` on those buffers. I am not quite sure why they are none there. I will investigate it for the next release.

jafd, 2018-10-28:
Preliminary testing shows that while the growth of wired memory slowed after recompiling and putting in new extensions, it hadn't stopped, unfortunately. I'll try to see if not loading VoodooI2C entirely would make any difference.

alexandred, 2019-01-14:
Is this issue still present on your system with the latest upstream?

penghubingzhou, 2019-03-05:
I noticed you  referenced it  in #135. Pls try the  new  commit and  test it.

jafd, 2019-03-22:
I can confirm that with the newer version the wired memory no longer grows, other things being equal. I cannot say that _this specific commit_ fixed it. Just that a newer release fixed it.

shrinivasshah, 2019-06-04:
hi, I have ASUS TUF FX504GE, i5 8300H, with ELAN1200 Precision TouchPad (pci8086,a368), I am unable to use my trackpad.

whenever I paste VoodooI2C + VoodooI2CHID + VoodooI2CUPDDEngine into EFI, with proper edited DSDT by author, My laptop keeps restarting
I followed this guide:
https://github.com/PoomSmart/ASUS-FX504GE-Hackintosh

closed 2020-04-29 by ben9923