VoodooI2C/VoodooI2C #125: VoodooI2CHID Doesn't Load After Update

Andrw0830, 2018-10-24:
Anytime I install an OS Update, after booting up the system, the VoodooI2CHID.kext doesn't load and I have to rebuild kext cache with Kext Utility. I have both VoodooI2C.kext and VoodooI2CHID.kext in EFI/Clover/kexts/Other. With kextstat, it does show VoodooI2C.kext load during the first boot, but the other kext does not. This has happened with the last few releases of VoodooI2C I've tried and don't have the issue with VoodooPS2Controller. I am using 10.14 on a Dell Inspiron 7567.

Let me know if you need anything else.

alexandred, 2018-10-25:
Can you provide a troubleshooting archive for when VoodooI2CHID.kext doesn't load?

Andrw0830, 2018-11-05:
Sorry for not getting back sooner. What will you need for the troubleshooting archive? Is there a command I can run? Do you need my Clover package? Please let me know.

alexandred, 2018-11-05:
See here: https://voodooi2c.github.io/#Troubleshooting/Troubleshooting

Andrw0830, 2018-11-06:
Here is my attached files per your link. The IORegistry is when VoodooI2C is working so let me know if you need one when it doesn't work as will have to wait for the next update to do so.

[Archive.zip](https://github.com/alexandred/VoodooI2C/files/2551414/Archive.zip)

alexandred, 2019-01-14:
Does this issue still occur if VoodooI2C and the satellites are installed in /L/E (with the appropriate Clover patch of course).

Andrw0830, 2019-01-25:
Hey man sorry about the delay, yep that worked! Is there a way to make it work without keeping it in /L/E similar to how VoodooPS2Controller works? I know if I ever have to boot into Recovery or reinstall Mac OSX, the same issue will happen.

darkhandz, 2019-02-08:
@Andrw0830 same situation as you

penghubingzhou, 2019-02-08:
> @Andrw0830 same situation as you

你试过重建缓存没？

stevezhengshiqi, 2019-02-08:
After some tests, I think I find a workaround for this issue.
From the log given by @Andrw0830, we can see 
`Kext com.alexandred.VoodooI2CHID - library kext com.apple.iokit.IOGraphicsFamily not found.`

So I add the following code in `config.plist` to force loading IOGraphicsFamily:
`<key>ForceKextsToLoad</key>
		<array>
			<string>\System\Library\Extensions\IOGraphicsFamily.kext</string>
		</array>`
under `<key>KernelAndKextPatches</key>`

It works on my device, not sure for other people's. It is still recommended to rebuild kextcache after update to let the kext automatically load.

darkhandz, 2019-02-09:
@stevezhengshiqi How did you test it? I add `IOGraphicsFamily` item to ForceKextsToLoad, then reboot into Recovery mode, touchpad still not work.

stevezhengshiqi, 2019-02-09:
@darkhandz I installed a fresh new system to see whether it works. Recovery mode is not tested, maybe it work in a different mechanism.
What I have tested is Mojave 10.14.3. At the language choose page, the trackpad is working with the patch; without the patch, the trackpad fails to load.

darkhandz, 2019-02-09:
@stevezhengshiqi I'll try it when next beta update available. 😀

Andrw0830, 2019-02-09:
Hey @stevezhengshiqi great find! I will test that too during the next update

penghubingzhou, 2019-02-09:
> After some tests, I think I find a workaround for this issue.
> From the log given by @Andrw0830, we can see
> `Kext com.alexandred.VoodooI2CHID - library kext com.apple.iokit.IOGraphicsFamily not [...]

这个方法看起来不错，不过似乎只是权宜之计。

Mateo1234454545, 2019-02-10:
> After some tests, I think I find a workaround for this issue.
> From the log given by @Andrw0830, we can see
> `Kext com.alexandred.VoodooI2CHID - library kext com.apple.iokit.IOGraphicsFamily not [...]

Working great !! Even on clean install and recovery environment.

Andrw0830, 2019-02-10:
> Working great !! Even on clean install and recovery environment.

It's weird I tried this by adding the ForeKextsToLoad under 'KernelandKextsPatches' and VoodooI2C and VoodooI2CHID is now in /Clover/kexts/Other, but recovery partition didn't make it work. Is your InjectKexts on Yes or Detect? Mine is Detect so wonder if the Recovery partition is just not thinking it's needed to be injected.

penghubingzhou, 2019-02-12:
> It's weird I tried this by adding the ForeKextsToLoad under 'KernelandKextsPatches' and VoodooI2C and VoodooI2CHID is now in /Clover/kexts/Other, but recovery partition didn't make it work. Is your [...]

I  think  rec  mode may  not  allow  injecting  iographicfamily because it  actually  don't  load   in  rec  mode.

Andrw0830, 2019-02-12:
> I think rec mode may not allow injecting iographicfamily because it actually don't load in rec mode.

That must be it. I just did the 10.14.3 supplement update and VoodooI2CHID did load and the trackpad worked. Thanks everyone for the suggestion!

Andrw0830, 2019-07-05:
Hey everyone, I was wondering if anyone found a solution to using VoodooI2C if you need to boot up in recovery mode or have to reinstall Mac using a USB Installer? With the solution to force IOGraphicsFamily to load, doesn't seem to work in recovery or a USB installer. I read that VoodooI2CHID supports both polling and interrupts mode. According to the documentation, polling should be used when booting in recovery. If we are already patched for the GPIO Interrupts method, would that be why the touchpad doesn't work in recovery and is there a solution without reverting the patches we applied to get GPIO working for the times we do need to boot in recovery? Basically if VoodooI2C can automatically enable Polling for times when it knows you are booting in recovery and use GPIO Interrupts when booting normally, not sure if that's possible? Just wanted to ask if anyone has gotten VoodooI2C working in both recovery and when booting normally?

Let me know if you need anything else?

gongzhenjeef, 2019-07-25:
> Hey everyone, I was wondering if anyone found a solution to using VoodooI2C if you need to boot up in recovery mode or have to reinstall Mac using a USB Installer? With the solution to force [...]

I think VoodooI2C actually load flawlessly. If you checked the IOReg, you will find nothing's attached to VoodooI2C. The point we discussed is that the dependency of VoodooI2C HID, IOGraphicsFamily, not present in recovery and also the first boot after update. Since the kextcache is rebuilt at that time, and VoodooI2CHID are loaded by bootloader before that.

If we track the commits regarding IOGraphicsFamily, this dependency is introduced with auto rotation feature. IMHO, maybe separate VoodooI2CSensors into a satellite kext and have that dependency moved there could solved this issue?

onemanosx, 2019-08-30:
> After some tests, I think I find a workaround for this issue.
> From the log given by @Andrw0830, we can see
> `Kext com.alexandred.VoodooI2CHID - library kext com.apple.iokit.IOGraphicsFamily not [...]

This works for me as well (tested on Catalina Beta 7)! Thank you!

closed 2020-04-06 by kprinssu