VoodooI2C/VoodooI2C #128: Not working on macOS Mojave 10.14.1 Update

BesnikRrustemi, 2018-10-31:
Hi @alexandred,

MacOS Mojave 10.14.1 update was released yesterday which VoodooI2C isn't working after update.

Any approximate ETA or known workaround that we can do on our own until you have the time and the update is ready ?

hieplpvip, 2018-10-31:
Rebuild kextcache

BesnikRrustemi, 2018-10-31:
@hieplpvip it wasn't working till latest public beta of macOS Mojave 10.14.1, kextcache is a nobrainer thing i did that already, but it didn't work for me, is it working on public release of 10.14.1 for you ? have you updated yet ?

hieplpvip, 2018-10-31:
I haven't updated yet. Will do it soon

darkhandz, 2018-11-01:
XPS 9550， 10.14.1 beta4 working fine. Update to 10.14.1 release:

```
Anonymous UUID:       BD838D05-146F-2A90-40D5-F5C700EC7110

Thu Nov  1 20:19:59 2018

*** Panic Report ***
panic(cpu 7 caller 0xffffff800e4d84dd): Kernel trap at 0xffffff800ea0d9df, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff800ef0ed2c, CR3: 0x0000000022df8000, CR4: 0x00000000003626e0
RAX: 0xffffff800ef82ec0, RBX: 0xffffff800eefed58, RCX: 0x0000000000000000, RDX: 0x0000000000000ffd
RSP: 0xffffff81fdc5bb20, RBP: 0xffffff81fdc5bb70, RSI: 0x0000000000000030, RDI: 0xffffff81fc9cdfa0
R8:  0x0000000000000031, R9:  0xffffff800ebee2f6, R10: 0xffffff800e200de0, R11: 0x0000000000000001
R12: 0x000000000000ffd0, R13: 0x0000000000000ffd, R14: 0xffffff803e942470, R15: 0xffffff81fc9cdf70
RFL: 0x0000000000010296, RIP: 0xffffff800ea0d9df, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff800ef0ed2c, Error code: 0x0000000000000000, Fault CPU: 0x7, PL: 0, VF: 2

Backtrace (CPU 7), Frame : Return Address
0xffffff81fdc5b5f0 : 0xffffff800e3aca9d 
0xffffff81fdc5b640 : 0xffffff800e4e6893 
0xffffff81fdc5b680 : 0xffffff800e4d82ba 
0xffffff81fdc5b6f0 : 0xffffff800e359ca0 
0xffffff81fdc5b710 : 0xffffff800e3ac4b7 
0xffffff81fdc5b830 : 0xffffff800e3ac303 
0xffffff81fdc5b8a0 : 0xffffff800e4d84dd 
0xffffff81fdc5ba10 : 0xffffff800e359ca0 
0xffffff81fdc5ba30 : 0xffffff800ea0d9df 
0xffffff81fdc5bb70 : 0xffffff800ea0ba5d 
0xffffff81fdc5bbc0 : 0xffffff800ea04a7b 
0xffffff81fdc5bc50 : 0xffffff800ea05579 
0xffffff81fdc5bcc0 : 0xffffff800e9d2be1 
0xffffff81fdc5bda0 : 0xffffff800e9d0aac 
0xffffff81fdc5be10 : 0xffffff800e9de658 
0xffffff81fdc5be60 : 0xffffff800ea4dc0b 
0xffffff81fdc5be80 : 0xffffff800ea2b8e1 
0xffffff81fdc5bf00 : 0xffffff800ea2b4c7 
0xffffff81fdc5bf50 : 0xffffff800ea2cf96 
0xffffff81fdc5bfa0 : 0xffffff800e3590ce 

BSD process name corresponding to current thread: Unknown
Boot args: dart=0 brcmfx-country=#a -lilubetaall -v keepyms=1 debug=0x100 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 18.2.0: Fri Oct  5 19:41:49 PDT 2018; root:xnu-4903.221.2~2/RELEASE_X86_64
Kernel UUID: 5D53F7E4-472A-369D-97D8-4DD877A4BDFF
Kernel slide:     0x000000000e000000
Kernel text base: 0xffffff800e200000
__HIB  text base: 0xffffff800e100000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)

System uptime in nanoseconds: 2603991244
last loaded kext at 2169785287: com.apple.driver.usb.AppleUSBHostPacketFilter	1.0 (addr 0xffffff7f8f577000, size 20480)
loaded kexts:
(none)

EOF
```

darkhandz, 2018-11-01:
[image]

[image]

darkhandz, 2018-11-01:
[image]

[image]

BesnikRrustemi, 2018-11-21:
@alexandred it is working for me now, the problem was clover related with the kernelandkextpatches used to disable Apple I2C drivers to load.
The problem seems to be clover related as at matchOS it doesn't detect:
10.13.x, 10.14.x anymore, after i just changed to 10.14.1 and reboot, the touchpad started working again.

BesnikRrustemi, 2019-01-04:
It was different problem, fixed by removing matchos on kernel and kext patches for the patches that disable applei2c kext.

closed 2019-01-04 by BesnikRrustemi

at-tantv, 2019-04-01:
> It was different problem, fixed by removing matchos on kernel and kext patches for the patches that disable applei2c kext.

How to remove matchos on keynal and kext patchs and disable applei2c kext

yamazaki45, 2019-08-15:
I have the same problem what I do? Why start from time to time kernel kernel, restarted and started 2 times to work or run with - v verbose enabled, in case it goes straight without crashing, can you help me this ??