VoodooI2C/VoodooI2C #132: Mouse cursor randomly jumps to top corner screen

cookievu, 2018-11-05:
My trackpad is Elan 1200, use VoodooI2C v2.1.4 & VoodooI2CHID on Mojave.
Everything works well, except for the cursor that usually jumps to the top corner of the screen.
Is there a way to solve this problem?

Jasper0819X, 2018-11-05:
same problem here with elan 1200 and mojave 10.14.1

alexandred, 2018-11-05:
Provide your troubleshooting archives as requested on the troubleshooting page: https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

cookievu, 2018-11-06:
This is my archive. Hope can find the cause.
https://drive.google.com/open?id=1gg5J52WiVGS0bRz1YS-D8mWRbpCgINy9

Jasper0819X, 2018-11-11:
@cookievu Do you have VoodooI2CELAN in your Clover's kexts folder?

coolstar, 2018-12-17:
Try commit ecabd7461c549f29a8a5b776a8b10041232dff25, if it's a scaling issue, should be fixed now.

andrewwyatt, 2018-12-18:
I built it before work today, and just now had some time to test.  There's a noticeable improvement in cursor speed and multi-touch input but unfortunately the cursor jumping to the right corner problem still persists for me.

hohoho123456, 2018-12-21:
i have the same problem with ELAN trackpad. other than that and 4-5 finger gesture support, it's perfectly working.

LeeBinder, 2019-01-03:
@coolstar : I'm having [two odd issues with 2.1.4]([url](https://github.com/alexandred/VoodooI2C/issues/108#issuecomment-451006810)), with both, current official alexandre release and also self compiled in 10.13.6 (even with 10.13.6 as target OS in the workspace Xcode serttings).

v. 2.0.3 is close to perfect on my Asus Vivobook with ELAN 1300 - performs even smoother than the Windows driver! The ONLY thing which annoys me in 2.0.3 is this random but frequent pointer jumping.

After reading what you wrote, I'm now wondering: is it possible for you to also optimize the scaling in the corresponding section of the 2.0.3 code?

Looking at all the issues ever since the engine change, it seems a worthwhile idea to not completely abandon 2.0.3, but rather to also implement those 2.1.4 fixes which are possible and helpful.

It would make my day if you could see if you can also fix the scaling in the 2.0.3 source code. I'd be more than happy to compile myself and report back :)

LeeBinder, 2019-01-03:
@Jasper0819X : I am not using VoodooI2CELAN because accdg. to documentation it is NOT for ELAN 1300.

LeeBinder, 2019-01-07:
No more random mouse pointer jumps for me with my ELAN 1300 since I followed these steps:

Step 1: On your Mac desktop, click the Apple logo, and select System Preferences.

Step 2: In the preferences pane, click Accessibility.

Step 3: On the left pane, scroll down to locate the Mouse & Trackpad tab, see that "Ignore built-in trackpad..." option? Check it.

Q: "BUT WAIT - DOESN'T THAT DISABLE MY TRACKPAD???"

A: It disables any ORIGINAL APPLE BUILT-IN TRACKPAD. But we don't have such, we are having PC Laptop touchpads + hackintosh drivers (kexts) which hook into Apple's macOS trackpad layer. That layer gets deactivated, but our drivers and the trackpad pref pane BOTH keep on working.

When the reason for YOUR random pointer jumps is that trackpad movements get routed through both, your h'tosh driver's layer PLUS Apple's layer, then disabling Apple's layer very likely can solve your pointer jumps.

GOOD LUCK! If it works for you, then pass this on across the web!

mathieupost, 2019-01-09:
Wow thanks!

Disabled the built-in trackpad option and FINALLY got it working now without random jumps!

mathieupost, 2019-01-10:
Not sure which version I had installed as it's gone now, but I'll try the newest version in combination with your scripts then ;)

alexandred, 2019-01-14:
I am going to tentatively close this issue as it appears not to be related to VoodooI2C and seems to be more an issue with macOS itself. If this issue still occurs despite the fix mentioned in this thread then please feel free to reopen.

closed 2019-01-14 by alexandred

alexandred, 2019-01-15:
@andrewwyatt Yes I've noticed that too and that's certainly a VoodooI2C issue (not really a bug as it stands as the code is doing what it is intended, it should just not consider a touchscreen as a mouse in the future). I will open a separate issue for that.

hohoho123456, 2019-02-06:
> No more random mouse pointer jumps for me with my ELAN 1300 since I followed these steps:
> 
> Step 1: On your Mac desktop, click the Apple logo, and select System Preferences.
> 
> Step 2: In the preferences pane, click Accessibility.
> 
> Step 3: On the left pane, scroll down to locate the Mouse & Trackpad tab, see that "Ignore built-in trackpad..." option? Check it.
> 
> Q: "BUT WAIT - DOESN'T THAT DISABLE MY TRACKPAD???"
> 
> A: It disables any ORIGINAL APPLE BUILT-IN TRACKPAD. But we don't have such, we are having PC Laptop touchpads + hackintosh drivers (kexts) which hook into Apple's macOS trackpad layer. That layer gets deactivated, but our drivers and the trackpad pref pane BOTH keep on working.
> 
> When the reason for YOUR random pointer jumps is that trackpad movements get routed through both, your h'tosh driver's layer PLUS Apple's layer, then disabling Apple's layer very likely can solve your pointer jumps.
> 
> GOOD LUCK! If it works for you, then pass this on across the web!
> 
> Credit: Andreas Louis for his [guide](https://www.anysoftwaretools.com/fix-mac-cursor-trackpad-problem/)

this did not work for me. when i enable "Ignore built-in trackpad..." my trackpad stoped working altogether even if i take out my wired mouse and wireless receiver.

xiqin, 2019-03-05:
Temporary fixbug
My English is very poor, please see the code.
[image]

LeeBinder, 2019-03-05:
Thank you xiqin. Can you please attach a zip with compiled kexts also, please?

xiqin, 2019-03-05:
[Release.zip](https://github.com/alexandred/VoodooI2C/files/2931226/Release.zip)
test os:  macos mojave 10.14.3
build:  https://github.com/alexandred/VoodooI2C/commit/b3416134fced8ef4226c6092ed17a59e17576caa

LeeBinder, 2019-03-05:
@xiqin : well done! Seems like the smoothest kexts so far for my Asus ELAN 1300 in 10.14.3!!!

Two questions:
1. Why do you consider this temporary not final? You think you just found a work-around not the real and final solution?

2. What Laptop and trackpad model do you have?

andrewwyatt, 2019-03-05:
@xiqin do you have this change in a fork somewhere?

xiqin, 2019-03-06:
@LeeBinder It sometimes destroys gestures.
I don't understand driver development and didn't really solve the problem.

@andrewwyatt Not commit

LeeBinder, 2019-03-07:
@xiqin : OK

What Laptop and trackpad model do you have?

xiqin, 2019-03-07:
Xiaomi mi Ruby 15.6 
I2C HID device

LeeBinder, 2019-03-07:
well all current touchpads are I2C HID devices .. ;) ELAN 1200? Or ELAN 1300? Or even Synaptics xyz? That's what I mean .. :)

xiqin, 2019-03-08:
I use VoodooI2CHID.kext, so you know？

LeeBinder, 2019-03-10:
> I use VoodooI2CHID.kext, so you know？

alexandred:
> Note that there is sometimes an overlap between device classes. For example, some ELAN devices may also be I2C-HID devices.

rockavoldy, 2019-03-16:
> [Release.zip](https://github.com/alexandred/VoodooI2C/files/2931226/Release.zip)
> test os: macos mojave 10.14.3
> build: [b341613](https://github.com/alexandred/VoodooI2C/commit/b3416134fced8ef4226c6092ed17a59e17576caa)

best solution for now, this fix the cursor random jump to top corner, after this 1 day use your kext, thanks

andrewwyatt, 2019-03-16:
Here's a patch based on the screenshot above, @alexandred this does appear to at least temporarily resolve the jumping cursor issue.  It also seems to resolve a click & drag issue I'd been having where whatever I'm dragging would arbitrarily release.

`diff --git a/Multitouch Support/Native/VoodooI2CMT2SimulatorDevice.cpp b/Multitouch Support/Native/VoodooI2CMT2SimulatorDevice.cpp
index c55652e..91f96a9 100644
--- a/Multitouch Support/Native/VoodooI2CMT2SimulatorDevice.cpp
+++ b/Multitouch Support/Native/VoodooI2CMT2SimulatorDevice.cpp
@@ -65,6 +65,7 @@ void VoodooI2CMT2SimulatorDevice::constructReportGated(VoodooI2CMultitouchEvent&
     // finger data
     int first_unknownbit = -1;
     bool input_active = false;
+    bool is_error_input_active = false;

     for (int i = 0; i < multitouch_event.contact_count + 1; i++) {
         VoodooI2CDigitiserTransducer* transducer = OSDynamicCast(VoodooI2CDigitiserTransducer, multitouch_event.transducers->getObject(i));
@@ -93,6 +94,10 @@ void VoodooI2CMT2SimulatorDevice::constructReportGated(VoodooI2CMultitouchEvent&

         IOFixed scaled_x = (((transducer->coordinates.x.value()) * 1.0f) / engine->interface->logical_max_x) * 7612;
         IOFixed scaled_y = (((transducer->coordinates.y.value()) * 1.0f) / engine->interface->logical_max_y) * 5065;
+
+        if (scaled_x < 1 && scaled_y > 5064) {
+            is_error_input_active = true;
+        }

         IOFixed scaled_old_x = (((transducer->coordinates.x.last.value)* 1.0f) / engine->interface->logical_max_x) * 7612;

@@ -211,9 +216,12 @@ void VoodooI2CMT2SimulatorDevice::constructReportGated(VoodooI2CMultitouchEvent&

     int total_report_len = (9 * multitouch_event.contact_count) + 12;
     IOBufferMemoryDescriptor* buffer_report = IOBufferMemoryDescriptor::inTaskWithOptions(kernel_task, 0, total_report_len);
-    buffer_report->writeBytes(0, &input_report, total_report_len);
-
-    handleReport(buffer_report, kIOHIDReportTypeInput);
+
+    if (! is_error_input_active) {
+      buffer_report->writeBytes(0, &input_report, total_report_len);
+      handleReport(buffer_report, kIOHIDReportTypeInput);
+    }
+
     buffer_report->release();
     buffer_report = NULL;
`

andrewwyatt, 2019-03-16:
[jumping_cursor.patch.gz](https://github.com/alexandred/VoodooI2C/files/2973777/jumping_cursor.patch.gz)

LeeBinder, 2019-03-16:
So there is progress, great. @andrewwyatt , could you also attach your current compile?

andrewwyatt, 2019-03-16:
@LeeBinder Sure, no problem.

[VoodooI2c-031619.zip](https://github.com/alexandred/VoodooI2C/files/2974409/VoodooI2c-031619.zip)

snackole, 2019-03-16:
Agreed. This is the best yet.

LeeBinder, 2019-03-17:
Thank you @andrewwyatt 
performing very smoothly (apart from the known brief delays after each copy etc.) in my Mojave test install. No trackpad function at all in High Sierra, I guess because you compiled them for Mojave?

andrewwyatt, 2019-03-17:
That's possible, I only have the Mojave SDK installed.

hotbrandon, 2019-04-05:
Thank you @andrewwyatt!
VoodooI2c-031619.zip solved my problem! Lenovo IdeaPad 330s 14" Mojave