VoodooI2C/VoodooI2C #133: Kernel Panic after Sleep

wmchris, 2018-11-07:
Had this issue multiple times, KP after waking from sleep with 2.1.4. Backtrace is always the same. Setup is the XPS9550.

```
*** Panic Report ***
panic(cpu 4 caller 0xffffff8009ad84dd): Kernel trap at 0xffffff800a01e5b9, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000000248ee000, CR4: 0x00000000003626e0
RAX: 0x0000000020000000, RBX: 0xffffff80454b6600, RCX: 0x0000000024000000, RDX: 0xffffff8042ad9e00
RSP: 0xffffff82001b3d40, RBP: 0xffffff82001b3d70, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0xffffff8045386200, R9:  0xffffff800a29ace0, R10: 0x0000000000000000, R11: 0xffffff800a29af58
R12: 0x0000000000000000, R13: 0xffffff8042964858, R14: 0xffffff8042ad9e00, R15: 0xffffff80454b6600
RFL: 0x0000000000010202, RIP: 0xffffff800a01e5b9, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffff82001b3810 : 0xffffff80099aca9d 
0xffffff82001b3860 : 0xffffff8009ae6893 
0xffffff82001b38a0 : 0xffffff8009ad82ba 
0xffffff82001b3910 : 0xffffff8009959ca0 
0xffffff82001b3930 : 0xffffff80099ac4b7 
0xffffff82001b3a50 : 0xffffff80099ac303 
0xffffff82001b3ac0 : 0xffffff8009ad84dd 
0xffffff82001b3c30 : 0xffffff8009959ca0 
0xffffff82001b3c50 : 0xffffff800a01e5b9 
0xffffff82001b3d70 : 0xffffff7f8f578d4c 
0xffffff82001b3d90 : 0xffffff800a02864a 
0xffffff82001b3de0 : 0xffffff800a02a8ab 
0xffffff82001b3e60 : 0xffffff7f8a8edb45 
0xffffff82001b3ea0 : 0xffffff7f8a8dfa94 
0xffffff82001b3ec0 : 0xffffff80099e8655 
0xffffff82001b3f40 : 0xffffff80099e8205 
0xffffff82001b3fa0 : 0xffffff80099590ce 
      Kernel Extensions in backtrace:
         com.alexandred.VoodooI2CServices(1.0)[409F4A01-3D35-32FF-B7E5-DB77CB7F52AF]@0xffffff7f8f578000->0xffffff7f8f57cfff
         com.apple.iokit.IOUSBHostFamily(1.2)[922E7520-229B-3C68-B232-A8727A20D44D]@0xffffff7f8a8c4000->0xffffff7f8a93efff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[A4F310BD-F588-34FF-A756-7D751A092119]@0xffffff7f8a8b4000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[3C048DDF-4E3F-3603-B95B-5580F5A93F06]@0xffffff7f8a8bc000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 agdpmod=vit9696 -v brcmfx-country=#a darkwake=no -lilubetaall kext-dev-mode=1 

Mac OS version:
18B75

Kernel version:
Darwin Kernel Version 18.2.0: Fri Oct  5 19:41:49 PDT 2018; root:xnu-4903.221.2~2/RELEASE_X86_64
Kernel UUID: 5D53F7E4-472A-369D-97D8-4DD877A4BDFF
Kernel slide:     0x0000000009600000
Kernel text base: 0xffffff8009800000
__HIB  text base: 0xffffff8009700000
```

wmchris, 2018-11-08:
Had this issue multiple times in a row, now after setting the boot flag it didnt appear anymore. I'll post it as soon as the KP will occur again.

wmchris, 2018-11-12:
```

Mon Nov 12 07:42:38 2018

*** Panic Report ***
panic(cpu 6 caller 0xffffff8009ad84dd): Kernel trap at 0xffffff800a01e5b9, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000000248ee000, CR4: 0x00000000003626e0
RAX: 0x0000000020000000, RBX: 0xffffff80457b5000, RCX: 0x0000000024000000, RDX: 0xffffff8042b4c9c0
RSP: 0xffffff8202eabd40, RBP: 0xffffff8202eabd70, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0xffffff80455ec840, R9:  0xffffff800a29ac20, R10: 0xffffff82008ad000, R11: 0xffffff800a29af58
R12: 0x0000000000000000, R13: 0xffffff804299d9d8, R14: 0xffffff8042b4c9c0, R15: 0xffffff80457b5000
RFL: 0x0000000000010206, RIP: 0xffffff800a01e5b9, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x6, PL: 0, VF: 1

Backtrace (CPU 6), Frame : Return Address
0xffffff8202eab810 : 0xffffff80099aca9d mach_kernel : _handle_debugger_trap + 0x48d
0xffffff8202eab860 : 0xffffff8009ae6893 mach_kernel : _kdp_i386_trap + 0x153
0xffffff8202eab8a0 : 0xffffff8009ad82ba mach_kernel : _kernel_trap + 0x4fa
0xffffff8202eab910 : 0xffffff8009959ca0 mach_kernel : _return_from_trap + 0xe0
0xffffff8202eab930 : 0xffffff80099ac4b7 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff8202eaba50 : 0xffffff80099ac303 mach_kernel : _panic + 0x63
0xffffff8202eabac0 : 0xffffff8009ad84dd mach_kernel : _kernel_trap + 0x71d
0xffffff8202eabc30 : 0xffffff8009959ca0 mach_kernel : _return_from_trap + 0xe0
0xffffff8202eabc50 : 0xffffff800a01e5b9 mach_kernel : __ZN15IORegistryEntry16detachFromParentEPS_PK15IORegistryPlane + 0x29
0xffffff8202eabd70 : 0xffffff7f8f578d4c com.alexandred.VoodooI2CServices : __ZN17VoodooI2CServices12detachDeviceEPvS0_P9IOServiceP10IONotifier + 0x32
0xffffff8202eabd90 : 0xffffff800a02864a mach_kernel : __ZN9IOService14invokeNotifierEP18_IOServiceNotifier + 0xea
0xffffff8202eabde0 : 0xffffff800a02a8ab mach_kernel : __ZN9IOService15terminatePhase1Ej + 0x4ab
0xffffff8202eabe60 : 0xffffff7f8a9b2b45 com.apple.iokit.IOUSBHostFamily : __ZN15IOUSBHostDevice9terminateEj + 0x77
0xffffff8202eabea0 : 0xffffff7f8a9a4a94 com.apple.iokit.IOUSBHostFamily : __ZN16AppleUSBHostPort26terminateServiceThreadCallEP9IOServiceP11thread_call + 0x18
0xffffff8202eabec0 : 0xffffff80099e8655 mach_kernel : _thread_call_delayed_timer + 0xec5
0xffffff8202eabf40 : 0xffffff80099e8205 mach_kernel : _thread_call_delayed_timer + 0xa75
0xffffff8202eabfa0 : 0xffffff80099590ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.alexandred.VoodooI2CServices(1.0)[409F4A01-3D35-32FF-B7E5-DB77CB7F52AF]@0xffffff7f8f578000->0xffffff7f8f57cfff
         com.apple.iokit.IOUSBHostFamily(1.2)[922E7520-229B-3C68-B232-A8727A20D44D]@0xffffff7f8a989000->0xffffff7f8aa03fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[A4F310BD-F588-34FF-A756-7D751A092119]@0xffffff7f8a979000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[3C048DDF-4E3F-3603-B95B-5580F5A93F06]@0xffffff7f8a981000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 agdpmod=vit9696 -v brcmfx-country=#a darkwake=no -lilubetaall kext-dev-mode=1 keepsyms=1 

Mac OS version:
18B75

Kernel version:
Darwin Kernel Version 18.2.0: Fri Oct  5 19:41:49 PDT 2018; root:xnu-4903.221.2~2/RELEASE_X86_64
Kernel UUID: 5D53F7E4-472A-369D-97D8-4DD877A4BDFF
Kernel slide:     0x0000000009600000
Kernel text base: 0xffffff8009800000
__HIB  text base: 0xffffff8009700000
System model name: MacBookPro9,1 (XXXXXXXX)

System uptime in nanoseconds: 5161134277491
last loaded kext at 4006899425507: com.apple.driver.AppleXsanScheme	3 (addr 0xffffff7f8f791000, size 32768)
last unloaded kext at 1133821741406: ru.usrsse2.SMCLightSensor	1 (addr 0xffffff7f8f5ea000, size 16384)
loaded kexts:
(none)

EOF
```

alexandred, 2018-11-13:
A temporary fix for this is to simply remove VoodooI2CServices.kext from VoodooI2C.kext/Contents/PlugIns. It is merely a helper tool for me to speed up debugging. I will look into what is causing it to crash soon.

tprimus83, 2018-11-14:
Hi Alexandred!

I have also problem with sleep. Sometimes I recognize, that the lid is closed, CPU running like hell, and can't wake it up. After reboot that's what I found as KP:
`*** Panic Report ***
panic(cpu 2 caller 0xffffff800a1feb31): "zalloc: zone map exhausted while allocating from zone kalloc.8192, likely due to memory leak in zone kalloc.256 (2491183104 total bytes, 9731174 elements allocated)"@/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-4903.221.2/osfmk/kern/zalloc.c:3415
Backtrace (CPU 2), Frame : Return Address
0xffffff81f912b150 : 0xffffff800a1aca9d 
0xffffff81f912b1a0 : 0xffffff800a2e6893 
0xffffff81f912b1e0 : 0xffffff800a2d82ba 
0xffffff81f912b250 : 0xffffff800a159ca0 
0xffffff81f912b270 : 0xffffff800a1ac4b7 
0xffffff81f912b390 : 0xffffff800a1ac303 
0xffffff81f912b400 : 0xffffff800a1feb31 
0xffffff81f912b530 : 0xffffff800a1b8ae4 
0xffffff81f912b5a0 : 0xffffff800a740678 
0xffffff81f912b8c0 : 0xffffff800a7402ed 
0xffffff81f912b970 : 0xffffff800a6aeda0 
0xffffff81f912bca0 : 0xffffff800a6b53e1 
0xffffff81f912bd20 : 0xffffff800a6b47d7 
0xffffff81f912bf40 : 0xffffff800a7b5aab 
0xffffff81f912bfa0 : 0xffffff800a15a466 

BSD process name corresponding to current thread: xpcproxy
Boot args: brcmfx-country=#a darkwake=0 debug=0x100 -v abm_firstpolldelay=6000 

Mac OS version:
18B75

Kernel version:
Darwin Kernel Version 18.2.0: Fri Oct  5 19:41:49 PDT 2018; root:xnu-4903.221.2~2/RELEASE_X86_64
Kernel UUID: 5D53F7E4-472A-369D-97D8-4DD877A4BDFF
Kernel slide:     0x0000000009e00000
Kernel text base: 0xffffff800a000000
__HIB  text base: 0xffffff8009f00000
System model name: MacBookPro1 (Mac-827FB448E656EC26)
...
Backtrace suspected of leaking: (outstanding bytes: 19712)
0xffffff800a1ff287 
0xffffff800a1b8ae4 
0xffffff800a7c62f7 
0xffffff800a85fb44 
0xffffff7f8d764256 
0xffffff800a858d45 
0xffffff7f8d763d58 
0xffffff7f8d765cc9 
0xffffff7f8d761bbd 
0xffffff7f8d7af11f 
0xffffff7f8afb4bf1 
0xffffff7f8afb3170 
0xffffff7f8d7b2211 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[ECDBC8A9-D637-3FEE-BDA3-94025A03B709]@0xffffff7f8afab000->0xffffff7f8b028fff
         com.alexandred.VoodooI2C(2.1.4)[E7230338-1AFE-3830-B753-09751D6BABA3]@0xffffff7f8d75b000->0xffffff7f8d788fff
            dependency: com.alexandred.VoodooI2CServices(1)[409F4A01-3D35-32FF-B7E5-DB77CB7F52AF]@0xffffff7f8d5a3000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[DC1D9290-BFFC-309B-8529-CCB7AB322B1C]@0xffffff7f8b03f000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[ECDBC8A9-D637-3FEE-BDA3-94025A03B709]@0xffffff7f8afab000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[7DE930EC-AB73-3F5A-9D38-94D5EE90D197]@0xffffff7f8aa95000
            dependency: org.coolstar.VoodooGPIO(1.1)[125807B8-4E2E-341B-AE46-F06983D562A3]@0xffffff7f8d74e000
         com.alexandred.VoodooI2CHID(1.0)[0C2FCF61-54DA-3FE5-99E3-B9E62A0AC8D2]@0xffffff7f8d7ac000->0xffffff7f8d7bcfff
            dependency: com.alexandred.VoodooI2C(2.1.4)[E7230338-1AFE-3830-B753-09751D6BABA3]@0xffffff7f8d75b000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[DC1D9290-BFFC-309B-8529-CCB7AB322B1C]@0xffffff7f8b03f000
            dependency: com.apple.iokit.IOGraphicsFamily(530.12)[9D1E5FA7-267E-397E-9C91-1751221A05CF]@0xffffff7f8b222000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[ECDBC8A9-D637-3FEE-BDA3-94025A03B709]@0xffffff7f8afab000

last loaded kext at 6878980733161: com.apple.driver.AppleUSBAudio	315.6 (addr 0xffffff7f8db03000, size 413696)
last unloaded kext at 3074237311551: com.apple.driver.AppleIntelLpssGspi	3.0.60 (addr 0xffffff7f8d789000, size 45056)
`
Is this the same, or should I open another issue for it?

Thank you in advance 
Peter

alexandred, 2018-11-14:
I can't tell unless you provide me with the KP with "debug symbols" enabled in Clover.

tprimus83, 2018-11-15:
Sorry, but I can't find that option, how could I enable it? -v was there.

alexandred, 2018-11-15:
It's literally called "Keep Debug Symbols". The actual command is `keepsyms=1`.

tprimus83, 2018-11-16:
Thank you! I've set the flag. Since was only one crash, but not mean sleep instead of meanwhile usage. I had to restart, but there wasn't any panic file generated. Now I enabled ssh, and I can check next time the logs mean freeze (hopefully). 
What I see is, that memory usage goes constantly up. And at freeze CPU running on 100% till I restart it. What should I do or check next time? Any ideas, what could cause it?

And what I forgot, is to thank you firstly for developing it! 👍

alexandred, 2019-01-14:
The original issue should be fixed with the latest upstream.

closed 2019-01-14 by alexandred