VoodooI2C/VoodooI2C #135: Kernel Panic after Using Laptop  For A  Long Time

penghubingzhou, 2018-11-21:
Hi, I got this issue:  If I use my laptop for a long time, my laptop will get suspected unless pressing power button  to reboot. I can ensure it is caused by Voodooi2c. First, after I removed the i2c kexts this issue disappeared. Second, I dumped a log after panic reboot. What caused that?
[paniclog.txt](https://github.com/alexandred/VoodooI2C/files/2604408/paniclog.txt)

AeriusOhara, 2018-11-27:
Perhaps I can add to this. I noticed kernel_task hoarding a bunch of ram pretty quickly to the point where it would cause the laptop to completely lock up. During which kernel_task was consistently eating anywhere near 10-15% of cpu.

So I used https://github.com/nickbruun/osxkerneltracer to see what was causing kernel_task to act up and it came back with this 
[image]

I went back and went through the GPIO pinning again and seemed to have default values. At some point I believe I started afresh and must have forgotten to redo that. After that kernel_task's cpu usage went down and VoodooI2C no longer even shows up in the list. 

However it was not only cpu issues though, I also saw through activity monitor that kernel_task's ram usage increased significantly when I used the trackpad vs when I just used the keyboard and or left the laptop idle.

After the GPIO pinning the ram usage has dropped significantly but the problem still remains. When I erratically move my mouse around using the trackpad kernel_task's memory usage goes up by small amounts. I tested various things to make sure it only happened while using the trackpad (no other apps running, cpu idling)

I might be wrong of course, I just thought I'd report what I've come across so far which might be correct or false assumptions. :)

alexandred, 2018-11-27:
I think your KP is unrelated to the original OP's one. Having the wrong GPIO pin can definitely lead to high CPU/RAM usage and it seems you fixed that by using the correct pin. The high CPU usage while touching the trackpad is known and I have no idea what is causing it. That being said, I had no idea the tool you linked existed and may very well help me in finding the root cause of the issue, thanks for bringing it to my attention!

AeriusOhara, 2018-11-27:
I'm glad it might help. I do wonder though, what about kernel_task's ram usage going up as you use the trackpad? It is still happening, just on a smaller scale. Or is that one of the things you hope to figure out?

The  kernel panic seems to happen when the ram gets full (and in Activity Monitor you can see the ram pressure become red), so I'm certain you're correct about VoodooI2C not being the cause.

Though I do wonder why the ram usage of kernel_task goes up as you use the trackpad, which I guess will be my only issue until I find and fix why a crash occurs when the ram gets full.

alexandred, 2018-11-27:
The high RAM usage should be fixed with the latest commits, a memory leak was pinpointed and patched. No ETA for its release yet though.

AeriusOhara, 2018-11-27:
Oh hey that's awesome, I assume I can just build the latest rev and use the resulting kext? Or is that risky to do with certain things not being fully implemented/missing since last release?

alexandred, 2018-11-27:
The latest commit should be as stable as v2.1.4

jafd, 2018-12-01:
Isn't that a duplicate of #122?

alexandred, 2018-12-01:
Yes I believe so and I think the latest commit addresses both these issues. I shall leave them open until the commit finds its way to a release.