VoodooI2C/VoodooI2C #138: Coffee Lake Support

williambj1, 2018-12-08:
Hello,
I really admire the kext you wrote.
When will the kext support Coffee Lake platform? 
Thank you very much!

eamigo86, 2018-12-24:
Hi, Any progress to implement support for Coffee Lake Intel cpu family? I have a Dell XPS 9570(i7 8750H with 4K touchscreen). Thanks in advance !!!

williambj1, 2018-12-30:
Now I'm using the newest build with the latest code change and I'm facing another problem😟.
After I enter my password in the login menu, the system will completely stop working with a screen of my user avatar and a half loaded stucked Dock. I waited for about 30 minutes and there was neither a panic nor an automatic reboot. I have no idea where to get a report for this issue or check the cpu usage from the system when it stucks.
I'm sure I patched my dsdt correctly. And when I disable Voodoohid.kext I can use the system without any problem except the touchpad.
Are there any tips?
Thank you very much!

briwil, 2019-01-03:
Does this mean you got it to work (somewhat) with Coffee lake?

williambj1, 2019-01-03:
No, not at all.
By the way, I figured out that when I leave the login menu for about 2 minutes and then I type in my password the system will work well.😵
I still can't let Voodooi2chid.kext load properly and the system doesn't detect my trackpad.

Here are some details of my device

- CPU: Intel Core i7 8750H
- Trackpad: ELAN 1404
- ACPI Pin: 0x5F
- GPI0 Pin: 0x47
- IONameMatch:  pci8086,a369 (Deleted a368 because there is another device which is a USB Type-C Controller using it and it causes a panic when VoodooGPIO tries to load it.)

Thanks a lot!

BesnikRrustemi, 2019-01-03:
@williambj1 Did you try the Notouchid.kext which is a Lilu Plugin. It might fix your delay.

williambj1, 2019-01-03:
> Hi, Any progress to implement support for Coffee Lake Intel cpu family? I have a Dell XPS 9570(i7 8750H with 4K touchscreen). Thanks in advance !!!

It seem that someone made XPS 9570's  trackpad work although it has still has some issues. 
[XPS9570.zip](https://github.com/alexandred/VoodooI2C/files/2723628/XPS9570.zip)

This is patched by a Chinese called penghubinzhou.

williambj1, 2019-01-09:
Add a -v image
[图片]

alexandred, 2019-01-09:
As I've said many times before, coffelake only works in polling mode at the moment. It will not work with GPIO interrupts.

williambj1, 2019-01-20:
Any news for CFL GPIO interrupts?

leobudima, 2019-05-25:
Does anyone know what the current state of Coffee Lake GPIO support is? Thanks!

williambj1, 2019-05-25:
> Does anyone know what the current state of Coffee Lake GPIO support is? Thanks!

It works perfectly, but not on devices with buggy GPIO.

williambj1, 2019-05-30:
Thanks for all of the Great work!!!!!

closed 2019-05-30 by williambj1