VoodooI2C/VoodooI2C #142: Two-finger right click only works with tap-to-click

b2550, 2018-12-29:
I am using VoodooI2CHID and I can only use right-click when I have tap-to-click enabled.

This is annoying because I constantly make accidental clicks with my trackpad when typing on my laptop. Since there is no good way to disable the trackpad while typing, I feel like adding right-click without tapping is the easier fix.

ech1965, 2018-12-31:
I have the very same issue ( XPS 9550) and fully agree to the proposed solution ;-)

alexandred, 2019-01-01:
This is addressed with the latest commits and will find its way into release soon.

Also it is not true that there is no way to disable the trackpad while typing. The latest VoodooI2C + the latest VoodooPS2 implements this fine.

closed 2019-01-01 by alexandred