VoodooI2C/VoodooI2C #144: VoodooI2C Sleep issuse v2.1.4 on ELAN1000 - ASUS GL552VW

baobaoit, 2019-01-08:
Hello @alexandred , I know that you had fixed sleep problem on v2.1.4, but on my laptop (on the title) it still not working after I used sleep on my laptop. Is there any solutions that I don't know so you/anyone can link in the comment bellow. Thank you.

Some information: that is when I put my laptop to sleep, then I close the lid of my laptop. After I open it, everything is worked fine, but the trackpad is not working. I have shutdown then turn on again my laptop and trackpad is work again.

penghubingzhou, 2019-01-13:
I found  a similar issue  in  ELAN1000 too. A friend in my QQ group  has a ELAN1000 device and his touchpad can only worked after wake from  sleep. @ben9923

baobaoit, 2019-01-14:
Sometimes it has stop working, I think it related with sleep of laptop. Yesterday, I had installed app SwitchResX. After I shutdown my laptop, I open it again then the trackpad is not working. I have seen that SwitchResX had a function that used sleep. Then I have uninstalled it but trackpad is not working. I had installed VoodooI2C and VoodooI2CELAN in LE and CKO. So I try delete from LE and restart my laptop. Finally, it is working.

penghubingzhou, 2019-01-15:
Helped  a  people  with  his elan1000 ，only  can  use  trackpad  after  sleep

baobaoit, 2019-02-04:
I’m tracking the VoodooI2C and I seeing that sometimes when I close the lid of my laptop for a few minutes then when I open it, the trackpad is not working. I have reboot my laptop.

baobaoit, 2019-02-04:
Now, I have put these kexts (I2C and I2CELAN) in LE and CKO. After I reboot many times but don't work, then I shutdown and turn on again. BOOM, it working.

penghubingzhou, 2019-02-04:
As  you  said， it's kext permission that  caused this issue.  But it  needs  more  test

alexandred, 2019-02-04:
I'm closing this issue since if its a permissions problem (which could have been easily diagnosed if OP provided their troubleshooting archives as requested in the first place).

closed 2019-02-04 by alexandred