VoodooI2C/VoodooI2C #145: Multitouch HID Event Driver Out of Date

vettz500, 2019-01-10:
I went ahead and built the latest kexts from the repo, but after installing the kexts I get the attached error upon booting. 
[image]

If I remove the 2CHID kext, I can boot fine, just no trackpad. I've rebuilt the kexts a few times now with the same issue. The 2CHID kext is what I've used in the past with no issue, just wanted to run the newer version to fix the memory leak issue and fix the obnoxious Kernel Task CPU usage I was having because of it.

alexandred, 2019-01-11:
What version of macOS are you running and building against?

vettz500, 2019-01-11:
> What version of macOS are you running and building against?
> […](#)

Running 10.13.6 build# 17G3025

I just realized that in Xcode the Deployment Target was set to 10.11. Could the be the cause?

alexandred, 2019-01-11:
Could be the target but also check the SDK that you’re building with.

vettz500, 2019-01-11:
> Could be the target but also check the SDK that you’re building with.
> […](#)

Just checked and for some reason it's set for sdk macosx10.14 
[image]

vettz500, 2019-01-15:
> And what about changing the SDK you build with in the actual build settings of VoodooI2C and its satellites?

Same issue. I even tested with the latest commits you applied to the Satellite kexts from yesterday.

ben9923, 2019-04-16:
This is probably an SDK issue, as @alexandred suggested.
Are you sure you didn't build with the 10.14 SDK? What's the value of `DTSDKName`in the output `VoodooI2C.kext/Contents/Info.plist`?
It seems like the 10.14 SDK is not backwards compatible to 10.13.6 and below. Try the latest [10.12/10.13 SDK](https://github.com/alexey-lysiuk/macos-sdk/releases) (you can build VoodooI2C and all of the satellites with the same SDK using the `-sdk` parameter of xcodebuild). Refer to the [CircleCI build script](https://github.com/alexandred/VoodooI2C/blob/master/.circleci/config.yml).

The latest v2.1.6, for example, was built with the macosx10.12 SDK. It should work fine on your system.

closed 2019-10-25 by ben9923