VoodooI2C/VoodooI2C #149: Small regression in commits (up to b341613) since last release (v2.1.4)

michel-m, 2019-01-18:
Hello,
First, thanks a lot for the good work, I've been using your kext since v1.0.0, it's great and keeps improving.
I've built it from source (up to b341613) yesterday to check if the last commits since the release (v2.1.4) improved the responsiveness (and they do), and I've noticed a small regression that I didn't see mentioned in other issues: dragging (one finger double tapping and then move around) broke along the way (I've made sure it is enabled in Accessibility settings).
I'm using VoodooI2C.kext along with VoodooI2CELAN.kext, for an ELAN1000, on Mojave 10.14.2.
Thanks again!

michel-m, 2019-01-18:
It works in v2.1.4, it doesn't anymore since then (one of the commits between v2.1.4 and b341613 must have broken it).

michel-m, 2019-04-10:
Erratum: upon reinstallation of my system, it (dragging) doesn't work in v2.1.4. It works well in v2.1.3 though.

andrewwyatt, 2019-04-10:
This was fixed in master a while back and just hasn't been released yet.

michel-m, 2019-04-10:
I've just compiled the kext from an up-to-date master, dragging doesn't work (all the rest does though).

michel-m, 2019-04-18:
Update: dragging is broken in v2.1.6 also (maybe it was broken in the ELAN satellite? I have no idea how to debug that).

quilenle, 2019-05-15:
I have use the lastest version of VoodooI2C and VoodooI2CELAN, and it (dragging) is broken. But in v2.1.4, it is working for me. ELAN1000.

michel-m, 2019-06-14:
Hello,
Dragging is still broken in 2.2 (lastest version it works for me: 2.1.3).
Using ELAN1000.

quilenle, 2019-06-23:
> Hello,
> Dragging is still broken in 2.2 (lastest version it works for me: 2.1.3).
> Using ELAN1000.

I think we just can use v2.1.3(4) :D

ben9923, 2019-06-23:
Note: Until v2.2 release, releases were problematic. They were overriden once new commits were pushed to master. 2.1.3 and 2.1.4 should be the same, ELAN-wise, but the 2.1.4 release you're seeing is about the same as 2.1.5.
Please try building the kexts yourself. Start by building https://github.com/alexandred/VoodooI2C/commit/06c2b3091615201bf946d1135176d6c6f4e09acb (The actual 2.1.4 release), then try newer and newer commits to find where it's exactly broken for you.

I think the only commit to VoodooI2CELAN since 2.1.4 release was https://github.com/kprinssu/VoodooI2CELAN/commit/8013479451095ddb50856d4112e643276d9a9611. Try locally reverting it and see if something is changed (I don't think it'll help, but...).

alexviean, 2019-06-28:
@alexandred I have the same issue. Same ELAN1000. Simple as this: 2.1.4 and prior works, 2.1.5 broken.

michel-m, 2019-06-28:
@alexandred @ben9923 Alright so after trying a lot of different commits (git clone, git checkout commit, git submodule update --recursive, etc, rince and repeat), I identified 37febae as the culprit for breaking dragging in my case (using ELAN1000, ELAN satellite).
Any idea how to fix this downstream?

ben9923, 2019-06-28:
> @alexandred @ben9923 Alright so after trying a lot of different commits (git clone, git checkout commit, git submodule update --recursive, etc, rince and repeat), I identified [...]

Are you sure it's that commit? I could not identify anything that could be related in this commit.
Does it properly work when checking https://github.com/alexandred/VoodooI2C/commit/3aa7d1d9f4e45a74ca937c0b15b8825a203a90e0 (the previous commit) out?
Please verify again, see if the commit I linked really works, and the one you linked does not :)

michel-m, 2019-06-29:
@ben9923 thanks for your answer. I'm positive, I've tried the previous commit and dragging does indeed work. Thank a lot for the time and effort you put into this kext.

red-green, 2019-07-10:
I'm having a very similar issue with dragging in versions beyond 2.1.4, however I have a Synaptics 1A586757 while most people in this thread seem to have issues with ELAN models. I use the VoodooI2CHID satellite on version 2.1.4 currently but at some point i would like to update to fix the OpenCore unload kp once the fix for #187 is merged.

kprinssu, 2020-05-15:
Please use the latest version of VoodooI2C. Closing as there haven't been any responses in a while. Please create a new issue if you experience any issues.

closed 2020-05-15 by kprinssu