VoodooI2C/VoodooI2C #150: Strange trackpad with latest release/commit ELAN1200 V2.1.4

giuligon, 2019-01-18:
First thanks to all contributors of this project!
Using the latest commit/release, my trackpad works very strangely: cursor and gesture are getting stuck for about a second then they restarts working for about, another, second. The gestures feel like "incomplete", half done, and furthermore, the Dock crashes almost all times.
I tried every commit and I encountered these problems starting e3c9727efb77e6a9bfc2be09230837ef1eedd6cb.
I attached the logs of the Dock and a video documenting "unresponsiveness" 
Thanks in advance for support! 
[Archivio .zip](https://github.com/alexandred/VoodooI2C/files/2773342/Archivio.zip)
Right now I'm using Mojave, ELAN1200 (VoodooI2C 2.1.4 + I2CHID)

valinor61, 2019-01-20:
I had similar issue with that kext I undo all of my dsdt changes and put that SSDT on acpi/patched now it works well
[SSDT-XOSI.zip](https://github.com/alexandred/VoodooI2C/files/2776143/SSDT-XOSI.zip)

giuligon, 2019-01-20:
@valinor61 Thanks a lot for answering and trying to provide a solution(previously I patched my DSDT applying the Windows10 Patch), but unfortunately I have the same issue unless I swap the newer version with an older one. Anyway, this problem seems to be related to Actuator and also to "Stuck finger fix"

giuligon, 2019-01-27:
@muhammadasadbastomi Hi, yes I can help you! Please, in order to do this, report properly your problem with a copy of IOReg and attach your EFI Folder.

alexandred, 2019-04-06:
Is this still an issue with the latest upstream?

giuligon, 2019-04-06:
@alexandred Hi Alexander, thank you for answering, unfortunately I'm still facing the same issue !

kprinssu, 2020-05-15:
Please use the latest version of VoodooI2C. Closing as there haven't been any responses in well over an year.

closed 2020-05-15 by kprinssu