VoodooI2C/VoodooI2C #151: Surface Pro 3 touch screen does not work ver.2.1.4 MacOS 10.13.6

rina-inverse, 2019-01-26:
Surface Pro 3 Core i 5 4300 U 1.9 GHz
VoodooI 2 C.kext VoodooI 2 CHID.kext ver 2.1.4
install to EFI / CLOVER / ketxs / Other
MacOS 10.13.6

Looking at IOReg, VoodooI2C.kext and VoodooI2CHID.kext seem to be loaded.
Trackpad settings are displayed in the environment settings.

The touch panel does not respond.

SysPrefs> Accessibility> Mouse & Trackpad> Ignore built-in ... is disabled.

rina-inverse, 2019-01-26:
files made by gen_debug.sh
[debug_3277.zip](https://github.com/alexandred/VoodooI2C/files/2798765/debug_3277.zip)

5high, 2019-02-24:
Hi，rina-inverse

did you patched your DSDT?

ittat, 2020-04-03:
> Hi，rina-inverse
> 
> did you patched your DSDT?
How to patch it ?

ben9923, 2020-04-05:
That issue was inactive for over a year...
Please re-open with updated troubleshooting files if it still happens with latest release.

closed 2020-04-05 by ben9923