VoodooI2C/VoodooI2C #153: Probabilistic Kernel

penghubingzhou, 2019-01-28:


penghubingzhou, 2019-01-29:
By  comparing  IOKIT and `` VoodooI2CControllerDriver.hpp```  I figured out  a  simple  way to solve it :  changing ```UInt16 address;``` to ```UInt32 address;``` .

penghubingzhou, 2019-01-29:
I believe ```messages[bus_device->message_write_index].address;``` should be a uint32 rather  than uint16

persistz, 2019-02-21:
+1 panic too

penghubingzhou, 2019-07-24:
@alexandred
 Sorry for so lang  reply, now I have tested in  OC and it will be  a  real "kernel panic". I  have no ideas for now.

penghubingzhou, 2019-10-25:
I think it's time to close this issue temporary because after I replace XOSI with this code, my panic disappeared:
```
Scope (\)
    {
        If (_OSI ("Darwin"))
        {
            Store (One, TPDD)
        }
    }
```
I still need more information about it to prove it's VoodooI2C's issue not a DSDT issue. So if anyone have sth useful pls contact me, thx!

closed 2019-10-25 by penghubingzhou