VoodooI2C/VoodooI2C #155: (Elan 1200) 5 fingers touch breaks click to drag/single click

overnightmare95, 2019-02-05:
I'm using VoodooI2C+VoodooHID (last commit b3416134fced8ef4226c6092ed17a59e17576caa) on a Asus N580VD with Elan 1200. Everything is working perfectly but when I accidentally use 5 fingers, one click drag/one click won't work as expected but after sleep/waking from sleep it works again. I have a ASUS buggy DSDT so my Touchpad is in polling mode.
IORegistryExplorer log and video attached.

[Bug video.zip](https://github.com/alexandred/VoodooI2C/files/2832263/Bug.video.zip)
[DSDT+IORegistryExplorerLog.zip](https://github.com/alexandred/VoodooI2C/files/2832265/DSDT%2BIORegistryExplorerLog.zip)

MaximumOverflow, 2019-02-06:
I'm having the same issue.
 I'm also running the kext in polling mode due to Asus's buggy DSDT.

stonexing, 2019-04-08:
I'm using Asus S4000VA is works on 10.14.3.
But it not work on 10.14.4.

ben9923, 2019-04-16:
Does it still happen with the latest release?

(I've noticed ASUS released some BIOS updates to many devices recently. Try updating if there's an available update for you, might solve the GPIO problem)

overnightmare95, 2019-04-17:
> Does it still happen with the latest release?
> 
> (I've noticed ASUS released some BIOS updates to many devices recently. Try updating if there's an available update for you, might solve the GPIO problem)

5 fingers still break click to drag in version 2.1.6

ben9923, 2019-04-17:
> > Does it still happen with the latest release?
> > (I've noticed ASUS released some BIOS updates to many devices recently. Try updating if there's an available update for you, might solve the GPIO problem)
> 
> 5 fingers still break click to drag in version 2.1.6

Try looking at the finger data recognized by macOS using [FingerMgmt](https://github.com/jnordberg/FingerMgmt).
See there are no stuck fingers or something else that could break some gestures.

overnightmare95, 2019-04-18:
> > > Does it still happen with the latest release?
> > > (I've noticed ASUS released some BIOS updates to many devices recently. Try updating if there's an available update for you, might solve the GPIO problem)
> > 
> > 
> > 5 fingers still break click to drag in version 2.1.6
> 
> Try looking at the finger data recognized by macOS using [FingerMgmt](https://github.com/jnordberg/FingerMgmt).
> See there are no stuck fingers or something else that could break some gestures.
> Make sure VoodooI2CHID is updated as well.

Yes, both VoodooI2C and VoodooI2CHID are updated to last commit. 
I'm uploading two videos with fingermgmt before and after the 5 finger touch. Sleep/Wake cycle seems to fix the bug until you accidentally use 5 fingers again. 
Thanks for your time
[Videos.zip](https://github.com/alexandred/VoodooI2C/files/3093816/Videos.zip)

Jie2GG, 2020-04-30:
> > > > 最新版本是否仍会发生？ （我注意到，ASUS最近发布了许多设备的一些 BIOS 更新。请尝试更新是否有可用的更新，可能会解决 GPIO 问题）
> > > 
> > > 
> > > 5 个手指仍然中断单击以拖动版本 2.1.6
> > 
> > 
> > 尝试查看 macOS 使用[FingerMgmt](https://github.com/jnordberg/FingerMgmt)识别的手指数据。看到没有卡住的手指或其他东西，可能会打破一些手势。确保巫术I2CHID也更新。
> 
> 是的，巫毒I2C和巫毒I2CHID都会更新为上次提交。
> 我在5指触摸前后上传两个手指视频。睡眠/唤醒周期似乎可以修复错误，直到您不小心再次使用 5 个手指。
> 感谢您的时间
> [视频.zip](https://github.com/alexandred/VoodooI2C/files/3093816/Videos.zip)

My Notebook is Asus S5300Fn, Touchpad is ELAN1200. I attempt using latest VoodooI2C And work in GPIO interrupt. Already solve 5 fingers touch this isuess.

kprinssu, 2020-05-15:
Please use the latest version of VoodooI2C and create a new issue if you experience any issues.

closed 2020-05-15 by kprinssu