VoodooI2C/VoodooI2C #156: ELAN 0608

rogerzero, 2019-02-11:
Gesture doesn't work on ELAN0608.
System Preference doesn't detect the touchpad.
Any solution?

penghubingzhou, 2019-02-12:
ELAN0608 can  be  dectected  in  syspref  with  i2celan.   But now  it has  bug.

rogerzero, 2019-02-12:
mine doesn't, should i do something?
what bug you referenced?

penghubingzhou, 2019-02-12:
You  need  to follow  the  gpio  pinning  guide  to  makesure  it run in  interrupt  mode  and  then  use i2celan  as satellite kext. The  bug  is it only  display in syspref but has  no  gesture  in  reality.

**来自GitHub小程序客户端：**

[image]

penghubingzhou, 2019-02-25:
Now  it's  prooved  that elan0608 can  work  with i2celan satellite. Lenovo Chao 5000 succeed.

rogerzero, 2019-02-25:
what do you mean by satellite?

penghubingzhou, 2019-02-25:
> what do you mean by satellite?. I did the gpio pinning, but yet, not detected in syspref but the touchpad is working (no gesture).

You should  use VoodooI2CELAN as satellite kext. See  voodooi2c.github.io  for  more  details.

closed 2019-03-05 by rogerzero