VoodooI2C/VoodooI2C #161: Touchscreen gesture direction not changed when rotated

gongzhenjeef, 2019-02-26:
After rotated automatically with VoodooI2CSensorHub, single point input on the touchscreen matches but the gesture direction is still in vertical. Eg, three finger (haven't fixed yet) can do scroll up and down by swipe left and right.

jieeraoyun, 2019-04-05:
@gongzhenjeef 

How do you achieve automatic screen rotation?

gongzhenjeef, 2019-04-05:
There's just a SHUB device under the I2C0 controller. VoodooI2C could recognize it and get some of the data. While I suppose there might be more data like brightness signal from the sensor hub.

alexandred, 2019-04-06:
> @gongzhenjeef
> 
> How do you achieve automatic screen rotation?This function is cool。
> 
> #115 issue has been soloved by blankmac。

Sensor Hub support is in beta and only works for a select few SHUBs.

alexandred, 2019-04-06:
Closed as this is a clone of https://github.com/alexandred/VoodooI2C/issues/148.

closed 2019-04-06 by alexandred