VoodooI2C/VoodooI2C #168: VoodooI2CAtmelMXTTouchDriver::init_device: configuration space crc mismatch | ATML1000

hieubq90, 2019-04-01:
```
2019-04-01 03:15:03.237926+0700 0x40e      Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOSunrisePointLP::Loading GPIO Data for SunrisePoint-LP
2019-04-01 03:15:03.237980+0700 0x40e      Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOSunrisePointLP::VoodooGPIO Init!
2019-04-01 03:15:03.237982+0700 0x40e      Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 0
2019-04-01 03:15:03.238021+0700 0x40e      Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 1
2019-04-01 03:15:03.238044+0700 0x40e      Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 2
2019-04-01 03:15:03.238251+0700 0x7f       Default     0x0                  0      0    kernel: (VoodooGPIO) VoodooGPIOSunrisePointLP::GPIO Controller is already awake! Not reinitializing.
2019-04-01 03:15:03.260456+0700 0x3fe      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,9d61 Starting I2C controller
2019-04-01 03:15:03.260499+0700 0x3fe      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
2019-04-01 03:15:03.260504+0700 0x3fd      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,9d60 Starting I2C controller
2019-04-01 03:15:03.260521+0700 0x3fd      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
2019-04-01 03:15:03.260549+0700 0x3fd      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,9d60 Publishing nub
2019-04-01 03:15:03.260566+0700 0x3fe      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CPCIController::pci8086,9d61 Publishing nub
2019-04-01 03:15:03.263700+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Probing controller
2019-04-01 03:15:03.263707+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Found valid Synopsys component, continuing with initialisation
2019-04-01 03:15:03.263744+0700 0x42d      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Probing controller
2019-04-01 03:15:03.263749+0700 0x42d      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Found valid Synopsys component, continuing with initialisation
2019-04-01 03:15:03.263808+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,9d60 SSCN not implemented in ACPI tables
2019-04-01 03:15:03.263815+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,9d60 FMCN not implemented in ACPI tables
2019-04-01 03:15:03.263818+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Warning: Error getting bus config, using defaults where necessary
2019-04-01 03:15:03.263835+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Publishing device nubs
2019-04-01 03:15:03.263847+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Found I2C device: ATML1000
2019-04-01 03:15:03.263852+0700 0x42d      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,9d61 SSCN not implemented in ACPI tables
2019-04-01 03:15:03.263860+0700 0x42d      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,9d61 FMCN not implemented in ACPI tables
2019-04-01 03:15:03.263864+0700 0x42d      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Warning: Error getting bus config, using defaults where necessary
2019-04-01 03:15:03.263880+0700 0x42d      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Publishing device nubs
2019-04-01 03:15:03.264116+0700 0x42a      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointLP
2019-04-01 03:15:05.210335+0700 0x564      Default     0x0                  0      0    kernel: (VoodooI2CAtmelMXT) VoodooI2CAtmelMXTTouchDriver::Touch probe
2019-04-01 03:15:05.215734+0700 0x564      Default     0x0                  0      0    kernel: (VoodooI2CAtmelMXT) VoodooI2CAtmelMXTTouchDriver::init_device: configuration space crc mismatch 00000000/003ecf3e
2019-04-01 03:15:05.215738+0700 0x564      Default     0x0                  0      0    kernel: (VoodooI2CAtmelMXT) VoodooI2CAtmelMXTTouchDriver::Unable to find message processor
2019-04-01 03:15:05.215740+0700 0x564      Default     0x0                  0      0    kernel: (VoodooI2CAtmelMXT) VoodooI2CAtmelMXTTouchDriver::Failed to init device
```

alexandred, 2019-04-06:
This is not a suitable format for a Github issue. I will close this until you provide proper troubleshooting archives as requested on the troubleshooting page.

closed 2019-04-06 by alexandred

hieubq90, 2019-04-07:
[debug_24238.zip](https://github.com/alexandred/VoodooI2C/files/3051454/debug_24238.zip)
here is debug_archive of my system.
My system has an ATML1000 touch screen and running Mojave 10.14.3.
I've patched DSDT and installed VoodooI2C & VoodooI2CHID kext in /L/E, it worked but only support single touch.
I tried to replace VoodooI2CHID by VoodooI2CAtmelMXT kext but it does not work.

williambj1, 2019-04-07:
I suppose you should provide this file on Gitter